openscad/openscad #3404: Feature request: Set the default filename from the program.

rewolff, 2020-08-14:
I have a bunch of programs that can render to either different objects from the same assembly or different variants. 

Sometimes it's useful to save the STLs under the same name, so in the slicer you can just do "reload STL from disk" to get the other variant. But often it would be nice to save the objects under say myprogram_objectA.stl and myprogram_objectB.stl . Or myprogram_3x16.stl. This then results in the gcode files also to carry that basename and I can simply hit print again on the printer if I still have a sliced version from a while back. 

I could do all this by hand, but in practice I give up after once or twice and start saving to the default filename again after one or two iterations. 

I'm suggesting something like: 

$default_basename = $mybasename + "objectA";

Where $mybasename is an automatic variable holding the basename of the main file. 
$default_basename defaults to $mybasename, and holds the filename that is suggested when I do "save as STL".

t-paul, 2020-08-14:
Setting via variable has too many issues and does not scale well for multi file export. The current idea is to be able to annotate top level modules and that could include a default file name too.

See https://github.com/openscad/openscad/wiki/Meta-Data-Use-Cases

kelvie, 2020-08-30:
@rewolff I have a quick and dirty patch that does this since I also find myself doing this over and over when trying to export different sub-parts of a single scad file.

t-paul, 2020-08-30:
As already mentioned above, doing it via a variable is not seen as viable solution.

kelvie, 2020-08-30:
Right, I didn't intend to mainline this, just to share how to do it if someone wanted something (right now) for this use case!

DaleSchultz, 2021-01-28:
> Setting via variable has too many issues and does not scale well for multi file export. The current idea is to be able to annotate top level modules and that could include a default file name too.
> 
> See https://github.com/openscad/openscad/wiki/Meta-Data-Use-Cases

My use case for this is that I have a _single_ module, and I use parameters to alter the output that the module produces. When I export to .stl file I need that file name to reflect the parameters that were in effect. I can build that into a string myself, and we need to be able to inject that/append it to the default filename. So having one file name per module would not solve the problem.

For example say my module simply creates a cube based on L, W & H variables - which are in the customizer.
When I generate a 23 x 45 x 16 cube I want to be able to generate a file name like this:
$OutputFile = str(L,"x",W,"x",H);

When I then save the stl file, the file defaults to 23x45x16.stl

I don't see the need for additional customization or meta data, we already have the data we need, we just have no programmatic way to get it into the filename. And one filename per module won't work.

larry68k, 2023-04-21:
I have 40 years of software development experience and do not see a real scaling issue with the base request of setting the output file name programmatically! Failing that there should be a function such as: "export_stl (filename);" that would export the current model.

An alternative function could be: "export_stl (modulename (module_parameters));
This version would support multiple exports from the same file cleanly and resolve the many request for naming the output stl file.

This does not require giving access to the built-in file name, although I do not see any real problem with that solution!

jordanbrown0, 2023-04-22:
Having a way to set the default STL export name seems plausible.

Having a way to **cause** an STL export does not.  Running an OpenSCAD program should not do **anything** other than display a model and put some stuff into the console log.  It should not create files, it should not pop dialog boxes, et cetera.

A function like `export_stl(modulename(module_parameters))` is not possible in the current language, because modules are not functions; you cannot call them from expression context.

jordanbrown0, 2023-04-22:
I should not have said "a function like...", because `export_stl()` would presumably be a module, not a function.  Edited.

rewolff, 2023-04-22:
There are already "modules" that do "funny" things. Like "echo". 

I would certainly use a module 

    export_stl ("thecube.stl") cube (20);
    export_stl ("theprism.stl") cylinder (d=20, h=20, $fn=3);
    
This would certainly enable a lot of neat tricks. My first use would be to simply keep my STL uptodate with my model. Just this week I changed a model and 3-D printed it with forgetting the export step -> I got the old version. 
I think it is not productive to dismiss a good idea by pointing out the small syntactic problem with quick mockup someone provides.

jordanbrown0, 2023-04-23:
There's no way to control an open source project - absolutely nothing stops you from forking it and doing whatever you like - but in my opinion OpenSCAD programs should never gain the ability to create new files or overwrite existing files without confirmation.  One of the advantages of OpenSCAD over other environments (e.g. Python) is that it is **safe** to download an OpenSCAD model and run it.  It will not destroy your files or attack your system.  Adding capabilities like this opens opportunities for malware.

I won't say that it would be impossible to come up with a scheme that would allow such a thing - perhaps, for instance, you'd have to approve each program that you give write permission to.  However, getting features like that to be trustworthy would not be simple.

```
export_stl(".login") cube();
```
```
export_stl("AppData/NTUSER.DAT") text("can you recover from a trashed registry?");
```

Maybe it can only create files in a designated directory.
```
export_stl("..\\..\\something") text("have I just escaped from the jail?");
```

And if we extrapolate a little more, to allow creating text files...

```
output(".login") echo("\nrm -rf *\n");
```

And of course those concerns recurse into libraries...

innocent_model.scad:
```
use <somelibrary.scad>
export_stl("innocent_output.stl") somepart();
```
somelibrary.scad:
```
module somepart() {
    export_stl(".login") text("gotcha!");
}
```

So maybe you say "export_stl() always appends .stl, and output() always appends .txt".  Are you sure that there are no directories writable where creating a file yields ... exciting ... results?  No directories where, say, all of the files in the directory are executed as shell scripts?

Et cetera.

Too scary, for too little value.  Again, there's nothing stopping you from implementing it, but I'm not going to.

DaleSchultz, 2023-04-23:
The request is to set the filename.
That is all that is needed.

So when we control what gets drawn, we can set the filename to be used _if and when_ the stl is exported. 

All the discussion about side effect actions and security are moot.  Exporting the file stays a user operation with all the existing overwrite warnings.

jordanbrown0, 2023-11-18:
Comments in #4317 note that one of the problems with this proposal is that other proposals want to have a single .SCAD file generate multiple STLs in one run.

rewolff, 2023-11-19:
Well... Yes, that is most of the issue here too. Having one scad file that can produce multiple objects. Having the ability to set the filename for each of the possible output objects seems like a useful feature that would help. But maybe a generic "output multiple STL files" would be even better.  

Just to inform you that this feature has practical use: I 3D printed something this week and in the slicer I noticed a small thing I had forgotten. Change it, hit render, go do something else for 5 minutes, come back, hit save in openscad, hit reload-from-disk on the slicer and hit send-to-printer. 24 hours of printing later I have an object that lacks the improvement I made. Well.... first time around I did change the default filename and second time around I didn't . 

I agree that it could be scary to have an "export_stl" module. That would require some thinking about security. Do STL files have comments? Add "created by Openscad from xxxx.scad on dd/mm/yy" . Check for that comment before silently overwriting.

kelvie, 2023-11-19:
This is precisely how I've been using this feature for over 2 years now, for example:

https://github.com/kelvie/scads/commit/e97f6a922a27cd6f151c2456603ab9fcc4784f42

Again, the way I implemented is presumably hacky, I don't claim to know all the side effects of hijacking a variable like that in this context, but for my workflow it's *really* handy to know which revision a particular STL is.

rewolff, 2024-02-11:
Cleaning up old tabs... Seeing the original question again, having read the objections.... 

How about: The openscad program may set $fname_suffix or call fname_suffix ()  module to set the filename suffix. 
Where the default filename would be <source_name>.stl, it will now be <source_name>_<the_given_suffix>.stl with the _ being dropped if there is no suffix. This reduces the options for mischief a lot and would mimic what I do manually already.

MethylBromide, 2024-05-10:
I too need a solution for these use cases. I understand I can customize the application, but I'm writing scripts for others to use and cannot insist they use my customized code -- these are designers, not programmers. They don't know how to build a project.

rewolff, 2024-05-12:
I've now adopted this "mkstl" script. Just pass it the scad file and it will search for 

```
#!/bin/sh
# mkstl (C) by REW (R.E.Wolff@BitWizard.nl)

scad=$1
# Put something like this in your scad file: 
#// Here are the outputs:
#// output 1 top
#// output 2 bot
#
#if (output == 1) translate ([110,110,0])  reelsides (bottom=false);
#if (output == 2) translate ([110,110,0])  reelsides (bottom=true);

outbn=`basename $scad .scad`

nrs=`grep '// *output' $scad | awk '{print $3}'`
for p in $nrs; do
  #ext=`grep '// *output *'$p' ' $scad | awk '{print $4}'`
  ext=`awk '/\/\/ *output *'$p' /    {print $4}' $scad`
  openscad  --enable=manifold -o $outbn"_"$ext.stl -D view=0 -D output=$p $scad 
done
```

Stuff like preventing a downloaded STL file from messing with your system is left as an excercise for the reader....

t-paul, 2025-10-30:
Yes, it's telling you what you want to hear and what sounds good. No, that feature was never implemented. That said, in Musk time, saying 2019 probably means 2035, so who knows... 

LLMs can generate nice examples by randomly mixing lots of input. Asking for specific facts is like asking for the weather in 2 weeks.

jordanbrown0, 2025-10-30:
See also the `part()` feature proposed in #1608.

jimb15122, 2026-09-20:
The resistance to using a simple special variable seems unjustified and is causing years of delay.  This is not asking for multi file export.  Just a way to programmatically set a DEFAULT export file name.  $stl_default_name = str( whatever )

jimb15122, 2026-09-20:
Actually, it should be called $output_file_name and is sole purpose is to prepopulate the export dialog with something other than the scad file name (applicable for all export types, not just stl).  -o on cli would override this if specified.

jordanbrown0, 2026-09-20:
One of the key concerns is
```
$output_file_name = ".profile";
```
or
```
$output_file_name = "/etc/passwd";
```
or
```
$output_file_name = "../../../../../../../etc/passwd";
```

Yes, there are possible answers to that concern.

jimb15122, 2026-09-20:
If the new var contained only the base file name sans extension, and extension was always appended, then overwrite risk would be mitigated. This would have no effect on cli mode and -o would still be mandatory to specify the complete file name. (with extension).  By only populating the gui save dialog, the user still gets to confirm or change the final name.

rewolff, 2026-09-22:
I think that's a reasonable restriction. While some legit use would have output/part1, output/part2 as the suggested filenames, restricting this for security reasons to no-path-allowed just basenames is a reasonable compromise. 

In my personal workflow, this feature is now outdated: I would want the option to specify a BUNCH of output filenames, and the code would generate  a different sub-part for each file. 

With the suggestion above: I'd do say: 
```
output = 1
if (output == 1) {
  $default_filename = "part1";
  part1 ();
} 
if (output == 2) {
  $default_filename = "part2";
  part2 ();
} 
```

But this requires me to change the output =1  to output=2 for each part. 

More convenient might be: 
```
$output_names = ["part1", "part2"];
if ($current_output == "part1") part1 ();
if ($current_output == "part2") part2 ();
```
which allows openscad to iterate over the output options.

jordanbrown0, 2026-09-22:
That's one of the reasons why the "just have a variable that specifies output name" schemes haven't gotten traction.

More likely is some kind of markup on the CSG tree that splits it up into separate "parts", probably with per-part metadata that is passed through into the exports.

My proposal (which also hasn't gotten traction, but I'm hopeful) is at https://github.com/openscad/openscad/wiki/OEP9:-Generating-Multiple-“Parts”-in-One-Run .

jimb15122, 2026-09-22:
Your multi-part proposal is impressive.  Perhaps a simplification could be to make the part() designation simply an alternative to "module" eg: part part1(){...}.  

If a "part level module" is used from another part-module, the child part is simply embedded into the parent part as if it were a plain module.  To summarize:  a part-module is always a top level part, but it can also be an embedded component of another part (including unnamed plain modules).  This eliminates a lot of the devilish details with unions, differencing, nesting, etc.

jordanbrown0, 2026-09-22:
Caution:  when you edit a comment after posting it, GitHub does **not** send an e-mail notification.  It's very easy for somebody who is primarily following the discussion via e-mail notifications to miss such an update.

What I personally do is to make the change, then add a **new** comment describing the change, wait for the e-mail notification to go out, and then delete the new comment.  That way the e-mail notification stream includes a full description of the comments and the web transcript is reasonably clean.

jordanbrown0, 2026-09-22:
I think you are suggesting that
```
nut(10);
translate([20,0,0]) {
    nut(10);
    translate([0,0,10]) nut(10);
}
```
would yield two parts, not three.

That would defeat major use cases for the feature.  You want to be able to have some deeply nested component be a separate part so that it can be a separate color/material/et cetera.

For instance, consider this model of a house.  The body of the house, the roof, and the chimney should all be different colors, as shown in the preview:

[image]

```
module myHouse() {
    color("blue")
        cube(10, center=true);
    translate([0,0,5])
        myRoof([10,10]);
}

module myRoof(dims) {
    height = dims.x/2;
    color("gray")
        rotate([90,0,0])
            linear_extrude(dims.y, center=true) {
                polygon([
                    [ -dims.x/2, 0 ],
                    [ 0, height ],
                    [ dims.x/2, 0 ]
                ]);
            }
    translate([0,0,height])
        myChimney();
}

module myChimney() {
    color("red")
        translate([0,0,0])
            cube([2,2,5], center=true);
}

myHouse();
```

To get that to print as different colors, you need to have each component be a separate part.

This is especially an issue for libraries like BOSL2, where components are naturally nested to position them relative to one another.

```
include <BOSL2/std.scad>
cuboid(10) {
    attach(TOP, BOTTOM)
    cuboid(10);
}
```
where you want the two cuboids to be different colors or materials.

jimb15122, 2026-09-24:
Perhaps adding the concept of an ***assembly*** would help.  An assembly is just a module with a flag that indicates its name and all of the name/value pairs of its args are registered in a list.  during export, the list is made available to choose what to export, using the chosen assembly name as the default file name.

Given chimney(), roof(), wall() are existing modules;

assembly house(style="Cape Cod")
{// A MODULE that defines a specialized arrangement of walls, roofs, and chimneys based on the requested style ...}

assembly neighborhood(name="Bloomfield")
{// A MODULE that defines a specialized arrangement of houses  based on the requested name ...}

// instantation
house(style="Ranch");  
house(style="Row");
neighborhood();

Export List Options:
"All"
"house.style.Ranch"
"house.style.Row"
"neighborhood.name.Bloomfield"

The list options appear only if one or more assemblies are instantiated.

for advanced multi part export formats, 
the "all" option would preserve the named assemblies inside.

Note:  This addresses the naming and multi part issues only.  It would only help with color/material if assemblies were designed to be material specific. (I.e design an assembly for each material)

nophead, 2026-09-24:
NopSCADlib has assemblies and can export multiple STLs.

jordanbrown0, 2026-09-24:
> Note: This addresses the naming and multi part issues only. It would only help with color/material if assemblies were designed to be material specific. (I.e design an assembly for each material)

Requiring components of different colors to be separated at the top level makes them much much harder to work with... which was the point of my house demonstration.

> NopSCADlib has assemblies and can export multiple STLs.

With the assistance of an external helper script, I assume, since OpenSCAD itself doesn't have that capability.  Sure, I have that too, and it's the approach for multi-color printing that I recommend at the moment.  But it's not a **good** approach.