openscad/openscad #3405: Feature request: chamfer for linear_extrude

richfelker, 2020-08-14:
I'm well-aware that advanced loft-type functionality for `linear_extrude` is a hard open problem. However, a commonly-needed subset of that is the ability to chamfer the edges of a 2D shape extruded with `linear_extrude` at the beginning/end. Ultimately this can be reduced to (built up from) multiple `linear_extrude` instances where the middle one is normal and the first/last apply the equivalent of the 2D `offset` module with successively negative `delta` to the child. Note that this is different from using the `scale` parameter except for a very specific class of shapes.

Presently this can be approximated (fairly well) by applying an intersection of skew transformations at angles of 0..360 with fine enough step (really just needs to hit the angles that apppear in the 2D shape), but it would be really nice (better quality, less computational overhead) to have it as a first-class feature of `linear_extrude`.

I think this feature is doable without significant extension to the design (independent of any larger lofty goals, pardon the pun).

thehans, 2020-08-14:
Duplicate of #2077

It's actually a really difficult problem to get right, because of the way that offset may merge vertices, create new loops, remove loops entirely, etc.  
With the way clipper is implemented (the 2D lib we use for offsets), we don't get any information back about *which* vertices were removed, so knowing which vertices to connect between layers becomes the main problem.

closed 2020-08-14 by thehans

richfelker, 2020-08-14:
Indeed, that does make it somewhat harder. For the use I have in mind, you could assume new loop or loop removal don't hapen and treat it as an error condition if they do because the offset is just supposed to be a chamfer. But indeed I see why the more general problem is hard, and even why matching up vertices is nontrivial. I'll follow the issue you marked this duplicate-of since there does seem to be promising progress on it. Thanks!

your-friend-alice, 2021-11-07:
workaround (for certain cases):
```
minkowski() {
    linear_extrude(0.1) text("test");
    cylinder(d1=4, d2=0, h=4);
}
```