openscad/openscad #3408: feature requestion: why can't modules return values?  is there a better pattern for this?

jacksonloper, 2020-08-16:
Hi guys.  Loving openscad so far.  Looking at this bevel gear module, https://www.thingiverse.com/thing:3833810.  I've come across a scenario where it would seem super great if a module could return a vector *as well* as create an object.  But maybe I'm thinking about this wrong.  Here's my story:

The `bevel_pair` module makes two different objects (two bevel gear which line up together).  Along the way, it computes some handy things like the radius of each gear, center of each gear, etc.  

So now let's say I make my bevel pair, and then I want to run a cylinder through the middle of the second gear.  As far as I can tell, here's what I have to do:
```
bevel_pair(...)
center_of_second_gear = [.. do some calculations that actually were already done earlier ..]
radius_of_second_gear = [.. do some calculations that actually were already done earlier ..]
translate(center_of_second_gear) cylinder(height,radius_of_second_gear*.9,radius_of_second_gear*.9)
```

In other words I have to do a redundant recomputation to figure out where to put my cylinder.  That's silly.  

Is there a better way?  Allowing modules to return values (as well as create things) would be an obvious solution, but maybe that would be antipattern here and there's a Right Way to do this in openscad?  

Thoughts?

t-paul, 2020-08-16:
Discussion about possible extension to the language to support that in #3087/#3088. For general discussion, please use the [forum/mailing list](http://forum.openscad.org/).

closed 2020-08-16 by t-paul

thehans, 2020-08-16:
With the current design of the language, modules are "statements" and behave differently from "expressions", which function calls are classified as.

The distinction is important in terms of parsing, because this is how it knows whether a function is being called vs a module.  Basically if a call is on the right hand side of an assignment then its an expression/function call.  If its just on its own line of script then it must be a statement/module call.  modules and function have their own namespaces, so you can have both with the same name.  Letting modules be called like functions would mean mixing these namespaces together making that no longer possible.

t-paul, 2020-08-16:
Yep, function literals solve that by creating a language feature that represents a function which is assigned to a variable and therefor lives in the variable namespace. Doing something similar for modules is a bit trickier as can be seen in the linked discussions, but should be doable too.

jacksonloper, 2020-08-16:
i see!  great (function-literal-related) things are afoot.  godspeed!