openscad/openscad #3411: Feature Request: "segmentation" options for linear_extrude

thehans, 2020-08-19:
I figured I'd open an issue for this since there's currently a related  [discussion topic on the mailing list](http://forum.openscad.org/Problem-using-difference-with-linear-extrude-tc29998.html).

The problem is when using `linear_extrude` with `twist` (and to a lesser extent a non-uniform `scale` parameter) on objects with long thin edges, such as this:
```
linear_extrude(50, twist=360,slices=20) square([10,2], center=true);
```
Results in skinny triangle faces with large differences in their normal vectors.  Attempting to `slices` only makes the triangles skinnier, but the zigzag ripples are still there.
If the long edges were divided into smaller colinear line segments, this annoying modeling artifact would go away.

So I think that a builtin option for `linear_extrude` to split these long edges would be pretty helpful, and fairly simple to implement.  The main question is how we want to design the interface: what new parameter name(s) to use and how they work?

Based on previous discussion (see quotes below), I'm thinking that a new parameter should be added to `linear_extrude`, call it `max_edge_length` or something like that, and it would function analogously to how `$fs` does on circles: 
any edge longer than `max_edge_length` would be divided into `n` equal segments, where `n` is roughly `current_edge_length / max_edge_length` (I say roughly since we could use ceil or round on the result, but I'm not sure which would be preferred)

Alternatively we could have a boolean `split_edges` parameter, where setting to `true` would let this "segmentation" feature use `$fs` (and maybe `$fa`?) **or** `$fn` to calculate edge segment lengths, depending on which are defined in the current scope.
If this approach were used, then `split_edges` should probably default to false for backwards compatibility.

There was also some comments about this in my PR #3351, but I thought it would be better to separate this into its own issue / PR for further discussion.
For reference, I've quoted the relevant comments from that PR below:

_Originally posted by @nophead in https://github.com/openscad/openscad/pull/3351#issuecomment-643145170_
> When I extrude a shape like that I add extra vertices along along the edges so the quads are roughly square. That gets rid of the ugly stripes that give a rippled appearance. It would be nice if linear extrude did that when necessary.

_Originally posted by @thehans in https://github.com/openscad/openscad/pull/3351#issuecomment-643523179_
> @nophead I've been thinking about that issue too.  
> Particularly when long, thin rectangle is extruded with twist, it ends up looking like a staircase no matter how many slices, because the long edge just sweeps too much for its length.
> 
> A small example:
> ```
> translate([-20,0]) linear_extrude(50, twist=360,slices=20) translate([0,-.1]) square([10,0.2]);
> translate([  0,0]) linear_extrude(50, twist=360, slices=20) polygon([for(x=[0:10]) [x,-0.1], for(x=[10:-1:0]) [x,0.1] ]);
> translate([ 20,0]) for(x=[0:10]) linear_extrude(50, twist=360, slices=20) translate([x,0]) square([1,0.2]);
> ```
> First line illustrates the problem.
> Second line was an attempted workaround, but polygon seems to automatically merge colinear vertices, so that is just the same model as the first.
> Third line works, but then you gotta waste time in CGAL to union all the separate extrusions.
> 
> So yeah it would be best if linear_extrude could automatically split long edges into shorter segments.  There's a few different ways this could be done... 
> 
> First I would think `$fn` could count as a minimum # of vertices *per outline*.  So any outline with less than `$fn` vertices would have extras inserted, but then there's the question of how to divide up  new vertices among edges.  Thinking as I write this... you could basically track a list of `<edge_length, segment_count>` pairs.  Each edge would start at `segment_count=1`, and whichever `edge_length/segment_count` is largest gets its count incremented until `$fn is reached.  
> Then go through and update the polygon once you know where and how many splits to make.
> That should make things about as even as possible I think.
> 
> For `$fs`, the code could split each edge into segments as `segments = round(edge_length / $fs)` (or maybe ceil instead of round?). Definitely the most straightforward.
> 
> I don't think `$fa` would be applicable for segmenting a 2D polygon though.  I suppose you could cast rays out from the "center of mass" at specific angles and spit the first edge each ray hits, but that would only work for convex shapes, and probably would result in an ugly mess most of the time.
> 
> The other question is should `linear_extrude` always do this?  If user has `$fn` or `$fs` set should it just split according to those (with `$fn` overriding `$fs` as usual), or should it require a new parameter flag, like `split=true` to enable that feature?
> When `slices` is given, then linear_extrude uses that without the need for the special variables in splitting up the height.  In that case should it also use `slices` as a number of segments per outline?
> Or should it have additional, completely independent parameter(s) like `max_edge_length=[num]` or `min_outline_segments=[num]`?
> 

_Originally posted by @nophead in https://github.com/openscad/openscad/pull/3351#issuecomment-643594011_
> To get around the merging of colinear vertices I perturbed them slightly when modelling a fan blade with this:
> 
> ```
>     module squarish(s, n) {
>         polygon([
>             for(i = [0 : n]) [i * s.x / n,  s.y + (i % 2) * eps],
>             for(i = [0 : n]) [s.x - i * s.x / n, (i % 2) * eps],
>         ]);
>     }
> ```
> 
> [image]
> 
> I think the aim is to make the quads square, so dividing the helix length made from the height and the twist by the number of slices gives the quad height and then the perimeter is divided into segments roughly the same length.
> 
> It is probably only needed when there is a twist. I think it should be optional for backwards compatibility. What would happen to my fan blades that are already divided up roughly to the same rule? I wouldn't want extra vertices to appear. 
> 
> Making it configurable seems more logical, perhaps ```segments = [n]``` to match ```slices = [n]```' but it requires a little more work from the user assuming roughly square quads is always optimum. 
> 
> I personally don't think it should use $fs because I have it  and $fa globally set for 3D printing curves. I think it would in general be too fine for linear_extrude unless $fa was also used in the calculation. I.e. calculate the perimeter length and treat it as a circle circumference to apply $fn, $fa and $fs but that seems a bit hacky.
>

rcolyer, 2020-08-19:
I consider twist unusable for any of my designs in the manner it currently operates, specifically because of this staircase effect.  Pretty much any way you slice it (hah), twist is making a complicated curved object, but it currently acts like it's trying to reproduce whatever block shaped template was given as an input.  Therefore I would endorse that $fs be used here by default for segmentation.  It should provide a symmetrically smooth result the same as if it were trying to make a sphere.  If there is a role for $fa in this, to be comparable to the meaning of the variable elsewhere, it would be to calculate the amount of segmentation at which the angle between the normals of the staircase faces becomes less than $fa, and choose the number of the segments by the least of that value or the one given by $fs.

MichaelAtOz, 2020-08-19:
I think a more sensible default for slices should also be in the mix.
The low default often results in long thin triangles and generate self-intersections.
I'll try to dig up the cases, I recall covering it here a few times.

nophead, 2020-08-19:
Didn't @thehans fix the self intersections by splitting the quads along the diagonal that makes the ripples convex? 

I don't think it should reuse ```$fn```, ```$fs``` and ```$fa``` because ```slices``` is set explicitly and I think the optimum segments makes the quad patches square. So I think it the simplest backwards compatible consistent solution is to let the user also set segments explicitly.

rcolyer, 2020-08-21:
Reusing $fs and $fa should be done specifically because they are currently used by linear_extrude with twist, but they are used asymmetrically in just the vertical axis, causing this issue.  Making use of them is a highly desired behavior, because if I put them at the top of my scad file and don't override them anywhere, I expect all built-ins to obey the contract and exhibit a smoothness on the scale I specified with these values.  Currently linear_extrude does this partially, and what we need is to complete it on the other axis.  The slices variable is just a redundant vestigial parameter that behaves exactly like specifying $fn manually in the linear_extrude.

If one wants a linear_extrude with twist to be more blocky than the top-level $fs and $fa specify, then the correct practice in line with other built-ins to set them locally back to a larger value.

nophead, 2020-08-21:
I don't think that is true. It uses the slices parameter to set the number of vertical slices and that determines how much twist each slice has.

MichaelAtOz, 2020-09-04:
> I recall self intersections with the small number problem due to the grid.

I've been looking for the cases I mentioned, I found one, I haven't finished looking because fiddling with it, I'm either confused (probably) or there is a problem.

How is `slices=` meant to work, with and without twist/scale, and in Preview and either `render()` preview or F6 render?

Without other parameters (no twist) 2019.05 shows the triangulated slices in preview, single quads on render and in STL. 2020.08.21 (PR3351) the slices get to the STL. For a simple object:
[image]

I repackaged [this](https://github.com/openscad/openscad/issues/3168#issuecomment-568625496) problem, which produces self-intersections (SIs), into:
[pencil-b.zip](https://github.com/openscad/openscad/files/5172605/pencil-b.zip)

With 2020.08.21 (PR3351), no other parameters (no twist), slices show in Preview:
[image]
Not in Render:
[image]
Not on export:
[image]
With 4 SIs at the tip.
With the Original $fn's, more SI's:
[image]

With a twist it keeps the layers, and no SI's:
[image]

Without twist, the simple cuboid has layers, but the pencil does not, and results in SI's.

My next post will show a nasty triangulation.

MichaelAtOz, 2020-09-04:
The last image above had tw=20.

With tw=2; fnPillar=40; fnSharpner=60;

Preview:
[image]

Render:
[image]

Close up:
[image]

Those dots are where the layers were, but they are not segmented.

STL close up, SI's:
[image]

Long skinny triangles + small number problem + Grid = not good.

I suspect there will be many corner cases.

That needs fixing, but the above demonstrates that the default low `slices` causes problems.

thehans, 2020-09-04:
> How is `slices=` meant to work, with and without twist/scale, and in Preview and either `render()` preview or F6 render?

If you explicitly provide the `slices` parameter, then `linear_extrude` will use that value, full stop.  Nothing about that should have changed in the recent PR.  
The issue you're seeing with render is, after the mesh has gone through some CGAL operation, where I think it automatically merges co-planar faces.  I'm not sure if there's any way to change CGAL's behavior in that case.

Anyways, this is all off topic from the current issue which is about segmenting the *2D outline*, pre-extrusion, this would be orthogonal to `slices`

thehans, 2021-03-01:
What would people think about splitting edges as a separate operation, not tied to `linear_extrude`?  

Here's my proposed module call signature:  ```split_edges(max_length = $fs)```
This way the max_length can conveniently default to $fs without breaking any backwards compatibility.

I also like this idea because
1) It avoids further cluttering `linear_extrude`'s argument list.  
2) In case other schemes for splitting edges are desired in the future, it would be simple to add other optional parameters.
3) It might be useful for other cases besides `linear_extrude()` though I can't think of any right now... except maybe to use with the proposed function-literal transform discussed in this [forum thread](http://forum.openscad.org/Passing-function-literals-to-builtin-modules-tc31423.html)

nophead, 2021-03-01:
Makes sense but I think the target is always to make the quads square, so the user has to work out the slice thickness and pass that.

thehans, 2021-05-26:
For anyone wanting to test the feature, PR #3770 has segments implemented.

closed 2021-06-08 by t-paul