openscad/openscad #3412: Build dependencies error (macOS)

gingerbeardman, 2020-08-19:
I'm on 10.14.6 (18G4032)

```
$ ./scripts/macosx-build-dependencies.sh
Detected Mojave (10.14) or later
Building for 10.9 or later
Setting number of CPUs to 6
Using basedir: /Volumes/External/Users/matt/Projects/openscad/../libraries
double_conversion eigen gmp mpfr glew gettext libffi freetype ragel harfbuzz libzip libxml2 fontconfig hidapi libuuid lib3mf glib2 boost cgal qt5 opencsg qscintilla
Building packages: double_conversion eigen gmp mpfr glew gettext libffi freetype ragel harfbuzz libzip libxml2 fontconfig hidapi libuuid lib3mf glib2 boost cgal qt5 opencsg qscintilla 

double_conversion 3.1.5 already installed - not building
Building eigen 3.3.7 ...
tar: Unrecognized archive format
tar: Error exit delayed from previous errors.
tar: Unrecognized archive format
tar: Error exit delayed from previous errors.
```

verbose

```
$ ./scripts/macosx-build-dependencies.sh -v
+ BASEDIR=/Volumes/External/Users/matt/Projects/openscad/../libraries
+ OPENSCADDIR=/Volumes/External/Users/matt/Projects/openscad
+ SRCDIR=/Volumes/External/Users/matt/Projects/openscad/../libraries/src
+ DEPLOYDIR=/Volumes/External/Users/matt/Projects/openscad/../libraries/install
+ MAC_OSX_VERSION_MIN=10.9
+ OPTION_DEPLOY=false
+ OPTION_FORCE=0
+ PACKAGES=("double_conversion 3.1.5" "eigen 3.3.7" "gmp 6.1.2" "mpfr 4.0.2" "glew 2.1.0" "gettext 0.19.8" "libffi 3.2.1" "freetype 2.9.1" "ragel 6.10" "harfbuzz 2.3.1" "libzip 1.5.1" "libxml2 2.9.9" "fontconfig 2.13.1" "hidapi 0.9.0" "libuuid 1.6.2" "lib3mf 1.8.1" "glib2 2.56.3" "boost 1.65.1" "cgal 4.13" "qt5 5.9.7" "opencsg 1.4.2" "qscintilla 2.11.2")
+ DEPLOY_PACKAGES=("sparkle 1.21.3")
+ '[' '!' -f /Volumes/External/Users/matt/Projects/openscad/openscad.pro ']'
+ OPENSCAD_SCRIPTDIR=/Volumes/External/Users/matt/Projects/openscad/scripts
+ getopts 3lcdfv c
+ case $c in
+ echo verbose on
verbose on
+ getopts 3lcdfv c
+ OPTION_PACKAGES=
++ sw_vers -productVersion
++ cut -d. -f2
+ OSX_VERSION=14
+ ((  14 >= 14  ))
+ echo 'Detected Mojave (10.14) or later'
Detected Mojave (10.14) or later
+ echo 'Building for 10.9 or later'
Building for 10.9 or later
+ '[' '!' ']'
++ sysctl -n hw.ncpu
+ NUMCPU=6
+ echo 'Setting number of CPUs to 6'
Setting number of CPUs to 6
+ false
+ ((  0  ))
+ echo 'Using basedir:' /Volumes/External/Users/matt/Projects/openscad/../libraries
Using basedir: /Volumes/External/Users/matt/Projects/openscad/../libraries
+ mkdir -p /Volumes/External/Users/matt/Projects/openscad/../libraries/src /Volumes/External/Users/matt/Projects/openscad/../libraries/install /Volumes/External/Users/matt/Projects/openscad/../libraries/install/share/macosx-build-dependencies
+ false
++ all_packages
+++ seq 0 21
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=double_conversion 3.1.5'
++ echo -n 'double_conversion '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=eigen 3.3.7'
++ echo -n 'eigen '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=gmp 6.1.2'
++ echo -n 'gmp '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=mpfr 4.0.2'
++ echo -n 'mpfr '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=glew 2.1.0'
++ echo -n 'glew '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=gettext 0.19.8'
++ echo -n 'gettext '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=libffi 3.2.1'
++ echo -n 'libffi '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=freetype 2.9.1'
++ echo -n 'freetype '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=ragel 6.10'
++ echo -n 'ragel '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=harfbuzz 2.3.1'
++ echo -n 'harfbuzz '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=libzip 1.5.1'
++ echo -n 'libzip '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=libxml2 2.9.9'
++ echo -n 'libxml2 '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=fontconfig 2.13.1'
++ echo -n 'fontconfig '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=hidapi 0.9.0'
++ echo -n 'hidapi '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=libuuid 1.6.2'
++ echo -n 'libuuid '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=lib3mf 1.8.1'
++ echo -n 'lib3mf '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=glib2 2.56.3'
++ echo -n 'glib2 '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=boost 1.65.1'
++ echo -n 'boost '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=cgal 4.13'
++ echo -n 'cgal '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=qt5 5.9.7'
++ echo -n 'qt5 '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=opencsg 1.4.2'
++ echo -n 'opencsg '
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=qscintilla 2.11.2'
++ echo -n 'qscintilla '
+ ALL_PACKAGES='double_conversion eigen gmp mpfr glew gettext libffi freetype ragel harfbuzz libzip libxml2 fontconfig hidapi libuuid lib3mf glib2 boost cgal qt5 opencsg qscintilla '
+ echo double_conversion eigen gmp mpfr glew gettext libffi freetype ragel harfbuzz libzip libxml2 fontconfig hidapi libuuid lib3mf glib2 boost cgal qt5 opencsg qscintilla
double_conversion eigen gmp mpfr glew gettext libffi freetype ragel harfbuzz libzip libxml2 fontconfig hidapi libuuid lib3mf glib2 boost cgal qt5 opencsg qscintilla
+ '[' -z '' ']'
+ OPTION_PACKAGES='double_conversion eigen gmp mpfr glew gettext libffi freetype ragel harfbuzz libzip libxml2 fontconfig hidapi libuuid lib3mf glib2 boost cgal qt5 opencsg qscintilla '
+ echo 'Building packages: double_conversion eigen gmp mpfr glew gettext libffi freetype ragel harfbuzz libzip libxml2 fontconfig hidapi libuuid lib3mf glib2 boost cgal qt5 opencsg qscintilla '
Building packages: double_conversion eigen gmp mpfr glew gettext libffi freetype ragel harfbuzz libzip libxml2 fontconfig hidapi libuuid lib3mf glib2 boost cgal qt5 opencsg qscintilla 
+ echo

+ for package in '$OPTION_PACKAGES'
+ [[ double_conversion eigen gmp mpfr glew gettext libffi freetype ragel harfbuzz libzip libxml2 fontconfig hidapi libuuid lib3mf glib2 boost cgal qt5 opencsg qscintilla  =~ double_conversion ]]
++ package_version double_conversion
+++ seq 0 21
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=double_conversion 3.1.5'
++ '[' double_conversion = double_conversion ']'
++ echo 3.1.5
++ return 0
+ build double_conversion 3.1.5
+ local package=double_conversion
+ local version=3.1.5
+ local should_install=0
+ [[ 0 == 0 ]]
+ is_installed double_conversion 3.1.5
+ check_version_file double_conversion 3.1.5
+ versionfile=/Volumes/External/Users/matt/Projects/openscad/../libraries/install/share/macosx-build-dependencies/double_conversion.version
+ '[' -f /Volumes/External/Users/matt/Projects/openscad/../libraries/install/share/macosx-build-dependencies/double_conversion.version ']'
++ cat /Volumes/External/Users/matt/Projects/openscad/../libraries/install/share/macosx-build-dependencies/double_conversion.version
+ [[ 3.1.5 == 3.1.5 ]]
+ return 0
+ echo 'double_conversion 3.1.5 already installed - not building'
double_conversion 3.1.5 already installed - not building
+ return 0
+ [[ 0 == 1 ]]
+ for package in '$OPTION_PACKAGES'
+ [[ double_conversion eigen gmp mpfr glew gettext libffi freetype ragel harfbuzz libzip libxml2 fontconfig hidapi libuuid lib3mf glib2 boost cgal qt5 opencsg qscintilla  =~ eigen ]]
++ package_version eigen
+++ seq 0 21
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=double_conversion 3.1.5'
++ '[' eigen = double_conversion ']'
++ for i in '$(seq 0 $(( ${#PACKAGES[@]} - 1 )) )'
++ local 'p=eigen 3.3.7'
++ '[' eigen = eigen ']'
++ echo 3.3.7
++ return 0
+ build eigen 3.3.7
+ local package=eigen
+ local version=3.3.7
+ local should_install=0
+ [[ 0 == 0 ]]
+ is_installed eigen 3.3.7
+ check_version_file eigen 3.3.7
+ versionfile=/Volumes/External/Users/matt/Projects/openscad/../libraries/install/share/macosx-build-dependencies/eigen.version
+ '[' -f /Volumes/External/Users/matt/Projects/openscad/../libraries/install/share/macosx-build-dependencies/eigen.version ']'
+ return 1
+ return 1
+ should_install=1
+ [[ 1 == 1 ]]
+ set -e
+ build_eigen 3.3.7
+ version=3.3.7
+ echo 'Building eigen' 3.3.7 ...
Building eigen 3.3.7 ...
+ cd /Volumes/External/Users/matt/Projects/openscad/../libraries/src
+ rm -rf eigen-3.3.7
+ '[' '!' -f eigen-3.3.7.tar.bz2 ']'
++ tar tjf eigen-3.3.7.tar.bz2
++ head -1
++ cut -f1 -d/
tar: Unrecognized archive format
tar: Error exit delayed from previous errors.
+ EIGENDIR=
+ rm -rf
+ tar xjf eigen-3.3.7.tar.bz2
tar: Unrecognized archive format
tar: Error exit delayed from previous errors.
```

t-paul, 2020-08-19:
Hmm, looks like the URL changed again (Bitbucket complains about Repo unavailable). I'll try to find what the new URL is.

t-paul, 2020-08-19:
Looks like `eigen` moved from BitBucket to GitLab. New URL is https://gitlab.com/libeigen/eigen/-/archive/3.3.7/eigen-3.3.7.tar.bz2 so the script line would need to look like
```
curl -LO https://gitlab.com/libeigen/eigen/-/archive/$version/eigen-$version.tar.bz2
```

closed 2020-08-19 by t-paul

t-paul, 2020-08-19:
Scripts are updated, but I could only test on the Linux version. Let me know if that works for you.

I don't know if the download fail did create a broken (empty?) archive file locally, so if it's not trying to download eigen again, clean up the download/build folder (usually called `openscad_deps`).

clemens-tolboom, 2020-09-09:
FWIW I got a timeout on building libzip. Is seems to me https://libzip.org is offline.

### Commands

```
  502  git clone git://github.com/openscad/openscad.git
  503  cd openscad/
  504  git submodule update --init
  505  source setenv_mac.sh
  506  ./scripts/macosx-build-dependencies.sh
```

### "end" result

```
+ build_libzip 1.5.1
+ version=1.5.1
+ echo 'Building libzip 1.5.1...'
Building libzip 1.5.1...
+ cd /Users/clemens/Downloads/openscad/../libraries/src
+ rm -rf libzip-1.5.1
+ '[' '!' -f libzip-1.5.1.tar.gz ']'
+ curl -LO https://libzip.org/download/libzip-1.5.1.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:01:17 --:--:--     0curl: (7) Failed to connect to libzip.org port 443: Operation timed out
```

t-paul, 2020-09-09:
Looks like there's also releases via github: https://github.com/nih-at/libzip/releases/download/v1.7.3/libzip-1.7.3.tar.gz

clemens-tolboom, 2020-09-09:
Wow ... quick response ... I will not try again. I'll get the binary for now. I was just curious about the build process for no particular reason but do not want to run after dependencies :-/

clemens-tolboom, 2020-09-09:
See my PR https://github.com/openscad/openscad.github.com/pull/90 ;-) ... thanks

t-paul, 2020-09-09:
Thanks, right now we have the problem that there's nobody available with a recent enough MacOS to work on those MacOS specific issues. My old MacBook was left behind some time ago unfortunately.

clemens-tolboom, 2020-09-09:
Hmmm ... https://github.com/openscad/openscad/blob/master/.travis.yml#L33 build a MacOS and I see

```
      before_install:
        - test -d /usr/local/opt/opencsg/lib || { rmdir /usr/local/opt/opencsg; brew install opencsg; }
        - test -d /usr/local/opt/libzip/lib || { rmdir /usr/local/opt/libzip; brew install libzip; }
        - ./scripts/macosx-build-homebrew.sh
```

which suggest it downloads `libzip` through homebrew. On my 10.14.6 I get

```
brew info libzip
libzip: stable 1.7.3 (bottled)
C library for reading, creating, and modifying zip archives
https://libzip.org/
Conflicts with:
  libtcod (because libtcod, libzip and minizip2 install a `zip.h` header)
  minizip2 (because libtcod, libzip and minizip2 install a `zip.h` header)
/usr/local/Cellar/libzip/1.6.1 (135 files, 582.2KB)
  Poured from bottle on 2020-05-28 at 08:27:48
/usr/local/Cellar/libzip/1.7.3 (142 files, 609KB) *
  Poured from bottle on 2020-08-25 at 20:11:35
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/libzip.rb
License: BSD-3-Clause
```

which seemed to be installed 6 month ago.

```
brew list libzip
/usr/local/Cellar/libzip/1.7.3/bin/zipcmp
/usr/local/Cellar/libzip/1.7.3/bin/zipmerge
/usr/local/Cellar/libzip/1.7.3/bin/ziptool
/usr/local/Cellar/libzip/1.7.3/include/ (2 files)
/usr/local/Cellar/libzip/1.7.3/lib/libzip.5.3.dylib
/usr/local/Cellar/libzip/1.7.3/lib/cmake/ (4 files)
/usr/local/Cellar/libzip/1.7.3/lib/pkgconfig/libzip.pc
/usr/local/Cellar/libzip/1.7.3/lib/ (2 other files)
/usr/local/Cellar/libzip/1.7.3/share/man/ (122 files)
```

So I do not get my particular failure :-(

Is `brew install libzip` and `brew install opencsg` a good PR?

t-paul, 2020-09-09:
There's 2 build options:
- based on homebrew, this is fine for local builds, but has issues distributing to other system on other OS versions
- based on the build script to compile all dependencies from source, this is used for release builds

Travis uses homebrew as that version is not distributed but used for running the test suite.

That special treatment in 'before_install' is only due to the caching on Travis as the builds there are starting to get near the allowed time limit. I don't remember all the details though.

commit 67fafca8 (2020-08-19): Merge pull request #3414 from openscad/update-eigen-url

Update eigen download URL to gitlab (fixes #3412).
commit 8dc73116 (2020-08-19): Update eigen download URL to gitlab (fixes #3412).