openscad/openscad #3413: macOS checklist outdated

gingerbeardman, 2020-08-19:
I am trying to build on macOS to contribute.

http://svn.clifford.at/openscad/trunk/doc/checklist-macosx.txt

refers to `http://qt.nokia.com/downloads/qt-for-open-source-cpp-development-on-mac-os-x`

which was last online in 2013.

t-paul, 2020-08-19:
That file does not relate to this repo either, so I'm not sure what you are asking to do.

gingerbeardman, 2020-08-19:
OK, it's referenced from the wiki.

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Building_on_Mac_OS_X

Do what you want with this information.

I give up.

closed 2020-08-19 by gingerbeardman

t-paul, 2020-08-19:
I'm happy to help as much as I can with MacOS builds but I don't have a working MacOS system myself and I don't know all possible documents someone put out that might have broken links. So please understand the confusion here with just the information provided in the original message.

gingerbeardman, 2020-08-19:
Understood, thanks for following up.