openscad/openscad #3418: Rendering artifacts for difference during preview under virglrenderer

kelvie, 2020-08-24:
This is on my Pixelbook on chromeOS, which uses lxc containers inside KVM.

[image]

This is the logo example, notice the green outlines for differences. The difference outline is generally always shown in preview mode, they go away after rendering.

```
OpenSCAD Version: 2019.05
System information: Linux 5.4.40-04224-g891a6cce2d44 #1 SMP PREEMPT Tue Jun 23 20:21:29 PDT 2020 x86_64 Debian GNU/Linux bullseye/sid 2 CPUs 6.48 GB RAM
User Agent: OpenSCAD/2019.05 (Linux x86_64; Debian GNU/Linux bullseye/sid)
Compiler: GCC "5.4.0 20160609" 64bit
MinGW build: No
Debug build: No
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.7, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.11.2
QScintilla version: 2.9.1
InputDrivers: 
GLib version: 2.48.2
lodepng version: 20180910
libzip version: 1.0.1
fontconfig version: 2.11.94 (runtime: 2.13.1)
freetype version: 2.6.1 (runtime: 2.10.2)
harfbuzz version: 1.0.1 (runtime: 2.6.7)
lib3mf version: 1.8.1
Application Path: /tmp/.mount_OpenSCv86NmD/usr/bin
Documents Path: /home/kelvie/.local/share
User Documents Path: /home/kelvie
Resource Path: /tmp/.mount_OpenSCv86NmD/usr/share/openscad
User Library Path: /home/kelvie/.local/share/OpenSCAD/libraries
User Config Path: /home/kelvie/.config/OpenSCAD
Backup Path: /home/kelvie/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/kelvie/.local/share/OpenSCAD/libraries
/tmp/.mount_OpenSCv86NmD/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
/usr/share/fonts
/usr/local/share/fonts
/home/kelvie/.local/share/fonts
/home/kelvie/.fonts
/mnt/chromeos/fonts
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
/home/kelvie/Library/Fonts
/usr/share/fonts/X11
/usr/share/fonts/truetype
/mnt/chromeos/fonts/croscore
/mnt/chromeos/fonts/crosextra
/mnt/chromeos/fonts/google-sans
/mnt/chromeos/fonts/ko-nanum
/mnt/chromeos/fonts/lohit-cros
/mnt/chromeos/fonts/monotype
/mnt/chromeos/fonts/noto
/mnt/chromeos/fonts/notocjk
/mnt/chromeos/fonts/roboto
/mnt/chromeos/fonts/tibt-jomolhari
/mnt/chromeos/fonts/util
/usr/share/fonts/X11/encodings
/usr/share/fonts/X11/misc
/usr/share/fonts/X11/util
/usr/share/fonts/truetype/croscore
/usr/share/fonts/truetype/dejavu
/usr/share/fonts/truetype/roboto
/usr/share/fonts/X11/encodings/large
/usr/share/fonts/truetype/roboto/unhinted
/usr/share/fonts/truetype/roboto/unhinted/RobotoTTF
GLEW version: 1.13.0
OpenGL Version: 3.1 Mesa 19.2.8
GL Renderer: virgl
GL Vendor: Red Hat
RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8880), depth(0), stencil(0)

GL Extensions:
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_multi_draw_indirect
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_clear_buffer_object
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_parallel_shader_compile
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_polygon_offset_clamp
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robust_buffer_access_behavior
GL_ARB_robustness
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_bit_encoding
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_filter_anisotropic
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_draw_buffers
GL_ATI_fragment_shader
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_texture
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_buffer_object
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_R8
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_IBM_rasterpos_clip
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_KHR_blend_equation_advanced
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_texture_compression_astc_ldr
GL_KHR_texture_compression_astc_sliced_3d
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

MichaelAtOz, 2020-08-24:
First try Edit/Preferences/Advanced/Force-goldfeather, probably won't work, but it simple first test.

Failing that you probably need software render, @tp what's the Linux equivalent of opengl32.dl?

kelvie, 2020-08-24:
@MichaelAtOz I've tried that, and yup, it didn't work.

Software render of course is super slow. This is likely just a bug in the opengl implementation (virglrenderer), as it does claim to support OpenGL 3.1

The eqipvalent of opengl32.dll is probably libGL.so

nophead, 2020-08-24:
The problem seems to be that there is no depth buffer.

OpenGL Version: 3.1 Mesa 19.2.8
> GL Renderer: virgl
> GL Vendor: Red Hat
> RGBA(8880), depth(0), stencil(0)
> GL_ARB_framebuffer_object: yes
> GL_EXT_framebuffer_object: yes
> GL_EXT_packed_depth_stencil: yes Qt graphics widget: QOpenGLWidget
> QSurfaceFormat: RGBA(8880), depth(0), stencil(0)

t-paul, 2020-08-24:
I'd assume that would be [`LIBGL_ALWAYS_SOFTWARE=true`](https://docs.mesa3d.org/envvars.html).

Yeah, that looks like depth/stencil buffer not working.

kelvie, 2020-08-25:
OK thanks. I've filed a bug upstream (ish): https://bugs.chromium.org/p/chromium/issues/detail?id=1121453#c_ts1598333723