openscad/openscad #3420: Editor still receives key presses even when hidden

ali1234, 2020-08-25:
Version: openscad-nightly snap, version 2020.07.30 

Reproduce:

1. Open a file.
2. Tick view->hide editor.
3. Type some nonsense.
4. Untick view->hide editor.

Result: the nonsense you typed is in the file.

This is a problem because you can mess up your code with stray key presses, without realising it has happened.

rcolyer, 2020-08-25:
This bug happens as far back as 15.03.  In the master branch however it has become worse as tooltips even pop-up while the editor is hidden.  All text entry to the editor needs to be actively disabled upon hiding and re-enabled upon unhiding.

t-paul, 2020-08-27:
I can't reproduce that with locally built OpenSCAD, so I'm wondering if that's a Qt5 bug that was fixed between Qt5.5 (used in Snap package) and Qt5.14 (currently on my local system).

MichaelAtOz, 2020-08-28:
I can reproduce on W7/64 with 2020.07.05 & 2020.08.21, both show Qt 5.13.

t-paul, 2020-08-28:
I found https://bugreports.qt.io/browse/QTBUG-82939 which sounds similar but is still open.