openscad/openscad #3421: Including an empty file silences warnings

ali1234, 2020-08-25:
Code:

```
//include <empty.scad>

//data = [1, 2, 3, 4];
//unit = 10;

function f(arr, _i=0, _acc=[]) =
	_i==len(arr) ? _acc : f(
		arr, _i+1,
		concat(
			[_i==0 ? 0 : _acc[0]+(arr[_i]*unit)], _acc
		)
	);

//x = f(data);
echo(x);
```
`empty.scad` should be an empty file.

This code intentionally has two errors: `data` and `unit` are not defined. The former causes an infinite loop and a warning about `arr` being undefined. The latter causes a second warning about `unit` to be printed in every recursion. The call to `f` is commented for convenience - uncomment it to see the results. The include is commented because its presence affects the result.

When the include is commented, the code will spam two warnings - the limit of 5 warnings fails because there are two different errors. That is a separate problem, and is expected.

However, when the include is uncommented, no warnings at all will appear. Even if unit is defined so that the warning limit still works, then the 5 warnings about `arr` being undefined will not appear.

MichaelAtOz, 2020-08-26:
Use this for testing:

 

function f(arr, _i=0, _acc=[]) =

  _i>100 ? "stopped" :

...

 

So you don't have to kill OpenSCAD.

 

I get some different results. 2019.05.

 

> This code intentionally has two errors: data and unit are not defined. The former causes an
infinite loop and a warning about arr being undefined. The latter causes a second warning about
unit to be printed in every recursion. 

 

Same.

 

> However, when the include is uncommented, no warnings at all will appear. Even if unit is defined
so that the warning limit still works, then the 5 warnings about arr being undefined will not
appear.

Uncommented include, commented unit.

 

...

WARNING: Ignoring unknown variable 'unit', in file inf-loop-errors.scad, line 11.

WARNING: len() parameter could not be converted, in file inf-loop-errors.scad, line 8

WARNING: Ignoring unknown variable 'unit', in file inf-loop-errors.scad, line 11.

WARNING: len() parameter could not be converted, in file inf-loop-errors.scad, line 8

WARNING: Ignoring unknown variable 'unit', in file inf-loop-errors.scad, line 11.

ECHO: "stopped"

 

i.e. 'infinite' loop.

 

Compiling design (CSG Tree generation)...

WARNING: Ignoring unknown variable 'data', in file inf-loop-errors.scad, line 16.

WARNING: len() parameter could not be converted, in file inf-loop-errors.scad, line 8

WARNING: len() parameter could not be converted, in file inf-loop-errors.scad, line 8

WARNING: len() parameter could not be converted, in file inf-loop-errors.scad, line 8

WARNING: len() parameter could not be converted, in file inf-loop-errors.scad, line 8

WARNING: len() parameter could not be converted, in file inf-loop-errors.scad, line 8

ECHO: "stopped"

 

If I remove the "stopped" code, then I get the above, but an infinite loop.

 

Uncomment include, uncomment unit.

 

Same as above, ie 6 errors & stopped, and 6 errors & infinite loop.

 

With 2020.08.21.ci5800 (git df971963), different messages

 

Uncommented include, commented unit.

 

...

WARNING: len() parameter could not be converted, in file inf-loop-errors.scad, line 8 

WARNING: Ignoring unknown variable 'unit', in file inf-loop-errors.scad, line 11. 

WARNING: undefined operation (undefined * undefined) in file inf-loop-errors.scad, line 11 

WARNING: undefined operation (undefined + undefined) in file inf-loop-errors.scad, line 11 

WARNING: len() parameter could not be converted, in file inf-loop-errors.scad, line 8 

WARNING: Ignoring unknown variable 'unit', in file inf-loop-errors.scad, line 11. 

WARNING: undefined operation (undefined * undefined) in file inf-loop-errors.scad, line 11 

WARNING: undefined operation (undefined + undefined) in file inf-loop-errors.scad, line 11 

ECHO: "stopped"

 

Uncomment include, uncomment unit.

 

...

WARNING: len() parameter could not be converted, in file inf-loop-errors.scad, line 8 

WARNING: undefined operation (undefined * number) in file inf-loop-errors.scad, line 11 

WARNING: undefined operation (undefined + undefined) in file inf-loop-errors.scad, line 11 

WARNING: len() parameter could not be converted, in file inf-loop-errors.scad, line 8 

WARNING: undefined operation (undefined * number) in file inf-loop-errors.scad, line 11 

WARNING: undefined operation (undefined + undefined) in file inf-loop-errors.scad, line 11 

ECHO: "stopped"

 

If I remove the "stopped" code, then I get the above, but an infinite loop.

 

> when the include is uncommented, no warnings at all will appear.

 

i.e. I get warnings. 

On W7/64.

 

  _____  

 

 
 
 
 

<https://github.com/notifications/beacon/AAPBVYIJTEW44PPGVXGYQBTSCRD47A5CNFSM4QLGHXKKYY3PNVWWK3TUL5
2HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4KHCBIEQ.gif> 

 

This email has been checked for viruses by AVG.
https://www.avg.com

MichaelAtOz, 2020-08-26:
a. I don't know why all that whitespace got inserted.

 

b. Edit

 

i.e. 'infinite' loop.

 

Uncommented unit          <========== missed this bit.

 

Compiling design (CSG Tree generation)...

 

 

  _____  

 

<https://github.com/notifications/beacon/AAPBVYIJTEW44PPGVXGYQBTSCRD47A5CNFSM4QLGHXKKYY3PNVWWK3TUL5

https://www.avg.com

-
<https://github.com/notifications/beacon/AAPBVYJVL7V5ZIBJRLAOVQLSCRIZTA5CNFSM4QLGHXKKYY3PNVWWK3TUL5
2HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFCGU66I.gif>

rcolyer, 2021-01-04:
I greatly simplified the test case.  It has no dependency on the content of the include file (I switched to a built-in containing only some variable assignments), and it is not specifically silencing warnings, but instead the presence of an include is for some reason delaying all console output until the completion of the preview (which is never for the infinite loop case).

This echos an infinite series of "here":
```
//include <MCAD/constants.scad>
function f() = echo("here") f();
x = f();
```

This halts output and loops forever with no output:
```
include <MCAD/constants.scad>
function f() = echo("here") f();
x = f();
```

This iteratively counts to 200:
```
//include <MCAD/constants.scad>
function f(i=0) = echo(i) i<200 ? f(i+1) : 0;
x = f();
```

This waits until completion of the execution, then instantly dumps all of the output at once.  On a fast system you can see the delay more clearly if you increase the 200 to a larger value.
```
include <MCAD/constants.scad>
function f(i=0) = echo(i) i<200 ? f(i+1) : 0;
x = f();
```

In summary, the mere presence of any include postpones the output being flushed to the console until the completion of a run, which is an unexpected behavior.

thehans, 2021-01-04:
I can't reproduce on Linux Mint 20 running the OpenSCAD nightly, is it possibly a Windows-specific issue?

Running the @rcolyer's "infinite" case in GUI immediately starts printing to the OpenSCAD Console, although delays from Qt make it slow to actually reach recursion limit (it does have to update/repaint the Console GUI 1 million times before that happens after all).  I have found that hiding the console until its done dramatically speed things up though.  You can check in task manager when the openscad process goes back down to idle CPU usage levels.  Takes about 2 minutes on my Ryzen desktop if i keep console hidden from start to end.
```
...
ECHO: "here" 
ECHO: "here" 
ERROR: Recursion detected calling function 'f' in file test_console.scad, line 2 
TRACE: called by 'f' in file test_console.scad, line 3 
Compiling design (CSG Products generation)... 
Geometries in cache: 0 
Geometry cache size in bytes: 0 
CGAL Polyhedrons in cache: 0 
CGAL cache size in bytes: 0 
Compiling design (CSG Products normalization)... 
Normalized tree has 1 elements! 
Compile and preview finished. 
Total rendering time: 0:01:51.694 
```

And running from terminal / command line it also immediately starts printing the flood of echos and finishes in a couple seconds.
`time openscad-nightly test_console.scad -o test.png`
```
...
ECHO: "here"
ECHO: "here"
ERROR: Recursion detected calling function 'f' in file test_console.scad, line 2
TRACE: called by 'f' in file test_console.scad, line 3
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Geometries in cache: 0
Geometry cache size in bytes: 0
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.235

real    0m3.628s
user    0m1.798s
sys     0m1.630s
```

MichaelAtOz, 2021-01-04:
2021.01.03 & Windows 7/64.

I'm not getting the delayed console output with include<> either.

I do get the recursion limit (with hidden console).

 

  _____  

From: Hans Loeblich [mailto:notifications@github.com] 
Sent: Mon, 4 Jan 2021 17:20
To: openscad/openscad
Cc: Michael; Comment
Subject: Re: [openscad/openscad] Including an empty file silences warnings (#3421)

 

 

-- 
This email has been checked for viruses by AVG.
https://www.avg.com

MichaelAtOz, 2021-01-05:
...and recursion limit with visible console took 9m13s, include<> commented-out, 13m01s uncommented
include<>.

That's a big difference for a file with three variables in it.

 

Neither had delayed console output.

 

Ryan, 

Your symptoms are not those of the original post.

What OpenSCAD version & what O/S?

Are you running it from terminal or command-line, if so does any output show there?

Do you get the recursion limit if you leave it running long (hide console & wait for CPU use to
stop) - also see below?

 

Do you have console problems with normal, non-fabricated usage?

 

This gets to the limit quicker, with enough console activity to see it working;

 

include <MCAD/constants.scad>

function f3(i=0) = (i<4000000) 

                      ? (i % 10000 == 0)

                           ? echo(i) f3(i+1)

                           : f3(i+1)

                      : echo("finished");

x = f3();

 

 

 

  _____  

 

https://www.avg.com

-
<https://github.com/notifications/beacon/AAPBVYJPVUFO6JPOZC3Z2ULSYFSMTA5CNFSM4QLGHXKKYY3PNVWWK3TUL5
2HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFTXBAHI.gif>

MichaelAtOz, 2021-01-05:
I don't suppose anyone's come across this issue with blank lines

being inserted in email replies to GitHub?

This is what is in my sent mail:

 

  _____  

 

https://www.avg.com

-
<https://github.com/notifications/beacon/AAPBVYJPVUFO6JPOZC3Z2ULSYFSMTA5CNFSM4QLGHXKKYY3PNVWWK3TUL5

<https://github.com/notifications/beacon/AAPBVYOAJPOG6HOD5OFXZWLSYKEVBA5CNFSM4QLGHXKKYY3PNVWWK3TUL5
2HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFT3NYWQ.gif>

MichaelAtOz, 2021-01-05:
This post is typed into GitHub on the web.
That previous post via email had an image in it, below, I've got something broken...
[image]

ali1234, 2021-01-05:
> Ryan, Your symptoms are not those of the original post.

On the contrary. I'm the original reporter and what Ryan described exactly matches the problem I was reporting. I was using Ubuntu 20.04 and the openscad-nightly snap, so you would have to look at whatever version was pushed to that repository in August.

MichaelAtOz, 2021-01-05:
Doesn't happen for me on Debian or Windows-7 with 2021.01.03.

@ali1234 & @rcolyer 

Are you running it from terminal or command-line, if so does any output show there? 
Do you get the recursion limit if you leave it running long (hide console & wait for CPU use to stop) - also see below?
Do you have console problems with normal, non-fabricated usage? 

This gets to the limit quicker, with enough console activity to see it working; 

include <MCAD/constants.scad> 
function f3(i=0) = (i<4000000) ? (i % 10000 == 0) ? echo(i) f3(i+1) : f3(i+1) : echo("finished"); 
x = f3();

Please attached the output of Help/Library-info.

ali1234, 2021-01-05:
No output is produced on the console either with or without the include.

I don't know what you mean by "recursion limit". This bug has nothing to do with recursion, that is simply an easy way to make a long running script which demonstrates the bug.

I encountered this bug during normal use, when I made a mistake in my code which introduced an infinite loop and noticed that my debug statements did not work.

ali1234, 2021-01-05:
The bug can be reproduced without recursion, eg:

```
//include <MCAD/constants.scad>

for (i = [0 : 500]) {
    echo(i);
}
```

ali1234, 2021-01-05:
Library info

OpenSCAD Version: 2021.01.03
System information: Linux 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 Ubuntu Core 16 8 CPUs 31.30 GB RAM
User Agent: OpenSCAD/2021.01.03 (Linux x86_64; Ubuntu Core 16)
Compiler: GCC "5.4.0 20160609" 64bit
MinGW build: No
Debug build: No
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.14.3, Cartesian, Extended_cartesian, Epeck
OpenCSG version: unknown, 
Qt version: 5.5.1
QScintilla version: 2.9.1
InputDrivers: 
GLib version: 2.48.2
lodepng version: 20180910
libzip version: 1.0.1
fontconfig version: 2.11.94
freetype version: 2.6.1
harfbuzz version: 1.0.1
cairo version: 1.14.6
lib3mf version: 1.8.1
Features: input-driver-dbus, lazy-union
Application Path: /snap/openscad-nightly/2280/usr/bin
Documents Path: /home/al/snap/openscad-nightly/2280/.local/share
User Documents Path: /home/al/snap/openscad-nightly/2280
Resource Path: /snap/openscad-nightly/2280/usr/share/openscad-nightly
User Library Path: /home/al/snap/openscad-nightly/2280/.local/share/OpenSCAD/libraries
User Config Path: /home/al/snap/openscad-nightly/2280/.config/OpenSCAD
Backup Path: /home/al/snap/openscad-nightly/2280/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/al/snap/openscad-nightly/2280/.local/share/OpenSCAD/libraries
/snap/openscad-nightly/2280/usr/share/openscad-nightly/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
/usr/share/fonts
/usr/local/share/fonts
/home/al/snap/openscad-nightly/2280/.local/share/fonts
/home/al/snap/openscad-nightly/2280/.fonts
/home/al/.local/share/fonts
/snap/openscad-nightly/2280/usr/share/fonts
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
/home/al/snap/openscad-nightly/2280/Library/Fonts
/usr/share/fonts/X11
/usr/share/fonts/cMap
/usr/share/fonts/cmap
/usr/share/fonts/eot
/usr/share/fonts/opentype
/usr/share/fonts/svg
/usr/share/fonts/truetype
/usr/share/fonts/type1
/usr/share/fonts/woff
/snap/openscad-nightly/2280/usr/share/fonts/truetype
/usr/share/fonts/X11/Type1
/usr/share/fonts/X11/encodings
/usr/share/fonts/X11/misc
/usr/share/fonts/X11/util
/usr/share/fonts/eot/font-awesome
/usr/share/fonts/opentype/cantarell
/usr/share/fonts/opentype/font-awesome
/usr/share/fonts/opentype/fonts-hosny-amiri
/usr/share/fonts/opentype/malayalam
/usr/share/fonts/opentype/noto
/usr/share/fonts/opentype/urw-base35
/usr/share/fonts/svg/font-awesome
[... 445 lines cut ...]
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vdpau_interop
GL_NV_vdpau_interop2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum

rcolyer, 2021-01-05:
> Doesn't happen for me on Debian or Windows-7 with 2021.01.03.
>
> @ali1234 & @rcolyer
>
> Are you running it from terminal or command-line, if so does any output show there?
> Do you get the recursion limit [...]

I'm launching from terminal, but this does not happen for terminal-only runs.  It is an issue related to the behavior of the gui and the gui console.  No additional terminal outputs occur while doing the gui runs which produce any of these behaviors.

Hiding the console while running is not an option on the version that does not output, as it freezes the gui.  On reflection this might be important to the nature of the problem.  Perhaps something about the presence of the include causes something to run on the gui thread which shouldn't, or holds the gui thread in reserve, whereas without the include the gui thread is left open for events?

@thehans and I confirmed that this bug does or does not appear on different systems running in almost identical conditions.  It's not currently clear what the operative difference is.  The bug appears fairly old though because I'm seeing it on everything from 2019.01-RC3 (the oldest I have functioning) through the 2021.01.03.ai6611 nightly.  It's just that on the older ones, the recursion limit was detected quickly, causing the paused output to all appear once it hit the end.  But you can clearly distinguish that the output is frozen until completion when the include is present, but concurrent with execution without the include.  On the new ones the recursion limit happens to not be detected in this example, even with console hidden for the no-include one which has an accessible gui, and I think it's the lack of recursion limit being hit quickly (perhaps because the limits were increased intentionally recently) which is what made this bug obvious enough to result in a report.  In retrospect I probably observed this many times, but simply didn't recognize it was a delay triggered by the presence of includes.  The loss of multithreaded interface capability during execution with the include is probably the most serious aspect of this bug.

So as I said, it happens on a wide variety of runs, but here's the 2021.01.03.ai6611 library info relevant section:
```
OpenSCAD Version: 2021.01.03.ai6611 (git a980a3a)
System information: Linux 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 Ubuntu 20.04.1 LTS 8 CPUs 31.25 GB RAM
User Agent: OpenSCAD/2021.01.03.ai6611 (git a980a3a) (Linux x86_64; Ubuntu 20.04.1 LTS)
Compiler: GCC "5.4.0 20160609" 64bit
MinGW build: No
Debug build: No
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: unknown, 
Qt version: 5.14.2
QScintilla version: 2.9.1
InputDrivers: 
GLib version: 2.48.2
lodepng version: 20180910
libzip version: 1.0.1
fontconfig version: 2.11.94 (runtime: 2.13.1)
freetype version: 2.6.1 (runtime: 2.10.1)
harfbuzz version: 1.0.1 (runtime: 2.6.4)
cairo version: 1.14.6
lib3mf version: 1.8.1
Features: input-driver-dbus, lazy-union
```

nophead, 2021-01-05:
It doesn't happen for me as I always have includes in my project and the console outputs echos as they occur, and always have done. Looking my library info I have a different version of everything.

OpenSCAD Version: 2021.01.01.ci6570 (git 52e26f2fb)
System information: Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833 4 CPUs 7.90 GB RAM
User Agent: OpenSCAD/2021.01.01.ci6570 (git 52e26f2fb) (Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.3.7
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: unknown, 
Qt version: 5.13.0
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.2
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.4.0
cairo version: 1.16.0
lib3mf version: 1.8.0
Features: input-driver-dbus, lazy-union

MichaelAtOz, 2021-01-05:
As it is not easily reproduced, this will have to be a bug hunt, looking into all the undergrowth...

> Perhaps something about the presence of the include causes something to run on the gui thread which shouldn't, or holds the gui thread in reserve, whereas without the include the gui thread is left open for events?

The parse & compile (& CSG preview) happen in the GUI thread (there is a request to split it off).

The logs start with
```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
ECHO:0
...
```
Do either of the first two lines get logged before the freeze?

> 
> @thehans and I confirmed that this bug does or does not appear on different systems running in almost identical conditions. It's not currently clear what the operative difference is. I think it's the lack of recursion limit being hit quickly (perhaps because the limits were increased intentionally recently) 

IIRC yes it was increased a while ago.

@rcolyer  
The different systems you mention, are they all Ubuntu?
Can you get the Help/Library-info of one where it works OK?
Could you include the additional (to what you posted above) Library-info `Application Path:` ... `OpenSCAD library path:`, ie on your problematic system.
Are you also using snap?

>
> In retrospect I probably observed this many times, but simply didn't recognize it was a delay triggered by the presence of includes. The loss of multithreaded interface capability during execution with the include is probably the most serious aspect of this bug.

This is presumably related to file I/O.
Is the include file, or any of the paths, located in any non-basic-conventional way, network drive etc?
If the include is in the same directory as the main .scad file, does it also freeze?

MichaelAtOz, 2021-01-06:
I just installed the snap nightly, it works normally in terms of the console (but some problem with PulseAudioServices).

My snap Library-info:
```
OpenSCAD Version: 2021.01.05
System information: Linux 4.9.0-12-amd64 #1 SMP Debian 4.9.210-1 (2020-01-20) x86_64 Ubuntu Core 16 8 CPUs 31.32 GB RAM
User Agent: OpenSCAD/2021.01.05 (Linux x86_64; Ubuntu Core 16)
Compiler: GCC "5.4.0 20160609" 64bit
MinGW build: No
Debug build: No
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.14.3, Cartesian, Extended_cartesian, Epeck
OpenCSG version: unknown, 
Qt version: 5.5.1
QScintilla version: 2.9.1
InputDrivers: 
GLib version: 2.48.2
lodepng version: 20180910
libzip version: 1.0.1
fontconfig version: 2.11.94
freetype version: 2.6.1
harfbuzz version: 1.0.1
cairo version: 1.14.6
lib3mf version: 1.8.1
Features: input-driver-dbus, lazy-union
Application Path: /snap/openscad-nightly/2288/usr/bin
Documents Path: /home/mebd/snap/openscad-nightly/2288/.local/share
User Documents Path: /home/mebd/snap/openscad-nightly/2288
Resource Path: /snap/openscad-nightly/2288/usr/share/openscad-nightly
User Library Path: /home/mebd/snap/openscad-nightly/2288/.local/share/OpenSCAD/libraries
User Config Path: /home/mebd/snap/openscad-nightly/2288/.config/OpenSCAD
Backup Path: /home/mebd/snap/openscad-nightly/2288/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/mebd/snap/openscad-nightly/2288/.local/share/OpenSCAD/libraries
/snap/openscad-nightly/2288/usr/share/openscad-nightly/libraries
```
Pretty much matches @ali1234's except Linux version. 

----
Try running `openscad-nightly --debug=all`
For me, with snap, gives:
```
Loaded design '/home/mebd/Documents/OpenSCAD/issue #3421.scad'. 
 
Parsing design (AST generation)... 
FileModule: registerInclude(): (0x40f33c0) 5, 1 - 5, 30 (MCAD/constants.scad) -> /snap/openscad-nightly/2288/usr/share/openscad-nightly/libraries/MCAD/constants.scad 
Compiling design (CSG Tree generation)... 
ECHO: 0 
...
```

MichaelAtOz, 2021-01-07:
I can now reproduce this freeze.
2021.01-RC5 on Windows.
This was via the Install.exe rather than ZIP, so I'll try the ZIP just in case.

Library-info
```
OpenSCAD Version: 2021.01-RC5
System information: Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833 4 CPUs 6.06 GB RAM
User Agent: OpenSCAD/2021.01-RC5 (Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.3.7
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.13.0
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.2
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.4.0
cairo version: 1.16.0
lib3mf version: 1.8.0
Application Path: C:/Program Files/OpenSCAD-RC5
Documents Path: C:\Users\MeB\Documents
User Documents Path: C:\Users\MeB\Documents
Resource Path: C:/Program Files/OpenSCAD-RC5
User Library Path: C:/Users/MeB/Documents/OpenSCAD/libraries
User Config Path: C:\Users\MeB\AppData\Local/OpenSCAD
Backup Path: C:/Users/MeB/Documents/OpenSCAD/backups
OPENSCADPATH: C:\Users\MeB\Documents\3D-REPRAP\OpenSCAD\OpenSCAD Libraries;C:\Users;C:\test;C:\test2;C:\test-uc-?
OpenSCAD library path:
C:/Users/MeB/Documents/3D-REPRAP/OpenSCAD/OpenSCAD Libraries
C:/Users
C:/test
C:/test2
C:/test-uc-?
C:/Users/MeB/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD-RC5\libraries
```
No significant difference to my ZIP 2021.01.03-nightly.

MichaelAtOz, 2021-01-07:
ZIP version also reproduces the freeze. It's Library-info is identical.

The freeze is not dependent on `include<>`, ie commented-out.
```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
```
Are logged to console, then freeze, then the rest logged quickly.
So the freeze happens during compile.

MichaelAtOz, 2021-01-07:
Also reproducible with 2021.01.06 snapshot.
Not with 2021.01.03.

----
Differences in Library-info

Freeze `OpenSCAD Version: 2021.01.06.ci6702 (git fb5690a63)`
```
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Application Path: C:/Program Files/openscad-2021.01.06.ci6702
Resource Path: C:/Program Files/openscad-2021.01.06.ci6702
C:/Program Files/openscad-2021.01.06.ci6702\libraries
```
----

No freeze `OpenSCAD Version: 2021.01.03.ci6609 (git a980a3a18)`
```
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: unknown, 
Application Path: C:/Program Files/openscad-2021.01.03.ci6609
Resource Path: C:/Program Files/openscad-2021.01.03.ci6609
C:/Program Files/openscad-2021.01.03.ci6609\libraries
```

----
That is not a difference which seems likely to cause a freeze.

Can someone bisect those two.

There is a perceived difference, .06 processes the recursion faster,
I had to go to 1000 to notice it freeze;
```
//include <MCAD/constants.scad>
function f2(i=0) = echo(i) i<1000 ? f2(i+1) : 0;
x = f2();
```
Whereas non-freeze I see the ECHO's logging.
That could be a red-herring, as the freeze does not need
to repaint the console each time.

MichaelAtOz, 2021-01-07:
Also freezes on Debian 2021.01.06, previously .03 did not freeze.

PR #3602 (2021-01-16): Buffer messages to GUI Console for better performance.
Also should fix #3421.
commit 45be0f56 (2021-01-16): Buffer messages to GUI Console and append multiple lines at a time for ~7x better performance.
Also should fix #3421.
commit 662db3ad (2021-01-16): Merge pull request #3602 from thehans/console_speedup

Buffer messages to GUI Console for better performance (fixes #3421).
closed 2021-01-16 by t-paul