openscad/openscad #3430: SVG import does not close open fill paths.

rickan54, 2020-09-06:
The spec says it should.

Instead, it strokes the open path with some default line thickness.

I don't know how to provide an example since I'm not allowed to upload an *.svg file. (Why?) I will provide it if given instructions how.

Anyway, my example is displayed correctly in Firefox and if I edit the file to close the fill paths it works in OpenSCAD too, but, unfortunately, the utility I use to convert from PostScript to SVG does not close the fill paths. Why should it? They are closed in the PS file the SVG is created from.

t-paul, 2020-09-06:
Just rename to *.txt, that should allow to upload the file anyway.

t-paul, 2020-09-06:
Also it would be useful to link to the part of the spec defining the expected behavior.

rickan54, 2020-09-06:
Sorry about not linking to the spec.

https://www.w3.org/TR/SVG2/painting.html#SpecifyingFillPaint

Second paragraph.

Example SVG file:

[junk.txt](https://github.com/openscad/openscad/files/5180294/junk.txt)

Change the extension to svg to get Firefox to display it properly.

Older version of spec:
https://www.w3.org/TR/SVG11/painting.html#FillProperties

t-paul, 2020-09-06:
I'm not sure there's a good way to handle fill to make the example file work. The current behavior is is shown in https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/SVG_Import

rickan54, 2020-09-06:
H'h. I'd sure have called it a bug:

 - It's non-conforming,

 - The case is handled in OpenSCAD but in a rather astonishing manner just considering the choices and not the spec.

 - It seems likely that SVG generators will create open fill paths because they can and it reduces file size, i.e. I don't have any data to prove it, but it sure seems to cripple SVG in OpenSCAD. (But maybe no one ever tried to use open filled paths before.)

 - based on my experience the OpenSCAD behavior in this case seems likely to mystify anyone who encounters it potentially causing wasted effort to correct it.

t-paul, 2020-09-06:
OpenSCAD will never be able to import just any SVG, specifically those that work based on visual effects. The import needs to extract geometry information from a format intended for rendering images. What we can try is finding some common styles and treat those special cases.

rickan54, 2020-09-06:
As I said in my original post, explicitly closing the path in the SVG file makes the example work in OpenSCAD. 

If the article you link to stated that filled paths are not supported, then I'd call this an enhancement, but I've shown that filled paths ARE supported but not correctly, It is trivial to close an open path and it's the only thing that makes sense for open fill paths. The way OpenSCAD handles open stroked and not filled paths makes sense and is the best option (IMO) in that case.

rickan54, 2020-09-06:
I'm not an SVG expert so I don't know what all is possible in an SVG file but it sure seems like handling the path geometry  correctly (including fill which OpenSCAD almost does) would be a good thing. Disregarding color, patterns, etc. shouldn't be a big deal IMO.

rickan54, 2020-09-06:
Another thought I have on this issue is that closing an open path is trivial whereas filling a closed path is a big deal of a feature. So the investment of effort in having filled paths which has already been made  is only (what seems to me as) a trivial fix away from being complete and more useful.

rickan54, 2020-09-06:
OpenSCAD behavior with filled paths matches the link you provide above which, in effect, ignores filled paths. So I guess this really IS an enhancement request.

Still, I think it's a real shame that the feature was designed that way when closing a path meant to be filled so it can be a legal OpenSCAD path and thus filled, as would any closed path, is so easy and logical. But maybe not trivial if it involves parsing and taking into account the filled attribute of the path. Just almost trivial.

So maybe it's also a design flaw (bug).

t-paul, 2020-09-07:
There is no such thing as an open path or fill in OpenSCAD. All objects 2D or 3D are required to be closed/solid to support CSG operations. At some point there might be extensions for additional support of lines or open polygons to construct closed/solid objects, but OpenSCAD is still a "Solid Modeller" at the core.

One option might be to close the path if it has no stroke-width set but any fill (!= "none") is defined.

rickan54, 2020-09-07:
What OpenSCAD does is fill closed paths. If the path is not closed it's converted to a closed path corresponding to a line with some width.

According to the SVG spec any filled path should be closed before filling it.

It seems to me the only ambiguity as far as OpenSCAD is concerned is what to do with paths which are BOTH stroked and filled, presumably with different colors or other attribute and filled first and then stroked (to preserve the line width).

The spec seems to show that fill defaults to black and stroke defaults to none kind of indicating that fill takes precedence. I interpret this to mean that if no fill or stroke are specified, then the path is to be filled, and if no fill-* attributes are specified the defaults are to be used. If no stroke is specified then the path will not be stroked and the stroke-* attributes will be ignored.  

If a path is to be BOTH stroked and filled, then the stroke must be specified to some !="none". and in this case maybe stroke behavior should be exhibited by OpenSCAD, a wide line with a width of 1 by default, rather than fill.

So it seems to me that the only path attributes related to stroking and filling that OpenSCAD needs are fill, stroke and stroke-width (and maybe fill-rule) and the cased are:

no fill and no stroke               - close if necessary and fill because of the default values
fill (bare) only                        - close if necessary and fill
fill  != none only                     - close if necessary and fill
stroke (bare) only                   - close if necessary and fill since stroke defaults to none
stroke != none only                - fill the path as a line stroke-width wide or 1 wide if ho stroke-width 
fill (bare) and stroke != none  - fill the path as a line stroke-width wide or 1 wide if ho stroke-width
fill and stroke both != none    - fill the path as a line stroke-width wide or 1 wide if ho stroke-width

OpenSCAD should prefer stroke if it's specified != none since it defaults to none.

I think the fill-rule is a different can of worms since it should apply to OpenSCAD in any case of a closed path (not just paths the spec says OpenSCAD should close). I was unable with a quick search to find what fill/winding rule OpenSCAD uses or whether it's an option. (I use PostScript for 2D drawing, and have done extensively and for ages now, and I don't remember ever having occasion to specify a fill/winding rule. But I'm clearly an odd case.)

(BTW, PostScript closes open paths (and sub-paths) prior to filling (but I don't remember ever taking advantage of this), and it uses the non-zero winding rule by default claiming it's more versatile, but it does have operators which use the even-odd rule you can use instead of the "normal" ones. It seems SVG takes after PostScript rather heavily)