openscad/openscad #3434: Restore multi-window operation as a first-class citizen

jordanbrown0, 2020-09-20:
Introduction of tab support relegated multi-window operation to a second-class citizen.  Before that, you could open a new blank window with File/New (at the top of the menu), or with ^N, or with the new-document button, and an existing file in a new window with File/Open (second in the menu), or with ^O, or with the open-document button.  After, new windows are available only with File/New Window and File/Open in New Window, halfway down the menu.  There's no keyboard shortcut and no buttons.  File/Recent Files and File/Examples because new-tab only.

I'd like to see a preference that restores the old behavior, with the addition of the new behavior allowing one to open several files at once.  The default would be the new tab-based behavior.

Proposed behaviors:

New preference "Use tabs", at the top of Advanced / User Interface, defaulting to checked.

When checked, behavior is unchanged from the current tab-based behavior.

When cleared:

File/New (and ^N and the New button) open a new window.

File/Open (and ^O and the Open button), File/Recent Files, and File/Examples open the files in the current window if it is new and unchanged, and otherwise in new windows.  (Note that this means that, as for tabs, opening multiple files may open the first in the current new-and-unchanged window; the second and later will open in new windows.)

File/New Window and File/Open in New Window are hidden.

jordanbrown0, 2020-09-20:
I have a prototype that seems to work in jordanbrown0/openscad, but need to figure out how to navigate both github's mechanisms and OpenSCAD's approval process.

t-paul, 2020-09-20:
Have you looked at #3338? I think a more general solution giving configuration options to the user is better than strange modes that switch behavior of the UI.

jordanbrown0, 2020-09-20:
Hmm.  No, I hadn't looked at #3338. Keyboard shortcuts would help, but aren't the whole picture - there's also the buttons and, to a lesser extent, positioning in the menu and behavior of the "secondary" open mechanisms like Recent Files and Examples.  It seems like a shortcut-based answer will lead to 80% solutions.  Those are better than none, but we're talking about the overall model for the UI and consistently switching between one model and the other will likely require multiple steps that you'd want to replicate for each installation that you use, and that new users would need to rediscover.  (One could imagine the shortcut mechanism having "canned" sets of shortcuts that could be selected en mass, but that's almost like having a mode.)

Also, making it clear that there's the two distinct UI models may make it less likely that tab-only behaviors will crop up in the future.  (Try to get Thunderbird to put Lightning or Options in a separate window.  Can't do it.)

I took a rather "top-down" approach, having each of the commands choose between the two models.  I do wonder whether it would be better to have a single "create new workspace" function that would either create a new tab or a new window depending on the mode.  That might more readily extend to future analogous cases.

t-paul, 2020-09-20:
I fear adding multiple UI modes will actually cause a big maintenance burden. That will mean GUI changes need to be tested in all modes multiplied by all operating systems. Right now we are already blocked in some areas as there's not really anyone available for testing things out on MacOS.

jordanbrown0, 2020-09-21:
That's fair, though I would hope that the mode could simply rearrange behaviors that are present in all cases and so be very low risk for different behavior on different platforms.  I'll think about it some more.

rcolyer, 2020-10-15:
I personally find the old multi-window modality a bit of a misguided feature.  It presented two "apparently" fully functioning interfaces to a program, except they're throttled to only be able to process one of them at a time.  You get exactly the same window appearance by running two copies of OpenSCAD, but then you can actually use them in parallel, previewing one while the other is rendering.  If it helped people to have some internal method for opening another copy of OpenSCAD for proper multitasking, that would at least sound more reasonable to me than going back to the mutually blocking windows design.

Tabs are a philosophically different thing, and provide a sensible interface that can be used for example to edit multi-file designs with includes, where parallel operation is not an issue.

nophead, 2020-10-15:
Another big difference is the multi-window view shares the same data. I.e. the module and geometry caches.  So if the views are separate aspects of one big project is makes some sense to use that mode rather than open separate instances.

rcolyer, 2020-10-18:
That cache difference should essentially disappear with #3316 from the GSoC project.

jordanbrown0, 2025-03-09:
I admit defeat.

closed 2025-03-09 by jordanbrown0

kintel, 2025-03-09:
I kind of want better multi-window support back as well, as I generally dislike the tab style. I guess this is mostly since I got used to macOS style MDI apps. But yeah, someone needs to care about the extra complexity (or make it clean enough).