openscad/openscad #3441: OpenSCAD crashes while rendering division of STL with cylinder

Knochi, 2020-09-25:
Hi,

I'm trying to modify (remix) a model for metric nuts and bolts.
https://www.thingiverse.com/thing:4500713

When I try to subtract a cylinder from the imported mesh OpenSCAD crashes while rendering, preview works fine.
Rendering of imported mesh without substraction works as well.

Version Information:
OpenSCAD Version: 2020.06.14.ci5408 (git be16bd85)
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 4 CPUs 7.85 GB RAM

[DiffFromSTLCrash.zip](https://github.com/openscad/openscad/files/5282844/DiffFromSTLCrash.zip)

MichaelAtOz, 2020-09-26:
The STL is bad.
Meshlab:
[image]
It's a mix of holes, degenerate faces and self-intersections.

Knochi, 2020-09-26:
Ok but then I would expect an error message rather than a crash. I will try to fix up the STL using meshmixer.

MichaelAtOz, 2020-09-26:
Possibly #2847.

MichaelAtOz, 2020-09-26:
> And why does the unmodified mesh render fine?

If there is only one object CGAL does not get called.

militaryCoder, 2021-02-12:
@Knochi, havve you tried to remesh this item and perform subtraction again?

Knochi, 2021-02-12:
No.. I gave up on the idea of remixing this item. The point about the issue was robustness, so even if something is wrong with the operation the application shouldn’t crash.

Cu
David

ChrisCoxArt, 2022-01-18:
The example given no longer crashes, but gives "ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron."

thehans, 2022-01-18:
Fixed via #3526.

closed 2022-01-18 by thehans