openscad/openscad #3442: No tabs to select between multiple open files when the editor is hidden

nophead, 2020-09-26:
If you open a second file with editor hidden it looks like it replaces the first file but if you unhide the editor you can see both files and switch between them with the tabs.

I think the tabs should be shown when multiple files are open even if the editor is hidden.

OpenSCAD 2020.09.25.ci5934

t-paul, 2020-09-26:
I'm not convinced it should work that way. Maybe there are other options to handle that case, but showing tabs without the editor looks wrong to me.

nophead, 2020-09-26:
Looks wrong visually or logically?

The tab bar extends all the way across the main window, so isn't visually coupled to the editor. Selecting a tab changes the main window title and F5 previews it. Without the tab bar the other files are invisible and inaccessible unless you reload them, but they are still loaded as a document with an invisible editor.

t-paul, 2020-09-26:
Yes, I mean more logically, or maybe even more specific workflow wise. Hiding the editor and then still using it partially likely opens a can of worms. So I wonder if the option should be to go more restrictive with the editor hidden. A similar discussion happened for the new error window too. Clicking on an error is supposed to open the file if it's not already open. With the editor hidden we plan to disable that opening of files instead of unhiding the editor and showing it.

nophead, 2020-09-26:
You could disable file open or force it to open in a new window but that is less functional and doesn't cover the case of hiding the editor with more than one file open.

I use an external editor, so only have the editor window open to view examples on the forum or to locate tricky syntax errors. I was quite happy without the tabbed view when file open opened a new window and everything worked and the model was shared. I can still do that with "open in new window". I do have two large monitors though and can see that with one monitor on a laptop it would be nice to have one window with tabs.

The tabbed view has always seemed a bit broken to me as it should maintain a preview window for each tab. In that case the tab bar would make even more sense with the editor hidden.

nophead, 2020-09-26:
Another thing that doesn't work properly with tabs is the customiser. If you switch to another tab and do F5, when you switch back the customiser loses and setting you made and goes back to the defaults.
Tabs only make sense if all the views hande multiple documents, not just the editor.

nophead, 2020-09-26:
The more I think about it I think the tabbed view should behave just the same as multiple windows except they share the same window and all the views are multiplexed with the tabs. You should even be able to pull off a tab and get a separate window with all the views in it, or dock a separate window to get a new tab.

jordanbrown0, 2020-09-27:
When thinking about these behaviors, don't forget the "undocked into separate windows" mode.

I'm an "all separate windows" person, at least on my desktop system.  Editor, console, and view are separate, and I always use separate sets of each for each file that I have open.  (Perhaps I should mention that my desktop has two 32" monitors.)

jmckeown2, 2022-06-19:
Because of my work, I prefer VSCode as my editor of choice, so I use OpenSCAD exclusively with editor hidden. For my workflow, I usually work in a split fullscreen mode with VSCode left and OpenSCAD right.
 
It would seem most intuitive that the Tabs remain visible whether or not Editor is hidden AND that on switching tabs the the preview is updated to the selected tab. (as happens with Customizer) -- I believe this is what @nophead was saying?

This is exactly how it worked in 2021.01 on Mac in "Fullscreen." but I suspect it was considered a bug because it ONLY worked this way in full screen and only when "Open in new window" was used.

I generally find that multi-window workflows are where I go for "research/learning/WTF mode" but I prefer more focus/less distraction when doing "in-the-zone" coding; point being, I appreciate the "all separate windows" crowd. So I would generally prefer if windows & tabs could be combined and ripped apart at will as my "head space" changes -- much like tabs vs windows works for browsers