openscad/openscad #3444: Preview flips between two different view depending on viewport position/angle

swthemathwiz, 2020-09-28:
I have this small example of flipping between views in preview mode. I'm trying to split a volume in two by moving it's Z axis to the origin and taking the top (i.e. positive Z) part. This comes from more complex software for generating a case with a separate lid and base (thus the module names).

In preview mode, the part of the box display switches between the top (which the code is supposed to show) and the "erased" bottom (which should not be shown) when changing view port position.

The problem does NOT occur if render is used (in place where commented out in source). Render always works as expected.
The problem does NOT occur if A_BIG_NUMBER is lowered to 100 from 1000, indicating there could be some numeric overflow problem (or a limitation I'm not aware of).

The software used is generic Linux Fedora 32 Workstation with an older NVidia card with up to date packages and drivers.

------------------- Software  Info -----------------------

OpenSCAD Version: 2019.05
System information: Linux 5.8.11-200.fc32.x86_64 #1 SMP Wed Sep 23 13:51:28 UTC 2020 x86_64 Fedora 32 (Workstation Edition) 8 CPUs 15.58 GB RAM
User Agent: OpenSCAD/2019.05 (Linux x86_64; Fedora 32 (Workstation Edition))
Compiler: GCC "10.1.1 20200507 (Red Hat 10.1.1-1)" 64bit
MinGW build: No
Debug build: No
Boost version: 1_69
Eigen version: 3.3.7
CGAL version, kernels: 5.0.2, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
GLEW version: 2.1.0
OpenGL Version: 4.6.0 NVIDIA 450.66
GL Renderer: GeForce GTX 750 Ti/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8880), depth(24), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8880), depth(24), stencil(0)

----------------- EXAMPLE -----------------

A_BIG_NUMBER = 1000;
//A_BIG_NUMBER = 100;

module all_positive_z(size=A_BIG_NUMBER) {
  intersection() {
    translate( [0,0,+size/2] ) cube( size, center=true );
    children();
  }
} // end all_positive_z

module top_of(split_z) {
  /*render( convexity=10 )*/ all_positive_z() translate([0,0,-split_z]) children();
} // end top_of

module base_layout_2d() {
  square( 10 );
} // end base_layout_2d

module side_layout_2d(thickness) {
  difference() {
    base_layout_2d();
    offset( r=-thickness ) base_layout_2d();
  }
} // end side_layout_2d

module sides(height,thickness) {
  linear_extrude( height=height, convexity=10 )
    side_layout_2d(thickness);
} // end sides

top_of(6) sides(10,2);

nophead, 2020-09-28:
The limitation you are not aware of is that all the intermediate objects need to fit inside the clipping volume for preview to work properly. If you zoom out far enough to view the huge cube it works but obviously looks tiny. If you use the orthogonal view it also works.

Use render() if if you need to use the perspective view with large objects that get clipped.

swthemathwiz, 2020-09-28:
Got it, thanks.

closed 2020-09-28 by swthemathwiz