openscad/openscad #3447: Feature suggestion: Fillet and chamfer

logiclrd, 2020-09-30:
I've done some searching of historical records regarding OpenSCAD and Fillet and Chamfer type operations, and have basically found only 3 things being said:

- Fillet and Chamfer are really complex and hard to implement transformations.

- Your best bet is to try to design from the ground up so that the geometry you need is baked in from the beginning.

- If you really need a computed chamfer, use a Minkowski sum of your shape with a sphere the size of the chamfer you want.

The last option I basically discount, because if you have smooth geometry with millions of faces, the Minkowski sum can take literally days to compute -- completely impossible to iterate with.

I was thinking about ways this might be implemented, even contemplating the lofty goal of perhaps trying to contribute it myself, though I suspect it is beyond my expertise. But, I did have some thoughts, and these thoughts led to something that _feels like_ a possible plan for how it might be implemented.

So, to address the first point, yes, these operations are complicated transformations, but it seems to me that there is already a robust open source implementation of them within Blender -- and Blender is licensed under GPLv2 the same as OpenSCAD! So, that seems like a potential win there. If a transformation were created between the "bag of polygons" that OpenSCAD seems to work with internally and the "linked list of loops" that I'm under the impression, from a brief glance at its documentation, that Blender works with, then it could be used to take any OpenSCAD geometry, pass it off to BMesh, and then the result would be easy enough to bring back into the OpenSCAD world.

I was thinking about what would be needed in order to do a Fillet or Chamfer like operation as an operator in OpenSCAD's computation of geometry, and it seems to me that, as complicated as the actual operation is, given that it is actually already implemented, the hardest remaining problem is reliably selecting the series of edges that make up the edge on which the operation needs to be performed. Since OpenSCAD's geometry isn't generated with that type of structure to begin with, and since it needs to select these things without inviting the user to simply click on the line they want. So, that got me thinking about ways to _programmatically_ identify an edge (series of edge segments) to which the operation should apply, and the best I could come up with, which seems eminently implementable, is:

1. Define a vector that determines a "search direction".

2. Sort all the points in the mesh in the direction of this vector -- basically, rotate them so that the vector is [1, 0, 0] and then sort by X.

3. Go through the points in this order, and for each one, follow all edge segments that increase in X, again sorting by X, so that the order in which they are enumerated is well-defined across repeated renderings, possibly with some transformations (changes of scale, etc.).

4. Search for edge segments that have an angle between their faces falling within a certain range, and when one is found, follow it out as far as it will go in either direction, with connected edge segments that have an angle not too deviant from the edge they're connected to, enumerating all the connected edge segments that have the same angle between their faces.

I'm referring to each pair of connected points as an "edge segment", so that the compound structure of a series of them linked end-to-end can be called an "edge".

5. Allow the user to specify that the desired edge is the _n_ th edge meeting these criteria.

Given these steps, the parameters to a Fillet or Chamfer operator would fall into two categories:

* Parameters that are used to identify the edge to be operated on.
* Parameters that determine what exact Fillet/Chamfer/Bevel type operation should be done.

Specifically, the first set of parameters would be:

* Search vector.
* Range of acceptable angles between faces.
* Maximum angle between edge segments to consider it part of the same edge.

This could potentially also be augmented with some parameters that place other limits on the edge length, such as the maximum count of segments or the maximum overall length.

My speculation is that these parameters would allow for a reasonable degree of repeatability across the design iteration of an OpenSCAD program.

If we call this step "Identify", then the overall plan for a Chamfer/Fillet/Bevel type operation would become:

1. Identify the object edge.
2. Transform to BMesh format, creating strips of edge segments starting with the output of Identify and then walking along the mesh in either direction.
3. Ask BMesh to alter the geometry.
4. Transform the resulting Blender-style mesh back to the format OpenSCAD works with natively.

Just looking at the BMesh Python API (I suspect OpenSCAD would probably want to directly call BMesh, which is written in C), I see that it has an operator that converts a `mesh` to a `bmesh` -- I don't what relation this has to step 2, though my gut feeling is that doing step 2 directly would perhaps be necessary in order to guarantee that the BMesh representation actually defines the edge from step 1 as a BMesh loop. Maybe I'm misunderstanding the importance of BMesh loops, and the operator uses some completely different means of identifying the operator's object, in which case the exact way that the definition of the mesh is transformed would be less important.

No step of this is trivial, but it seems to me that if the assertion that Fillet and Chamfer are really hard to implement correctly is true, then this presents a possible plan where they might be possible to leverage from OpenSCAD by plugging conceptually simpler units together.

Thoughts?

t-paul, 2020-09-30:
My first though is that we should decouple the discussion about how to represent the feature in the language and the actual implementation. There's different ideas and suggestions of what backend to use (stay with CGAL but move to corefinement, use carve / opencascade / bmes).
Also note that querying the geometry would cause the mesh to be instantiated at this point which a) breaks preview and b) removes options of keeping the geometry tree, e.g. for STEP export.

That said it's obviously a feature that's asked for a lot and would be a good step forward for OpenSCAD. So even if you decide not do work on the actual implementation, driving this topic via the discussion would be extremely helpful. With the very limited amount of developer time we have available, it's going to take a while so just keeping the on-topic discussion alive and detailing what needs to be done will improve chances of an implementation a lot.

nophead, 2020-09-30:
If 3D minkowski could be made to run quickly I don't think you would need
any language changes to make fillets.

logiclrd, 2020-09-30:
But can it? :-) I'm under the impression that the Minkowski sum is inherently expensive to calculate -- at a very basic level, it involves considering an order of points that is the product of the input geometries.

nophead, 2020-09-30:
It is just a sum of vertices and a hull of the resultant points for any convex shape. And hull is relatively fast.

What takes the time is CGAL decomposing a concave shape into convex sections and then unioning the results together. CGAL takes an age to do anything. The number of vertices is somewhat proportional to the product of the two shapes, I think, so soon gets moderately large, so CGAL gets very slow.

t-paul, 2020-09-30:
Fast minkowski would enable some nice features that are currently not effectively usable. But it's still difficult to work with. If there would be simpler options for at least a good number of use cases, that would be helpful (like an offset extrude with a height-dependent function as offset value).

logiclrd, 2020-09-30:
Is it ever conceivable that Minkowski could be fast if, say, I want to produce rounded edges on something like this?:

[image]

The render output says it has about 340,000 facets joined by 530,000 edges. If I wanted rounded edges, then I'd Minkoski that against a `sphere($fn = 32)` or somesuch, which itself has about 1,000 facets. That means that the input to the union functions is conceptually on the order of 340 billion facets, right?

To make it fast, I suppose I could decompose that by chopping it up, so that I only feed parts that I know have sharp edges into the Minkowski function, but then the chopped edges also get softened, and I need to independently figure out how to inflate the smooth parts, and then chop off the rounded edges I don't want, and then merge it back together without seams -- it feels basically intractable.

Maybe I'm missing something?

logiclrd, 2020-09-30:
Also, what if I wanted beveled edges, not rounded? If the surface were flat, then I could just do Minkoski with a cube rotated into a particular orientation, but it's curved. I'm feeling like Minkowki is pretty cool and all that, but _isn't_ a generalized solution to the problem.

nophead, 2020-09-30:
Any operation on an object with that amount of detail will take forever with CGAL. I always limit the resolution to match FDM printing, so I don't generate things that smooth. But there are other 3D libraries that are many orders of magnitude faster.

To make chamfered cutouts like those I would hull a larger version of the shape with the shape to produce a chamfered version and then subtract it from both sides. Only works for convex cutouts. Linear_extrude with scale would do concave ones I think. 

Obviously it more effort than just selecting an edge and say chamfer that but that is the nature of OpenSCAD. It isn't a fast way of doing anything.

t-paul, 2020-09-30:
That is the nature of OpenSCAD *now*. To me it seems the point of this discussion would be if and how that could be changed.

logiclrd, 2020-09-30:
@nophead I am in fact printing that bottle, about 10cm wide at the base, and even with 150 edges forming each perimeter the printer generates visible facets. I don't want the visible facets, so having 400 edges on each perimeter and 1 _is_ matching the printer's resolution :-P Similar argument on the spacing in the Z direction across the curve up to the neck.

nophead, 2020-09-30:
I just think modelling mathematically is slower than point and click and always will be.

logiclrd, 2020-09-30:
I don't see why it should inherently need to be. Ultimately, a point-and-click interface is just a series of instructions that _could_ be delivered programmatically. I think the reason OpenSCAD has these performance issues is that the functional language to which it is tied means that shortcuts can't be taken for some kinds of operation. It's got a mathematical purity to it, which is awesome, but at the same time sometimes less practical. That mathematical purity, of course, has allowed for things like ImplicitCAD to exist.

nophead, 2020-09-30:
I don't think the reason is is slow is anything to do with the script language being functional. It is simply because CGAL is the slowest geometry library by several orders of magnitude.

t-paul, 2020-09-30:
Yep, the OpenSCAD language is neither fast nor optimized, but compared to the geometry calculation it's usually multiple orders of magnitude faster (obvious example: preview/F5 vs. render/F6). And in addition the boolean operation we use in CGAL are the slowest one can think of (while trying to be very exact). There's now even inside CGAL newer options that might be much faster (as claimed in #2360).

logiclrd, 2020-09-30:
Well, fair enough, but it's still a fact that Minkowski sums can only represent a narrow slice of all the possible chamfering and filleting operations one might want to do, the biggest limitation I think being that you can't limit the effect to only _part_ of a mesh.

nophead, 2020-09-30:
For those cases l union bits together.

logiclrd, 2020-09-30:
But that doesn't work terribly well when there are coincident faces, in my experience. If you're taking a bigger shape, chopping it up to do transformations on just parts, and then putting them back together, there are going to be issues at the joins.

[image]

[image]

dpellegr, 2020-09-30:
@logiclrd I usually workaround those by adding few microns here and there.

logiclrd, 2020-09-30:
It feels really funny to be using this language whose main goal is to allow extreme mathematical precision in representation, and then to be adding fudge factors in to make Boolean operations work properly. It is my understanding, though, that it is virtually an intractable problem to make Booleans work with coincident surfaces. If the language allowed a transformation on a mesh to be explicitly applied only to a part of it, it would be a non-issue!

dpellegr, 2020-09-30:
Extreme mathematical precision is an illusion. How do you represent sqrt(2) or sqrt(3)? You may say "who cares" well any time you apply a rotation of 30 or 45 degrees you get one of them.

Anyway something like you are saying is already implemented in [Cadquery](https://github.com/CadQuery/cadquery). It does not comes with another bunch of issues, see how it nicely fails on a pretty basic union: https://groups.google.com/g/cadquery/c/C1mMNHhBo8U/m/fdAbqQdvAwAJ

Maybe it will mature as will OpenSCAD do

nophead, 2020-09-30:
You have to separate z fighting from actual errors. I manage to make exact
objects that also preview correctly. If l can't avoid it l use render
arround difference.

logiclrd, 2020-09-30:
Isn't Z fighting where it's separate bodies that happen to be at exactly the same difference from the camera, so it isn't clear which should take precedence for any given pixel? A separate issue from a boolean operation where the desired output is a single body but the two inputs have a coincident surface that the boolean operation should be subsuming. I've read that the latter is basically an intractable problem, and that nobody has found a perfect solution for it, and off the top of my head I can see how that could result from floating point inaccuracy -- it's basically impossible to get to the same point by two different values and get _literally exactly_ the same floating point number. I've not worked with this at all, though, this is just my vague understanding from a difference.

nophead, 2020-09-30:
Yes z fighting is when opencsg has coincident faces that are different
colours. So it can be avoided with union if both parts are the same colour.
With difference a slight overlap or render works.

On the other hand CGAL can handle exactly coincident faces, so there is no
need to make inaccurate objects.

mrkmg, 2020-11-13:
This is the exact reason I started work on #2079 

Using an offset extrude, one should be able to chamfer or fillet (with a little cleverness).

TLC123, 2021-08-14:
#884 #3447
!!!! Solved in user space !!!!  :
Enabled by fast convex + convex minkowski sum. Thanks whoever implemented that.

Unions of _simple_  operands has a general FAST solution in all cases where as the joint area is convex 

    For non convex operands, a masking system enable us to isolate out only local convex areas to fillet.
    Masks are in practice a common volume that both operands are first intersected by.

Best of luck.

Write up:
https://forum.openscad.org/I-made-a-thing-unionRound-fillet-union-td33534.html

<a href="http://www.youtube.com/watch?v=gVk-Keg_nGQ">[on Youtube]</a> 
<a href="github.com/TLC123/OpenSCAD_stuff/blob/d6fa00e498313d98a9ea31902600cc8a0af1b9bc/2021/unionRoundSimple.scad">[on GitHub]</a>
<a href="www.thingiverse.com/thing:4932117/">
[on Thingiverse]</a>
<a href="openscadsnippetpad.blogspot.com/2021/08/openscad-unionround-module.html">
[on OpenSCAD Snippet Pad]</a>
<a href="www.reddit.com/r/openscad/comments/p43xqc/unionround_module/">
[on r /OpenSCAD]</a>

jordanbrown0, 2023-11-18:
Duplicate of #884.

closed 2023-11-18 by jordanbrown0