openscad/openscad #3455: Command Line API?

edmundsj, 2020-10-05:
Is there a command-line API for this tool, or do I have to use the GUI? I would much prefer to simply invoke this from the command line.

edmundsj, 2020-10-05:
JK, I see there is a section of the user manual dedicated to this. Closing issue.

closed 2020-10-05 by edmundsj

t-paul, 2020-10-05:
Plus new feature in snapshot versions as of yesterday :grinning: -> #3407.

edmundsj, 2020-10-05:
Great work! This is an awesome package, by the way. I'm currently using [openpyscad](https://github.com/taxpon/openpyscad) to generate the source files in python, it would be really neat if this had a python wrapper. I suggested to the author of openpyscad that he do just that by hooking into your CLI. I think that would greatly improve usability.