openscad/openscad #3457: Crash on rendering malformed polyhedron

dottedmag, 2020-10-10:
Reproducible on 2019.05 and 2020.10.09 (git 7c2074fa5), macOS 10.14 (haven't tried other OSes).

Rendering attached file crashes OpenSCAD with SIGSEGV.

Stacktrace:
<pre>
* thread #11, name = 'QThread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x000000010d55c6f8 OpenSCAD`void CGAL::polygon_mesh_to_nef_3<CGAL::Polyhedron_3<CGAL::Cartesian<CGAL::Gmpq>, CGAL::Polyhedron_items_3, CGAL::HalfedgeDS_default, std::__1::allocator<int> >, CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool>, CGAL::internal::Polyhedron_index_map_external<CGAL::internal::In_place_list_iterator<CGAL::HalfedgeDS_in_place_list_face<CGAL::I_Polyhedron_facet<CGAL::HalfedgeDS_face_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Plane_3<CGAL::Cartesian<CGAL::Gmpq> > > > >, std::__1::allocator<CGAL::HalfedgeDS_in_place_list_face<CGAL::I_Polyhedron_facet<CGAL::HalfedgeDS_face_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Plane_3<CGAL::Cartesian<CGAL::Gmpq> > > > > > > >, CGAL::internal::Polyhedron_index_map_external<CGAL::internal::In_place_list_iterator<CGAL::HalfedgeDS_in_place_list_halfedge<CGAL::I_Polyhedron_halfedge<CGAL::HalfedgeDS_halfedge_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true> > > >, std::__1::allocator<CGAL::HalfedgeDS_in_place_list_halfedge<CGAL::I_Polyhedron_halfedge<CGAL::HalfedgeDS_halfedge_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true> > > > > > > >(CGAL::Polyhedron_3<CGAL::Cartesian<CGAL::Gmpq>, CGAL::Polyhedron_items_3, CGAL::HalfedgeDS_default, std::__1::allocator<int> >&, CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool>&, CGAL::internal::Polyhedron_index_map_external<CGAL::internal::In_place_list_iterator<CGAL::HalfedgeDS_in_place_list_face<CGAL::I_Polyh
edron_facet<CGAL::HalfedgeDS_face_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Plane_3<CGAL::Cartesian<CGAL::Gmpq> > > > >, std::__1::allocator<CGAL::HalfedgeDS_in_place_list_face<CGAL::I_Polyhedron_facet<CGAL::HalfedgeDS_face_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Plane_3<CGAL::Cartesian<CGAL::Gmpq> > > > > > > >, CGAL::internal::Polyhedron_index_map_external<CGAL::internal::In_place_list_iterator<CGAL::HalfedgeDS_in_place_list_halfedge<CGAL::I_Polyhedron_halfedge<CGAL::HalfedgeDS_halfedge_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true> > > >, std::__1::allocator<CGAL::HalfedgeDS_in_place_list_halfedge<CGAL::I_Polyhedron_halfedge<CGAL::HalfedgeDS_halfedge_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true> > > > > > >) + 728
OpenSCAD`CGAL::polygon_mesh_to_nef_3<CGAL::Polyhedron_3<CGAL::Cartesian<CGAL::Gmpq>, CGAL::Polyhedron_items_3, CGAL::HalfedgeDS_default, std::__1::allocator<int> >, CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool>, CGAL::internal::Polyhedron_index_map_external<CGAL::internal::In_place_list_iterator<CGAL::HalfedgeDS_in_place_list_face<CGAL::I_Polyhedron_facet<CGAL::HalfedgeDS_face_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Plane_3<CGAL::Cartesian<CGAL::Gmpq> > > > >, std::__1::allocator<CGAL::HalfedgeDS_in_place_list_face<CGAL::I_Polyhedron_facet<CGAL::HalfedgeDS_face_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Plane_3<CGAL::Cartesian<CGAL::Gmpq> > > > > > > >, CGAL::internal::Polyhedron_index_map_external<CGAL::internal::In_place_list_iterator<CGAL::HalfedgeDS_in_place_list_halfedge<CGAL::I_Polyhedron_halfedge<CGAL::HalfedgeDS_halfedge_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true> > > >, std::__1::allocator<CGAL::HalfedgeDS_in_place_list_halfedge<CGAL::I_Polyhedron_halfedge<CGAL::HalfedgeDS_halfedge_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true> > > > > > > >:
->  0x10d55c6f8 <+728>: movq   (%rax), %rax
    0x10d55c6fb <+731>: movq   0x18(%rax), %rax
    0x10d55c6ff <+735>: movq   0x8(%rax), %rax
    0x10d55c703 <+739>: movq   %rax, -0x88(%rbp)
Target 0: (OpenSCAD) stopped.
(lldb)  bt
* thread #11, name = 'QThread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
  * frame #0: 0x000000010d55c6f8 OpenSCAD`void CGAL::polygon_mesh_to_nef_3<CGAL::Polyhedron_3<CGAL::Cartesian<CGAL::Gmpq>, CGAL::Polyhedron_items_3, CGAL::HalfedgeDS_default, std::__1::allocator<int> >, CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool>, CGAL::internal::Polyhedron_index_map_external<CGAL::internal::In_place_list_iterator<CGAL::HalfedgeDS_in_place_list_face<CGAL::I_Polyhedron_facet<CGAL::HalfedgeDS_face_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Plane_3<CGAL::Cartesian<CGAL::Gmpq> > > > >, std::__1::allocator<CGAL::HalfedgeDS_in_place_list_face<CGAL::I_Polyhedron_facet<CGAL::HalfedgeDS_face_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Plane_3<CGAL::Cartesian<CGAL::Gmpq> > > > > > > >, CGAL::internal::Polyhedron_index_map_external<CGAL::internal::In_place_list_iterator<CGAL::HalfedgeDS_in_place_list_halfedge<CGAL::I_Polyhedron_halfedge<CGAL::HalfedgeDS_halfedge_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true> > > >, std::__1::allocator<CGAL::HalfedgeDS_in_place_list_halfedge<CGAL::I_Polyhedron_halfedge<CGAL::HalfedgeDS_halfedge_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true> > > > > > > >(CGAL::Polyhedron_3<CGAL::Cartesian<CGAL::Gmpq>, CGAL::Polyhedron_items_3, CGAL::HalfedgeDS_default, std::__1::allocator<int> >&, CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool>&, CGAL::internal::Polyhedron_index_map_external<CGAL::internal::In_place_list_iterator<CGAL::HalfedgeDS_in_place_list_face<CGAL::I_Polyhedron_facet<CGAL::HalfedgeDS_face_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Plane_3<CGAL::Cartesian<CGAL::Gmpq> > > > >, std::__1::allocator<CGAL::HalfedgeDS_in_place_list_face<CGAL::I_Polyhedron_facet<CGAL::HalfedgeDS_face_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::__1::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Plane_3<CGAL: [... cut ...]
    frame #11: 0x000000010f743140 QtCore`QMetaObject::activate(QObject*, int, int, void**) + 2320
    frame #12: 0x000000010f54e868 QtCore`QThread::started(QThread::QPrivateSignal) + 56
    frame #13: 0x000000010f551f46 QtCore`___lldb_unnamed_symbol298$$QtCore + 326
    frame #14: 0x00007fff67ae82eb libsystem_pthread.dylib`_pthread_body + 126
    frame #15: 0x00007fff67aeb249 libsystem_pthread.dylib`_pthread_start + 66
    frame #16: 0x00007fff67ae740d libsystem_pthread.dylib`thread_start + 13
(lldb) 
[bug.txt](https://github.com/openscad/openscad/files/5358953/bug.txt)

</pre>

dottedmag, 2020-10-10:
[bug.txt](https://github.com/openscad/openscad/files/5358954/bug.txt)

MichaelAtOz, 2020-10-10:
Probably #2847.

t-paul, 2021-07-27:
Fixed by #3526.

closed 2021-07-27 by t-paul