openscad/openscad #3460: Option to use fixed origin for SVG export instead of semi-arbitrarily decide one

dogtopus, 2020-10-13:
Currently OpenSCAD just uses the minimum xy coordinates that won't crop anything as the origin point, which really messes up use cases that require precise alignment like laser engraving over previously cut objects. It would be nice to have user set an origin point (like (0, 0)) instead of latching on shapes which could have arbitrary size and alignment.

naggie, 2021-02-17:
I had previously resolved this by exporting a DXF instead. However, there are issues with the DXF support (like curves being a collection of independent lines instead of connected).

I suppose a workaround is to somehow put a small dot at the origin and max x/y.

This would be a welcome change.

jordanbrown0, 2021-02-17:
Not to ask excessively stupid questions, but why isn't the SVG origin simply the OpenSCAD origin?

t-paul, 2021-02-17:
Because they have a different Y axis direction.

jordanbrown0, 2021-02-17:
I exported an SVG from OpenSCAD and looked at it from Inkscape.

It appears that the SVG origin is in the same place as the OpenSCAD origin, but with Y negated (as is kind of required, since SVG uses a "plus Y is down" system, while OpenSCAD uses "plus Y is up").

That's not what the primary Inkscape UI says - its rulers have the origin in the top-left of the drawing - but that appears to be a transformation layered on top of the SVG coordinate system.  I can't say that I fully understand Inkscape's coordinate system, but if I set the viewbox origin to 0,0 (File, Document Properties, Page, Scale, Viewbox, X and Y) then the rulers line up with the SVG and OpenSCAD origins.

Doing that puts the "page" in a perhaps-stupid place, down-right of the origin, in what SVG calls +X/+Y and OpenSCAD calls +X/-Y, but it does put all of the origins in the same place relative to the object..

Does that do what you need?

(Note that Inkscape's display coordinate system got flipped somewhere around 1.0.  I'm referring here to 1.0's default settings, where [0,0] is in the top left and +Y is down.  Ref https://wiki.inkscape.org/wiki/index.php/Release_notes/1.0#Y-Axis_Inversion)

chrysn, 2024-03-14:
> Because they have a different Y axis direction.

That is merely a matter of 3D orientation – but granted, doing literal coordinate translation would look surprising to users.

I propose resolving this by introducing a `literal_coordinates` argument to the importer that only applies to import format such as SVG that have right-down axis semantics. It would be documented like this:

* **literal_coordinate**
   Boolean. If set to true, the numeric values of x and y coordinates are used without format specific transformations.

   This argument only applies to SVG (or similar future formats): SVG's coordinate system applies the *y* value toward the bottom of the screen on the drawing plane, whereas in OpenSCAD's top view, *y* points toward the top of the screen. By default (i.e., when literal_coordinate=false), the import command flips an SVG file's *y* coordinates, and shifts the whole drawing into the positive area.

   Using literal_coordinate=true is particularly useful when the OpenSCAD script has additional information about positions inside the SVG file's coordinate system.

   This is mutually exclusive with the "center" argument.

(Depending on the implementation it may not be necessary to make this mutually exclusive with center, but semantics would be weird if it was allowed)

chrysn, 2024-03-14:
Starting to look at the code, there's one more transformation that is applied to SVGs: SVG units are transformed to mm. Options are to either update the documentation to say that coordinates "in mm" are imported (and consider a different name for the argument), or to do literally literal coordinates, and also import coordinates in SVG native units (pixels, AIU). I'd prefer the latter, but either would resolve the issue for me.

jordanbrown0, 2024-03-14:
WRT units... it looks like SVG, strictly speaking, can use any of several units or, for some properties, an unspecified "user coordinate system".  (It looks like properties with no units are allowed in SVG1.1 properties but not in SVG2.)

https://developer.mozilla.org/en-US/docs/Web/SVG/Content_type#length

>A length is a distance measurement, given as a number along with a unit. The SVG2 specification aligns with CSS [<length>](https://developer.mozilla.org/en-US/docs/Web/CSS/length) data types and units for the attribute syntax and values. A length unit identifier must be provided [...]

> For SVG-specific properties defined in SVG1.1 and their corresponding presentation attributes, the unit identifiers in values are optional. If not provided, the length value represents a distance in the current user coordinate system.

Inspecting a simple Inkscape-generated SVG, the document dimensions are explicitly in mm, and there is an Inkspace-specific attribute setting the document units to mm, but the actual geometric objects have no specified units and there is no generic property selecting a global unit.

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!-- Created with Inkscape (http://www.inkscape.org/) -->

<svg
   width="210mm"
   height="297mm"
   viewBox="0 0 210 297"
   version="1.1"
   id="svg1"
   inkscape:version="1.3.2 (091e20e, 2023-11-25, custom)"
   sodipodi:docname="junk.svg"
   xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
   xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:svg="http://www.w3.org/2000/svg">
  <sodipodi:namedview
     id="namedview1"
     pagecolor="#ffffff"
     bordercolor="#000000"
     borderopacity="0.25"
     inkscape:showpageshadow="2"
     inkscape:pageopacity="0.0"
     inkscape:pagecheckerboard="0"
     inkscape:deskcolor="#d1d1d1"
     inkscape:document-units="mm"
     inkscape:zoom="1.6863847"
     inkscape:cx="397.00313"
     inkscape:cy="561.25984"
     inkscape:window-width="2902"
     inkscape:window-height="2073"
     inkscape:window-x="494"
     inkscape:window-y="494"
     inkscape:window-maximized="0"
     inkscape:current-layer="layer1" />
  <defs
     id="defs1" />
  <g
     inkscape:label="Layer 1"
     inkscape:groupmode="layer"
     id="layer1">
    <rect
       style="fill:none;stroke:#000000;stroke-width:0.264583"
       id="rect1"
       width="29.966719"
       height="26.201267"
       x="17.572109"
       y="17.572109" />
  </g>
</svg>
```

So it would seem that there are two questions for SVG import:
* What is the conversion between unspecified-unit SVG lengths and OpenSCAD units?
  * 1:1 seems obvious.
* What is the conversion between specified-unit SVC lengths and OpenSCAD units?
  * 1mm:1 seems obvious, because although OpenSCAD is nominally unitless the most common applications use one mm per OpenSCAD unit.  The `import()` module could have a `scale=` or `unit=` option, but it's not really necessary since you can always slap whatever `scale()` you want around it once you know that one SVG mm is one OpenSCAD unit.

Similarly, SVG export has one or more questions:
* Do SVG exports specify units?
  * "No" seems obvious, if it's practical.
* If SVG exports must specify units, what units do they use and what is the conversion from OpenSCAD units to those units?
  * Using mm, and converting 1:1mm seems obvious.
* Should there be an export option allowing the user to specify what units to use and what the conversion ratio is?