openscad/openscad #3462: [Feature requst] Running external editor from within openSCAD

pedjas, 2020-10-14:
It would be great to have option to run external editor from OpenSCAD.

Operating system can be set to run only one application on file double click. By default OpenSCAD runs on *.scad files. If one wants to use other editor he must manually start editor and load .scad file. That is cymbersome.

If there is option within OpenSCAD to run external editor, things would be much easier. User would run .scad file in OpenSCAD and then switch to external editor with simple click or keypress.

MichaelAtOz, 2020-10-14:
You can right-click/open-with (Windows - there is similar on Linux) to open with an editor.

pedjas, 2020-10-14:
On my computer I have option to open with but then selected application becomes default for oppening file extension.

In good old times, Windows had options to set default application sepearately for opening and editing file, but now it seems they decided to toss that functionality out.

MichaelAtOz, 2020-10-14:
With Windows-7, right-click/open-with:
[image]

choose-default-program:

[image]

Un-check always-use...
Select your editor.
Then next time your editor will be on the list of right-click/open-with.

I did similar on Windows-10, it is slightly different, but same outcome.

I'm not too familiar with Linux.

t-paul, 2020-11-08:
It seems we should leave that to the environment/OS then.

closed 2020-11-08 by t-paul