openscad/openscad #3463: Warnings when rendering model that uses text() on command-line

gingerbeardman, 2020-10-14:
I call text() as follows:
```
    text("藤原とうふ店", 4.7, "Corporate Logo Medium:style=Medium", halign="left", valign="left", spacing=1);
```

I call opens cad on command-line as follows, as part of a loop in a shell script:
```
    openscad \
    -q \
    -D rl=$RL \
    -D rr=$RR \
    -D rf=$RF \
    -D rb=$RB \
    -D shad=$SHADOW \
    --autocenter \
    --camera 0,2,28,45,0,$ANGLE,500 \
    --imgsize 320,320 \
    -o $FOLDER/$CAR-$FRAME-$NUMBER.png \
    _$CAR.scad &
```

```
Fontconfig warning: "/Applications/OpenSCAD.app/Contents/Resources/fonts/fonts.conf", line 84: unknown element "blank"
```

[fonts.conf.zip](https://github.com/openscad/openscad/files/5379890/fonts.conf.zip)

I also see an odd green colour in my render:

app
[image]

render
[image]

t-paul, 2020-10-14:
The warning is covered by #2888. Is it otherwise working?

gingerbeardman, 2020-10-14:
Ah, sorry I did not search before posting.

Yes, otherwise all is working OK with the render. Closing.

closed 2020-10-14 by gingerbeardman