openscad/openscad #3465: Manage testing & release of Multi-Process OpenSCAD

MichaelAtOz, 2020-10-17:
In the dim dark past [a challenge was issued](https://github.com/openscad/openscad/issues/391), it lay defiantly unanswered for years, until @devilman3d [tried valiantly](https://github.com/openscad/openscad/pull/1980) only to be smitten by CGAL, now @justbuchanan has taken the [gauntlet](https://github.com/openscad/openscad/pull/2399), buy yet again CGAL triumphs, the battlefield lay silent, until @slaviber chose a [different attack](https://github.com/openscad/openscad/pull/3193), so the story continues...

 I suspect this will need significant testing & may well require a few iterations. So I propose the pull can be used initially, but this should be used for the longer term.

MichaelAtOz, 2020-10-17:
This is for problems & related discussion, leaving the pull for code development topics.

MichaelAtOz, 2020-10-17:
@slaviber for testing the previous attempt I wrote something to make lots of work for CGAL, see [random-threadsV4.zip](https://github.com/openscad/openscad/files/5395157/random-threadsV4.zip).
Basically I run it in random mode, if I see something funny, I override with the specific `key=` for repeatability

I'm testing 2020.10.11 against your 2020.10.14.

On first try, with comparatively light load, I have a discrepancy.
```
random-threadsV4.scad
[2020, 10, 11]
Rendering Polygon Mesh using CGAL...
Geometries in cache: 1509
Geometry cache size in bytes: 2928328
CGAL Polyhedrons in cache: 14
CGAL cache size in bytes: 54927008
Total rendering time: 0:02:00.830
   Top level object is a 3D object:
   Simple:        yes
   Vertices:     8967
   Halfedges:   43466
   Edges:       21733
   Halffacets:  26830
   Facets:      13415
   Volumes:       457
Rendering finished.

[2020, 10, 14]
Rendering Polygon Mesh using CGAL...
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
Geometries in cache: 1509
Geometry cache size in bytes: 2928328
CGAL Polyhedrons in cache: 14
CGAL cache size in bytes: 55492840
Total rendering time: 0:01:07.044
   Top level object is a 3D object:
   Simple:        yes
   Vertices:     8960
   Halfedges:   43932
   Edges:       21966
   Halffacets:  27312
   Facets:      13656
   Volumes:       457
Rendering finished.
```

Aaargh! I just tried against 2019.05 and got a different set...
```
Rendering Polygon Mesh using CGAL...
Geometries in cache: 1508
Geometry cache size in bytes: 2876288
CGAL Polyhedrons in cache: 14
CGAL cache size in bytes: 54035728
Total rendering time: 0 hours, 1 minutes, 55 seconds
Top level object is a 3D object:
Simple: yes
Vertices: 8751
Halfedges: 42672
Edges: 21336
Halffacets: 26500
Facets: 13250
Volumes: 461
Rendering finished.
```

@t-paul shouldn't there be identical stats, edges/faces etc? They are the same CGAL version.

MichaelAtOz, 2020-10-18:
Hold-off looking at this. I found something funny with seeding...

MichaelAtOz, 2020-10-18:
OK, seeding difference is between 2019.05 and the other two. 
Version 2020.10.11 & 2020.10.14 (multi-process) produced identical echoes i.e. all  the `i=`,
so should produce identical geometry.

@slaviber Do you expect your optimisations to produce identical geometry?

slaviber, 2020-10-18:
@MichaelAtOz Yes, the end result should be identical.
The rendering pipeline is very different, though.

slaviber, 2020-10-18:
I ran your test on 2020.10.14 with and without the multi-processing feature enabled and got the same results as yours.  
The number of volumes is equal, objects are manifold. There is no visual difference, as far as I can tell, no CGAL errors with the  "--debug=cgalutils-multithreaded" command line parameter.  
The generated mesh is different but that is a natural consequence of the different triangulation path, I think.

MichaelAtOz, 2020-10-18:
I expect this will lead to discussions on what defines reality.

[random-threadsV5small.zip](https://github.com/openscad/openscad/files/5398337/random-threadsV5small.zip)
Export STLs with .11 & .14, in Netfabb volume is 1697.00 v's 1697.01.

[random-threadsV5small-STLs.zip](https://github.com/openscad/openscad/files/5398347/random-threadsV5small-STLs.zip)

Difference the two. If geometry is identical there should nothing left.

[image]
[image]

They look like co-incident faces.

MichaelAtOz, 2020-10-19:
Now I'm testing only 2020.10.14, but turning the feature & off.

Slightly evolved program 
[random-threadsV6small.zip](https://github.com/openscad/openscad/files/5398852/random-threadsV6small.zip)

F6-off:
[image]

F6-on:
[image]

They eyeball the same.

STLs in MeshLab

Off:

[image]

On:

[image]

Very different tessellation, particularly the spheres. That would explain the Netfabb volume difference.
And `difference()` differences.

Question is, is this a problem?

I think it is. But is this export generating the difference?

Surly the detached sphere in the middle, should be made the same?

Off:
[image]

On:
[image]

slaviber, 2020-10-19:
> ...this will lead to discussions on what defines reality.  

More like a discussion on expectations.  Reality is just that - imprefect.   
  
> If geometry is identical there should nothing left.  

That's only correct if one of the sets contains the other. As in the mathematical representation of the scene. But your difference tests clearly show that neither set contains the other one. Only up to some discrete precision - either a rounding or a grid error  

The tessellation is out of my control, however.  

Here I did the same test but with the native nef3 format without export floating-point error:  

[image]
The image above shows a simple difference.  
.  
.  
.  
[image]
On the above image you can see that I made a very rough and inaccurate approximation of a 10-micron centered shell and then subtracted the shell from the vanilla mesh.  

@MichaelAtOz Do you think that such sub-precision and tessellation differences would affect the end-user? I always thought that's the reason why there are PNG tests and not pure mesh tests. Moreover, the Documentation doesn't state anything on mesh compatibility between versions or tolerances.  

That's a comparison between two different approximations of the same mathematical model by using two completely different algorithms. But with finite precision.

MichaelAtOz, 2020-10-20:
I'm still mulling this over & doing other tests.

The "PolySet has nonplanar faces. Attempting alternate construction" is concerning.
It is indicative of introduced inaccuracies, not just the ones in the export conversion to floats.

nophead, 2020-10-20:
If the STLs change it would break my workflow as I require them to be the same unless they are actually changed.

Why does anything change at all with parallel processing? Isn't it exactly the same operations on the same data, just done in parallel instead of serially?

slaviber, 2020-10-24:
@MichaelAtOz 
From my experience with the code, the "PolySet has nonplanar faces. Attempting alternate construction" thing happens only when I'm merging multiple PolySets with sharp edges. Maybe CGAL predicts that it's default [meshing algorithm](https://en.wikipedia.org/wiki/Delaunay_triangulation) would produce inaccurate geometry and then changes it's [magic](https://doc.cgal.org/latest/Triangulation_3/index.html) so that the mesh could be properly evaluated. And if the source object "leaks" than CGAL defaults to a harder error.  

Now, I'm neither a CGAL nor an OpenSCAD expert. That's the end result when I apply my optimizations. If something breaks I try another approach and so on, but I can't control the mesh.  
I found a way to generate the same mesh in parallel but this blows away 50% of my optimization so I wonder why is mesh compatibility such an issue.  

@nophead Yes it's the same operations. But the order in which they are done apparently alters the way the geometry is generated.

nophead, 2020-10-24:
But the order does not need to change. It is just a tree where the parents are a geometric combination of the children, which must be evaluated first. So children can be evaluated in parallel instead of serially but that shouldn't affect the results. Are you saying you don't evaluate say a union in the order the code specifies? I make use of that to optimise my code. I try to keep the number of facets small until the end of the union / difference.

slaviber, 2020-10-24:
I see multiple misconceptions.

Specifically for the children that are getting evaluated:  
One thing is when you have to combine the objects sequentially in some order, another thing is when every children gets combined with every other children in parallel until only one (three in my case) object is left. 

Are you telling me about an optimization in the source code or an optimization in the .scad file?

nophead, 2020-10-24:
I mean optimising the scad file. For example, if you have a union of several items it is quicker to union ones with less facets first and add the complex ones at the end.

If you have a union of n children then if you evaluate all the children in parallel and then union them one at a time in the order specified in the source code then the results will be the same and deterministic. If you split up the union of n into doing pairs of children in parallel then yes it might give slightly different results, especially with the grid snap.

slaviber, 2020-10-24:
Uhm, I don't know from how much time, but the order of the items in the .scad file is irrelevant because they are sorted by number of facets on **master**. You can check that [here](https://github.com/openscad/openscad/blob/master/src/cgalutils-applyops.cc#L155), [here](https://github.com/openscad/openscad/blob/master/src/cgalutils-applyops.cc#L150) and so on. I'm also sorting the remaining objects because it's fast.   

The Nef Polyhedrons themselves aren't the bottleneck, the bottleneck, as stated in the first comment of my PR, are the boolean operations. So that's what's parallellized - the unions, the intersections and the differences.   

Yes, the grid snap amounts to a bit of error, but the main problem remains the merging of sharp PolySets before they are converted to Polyhedrons, only in that case the mesh changes compared to **master**.

dgrnbrg, 2021-01-27:
As a longtime user of openscad and a professional dev specialized in parallel scientific computing, I hope it's okay for me to add my $.02.

It's sometimes the case that certain parallelizations have precision diffs (both deterministic and nondeterministic). In my experience, for these systems whose outputs are ultimately turning into something real/simulated, manufacturing tolerances and texture/lighting application render the differences moot.

I hope that a backwards compatibility flag and/or the image-based testing lets this get shipped. Thank you for your consideration.

MichaelAtOz, 2021-05-20:
I'll do my 2¢ on the theology later.

Re my recent n=8000 freeze mentioned in #3193.

I think the later optimisations has changed how my stress test is processed.
Previously, the `part` loop actually created groups of unions, which formed growing clusters which were dolled out to various threads. 
As the test is basically n discreet objects which randomly overlap, the new optimisation doesn't care, it is just a big union, with no other operations.
So it spends a lot of time doing something (including often the "PolySet has nonplanar faces. Attempting alternate construction" - basically rebuilding structures. It looks like the something is generating the 8000 primitives, pre the CGAL union.
Then BANG, slam some large sets of unions into discreet processes. Potentially 4000, then 2000, 1000 etc (I'll have to have a look at the code).

Not ideal for a representative test, but one hell of a stress test...still valid.
Many designs frequently have lots of base objects.

I'll think about getting to do other load mixes on top of the union.

MichaelAtOz, 2021-05-20:
I'm actually impressed the system didn't meltdown BTW. A few processes got killed to free memory, but it chugged away, fans on full.

Note for later: test scenario, a spawnchild dies (maybe memory memory exhausted), what happens?

MichaelAtOz, 2021-05-20:
OK, it finished, Oh that was just n=3000, not 8000....perhaps try 1500 first next time.
RAM 31.1GB 99.4% & Swap 22.9GB 72.2%.

```
Compiling design (CSG Tree generation)...
ECHO:"random-threadsV8.scad"
ECHO:version = [2020, 10, 14]
ECHO:""
ECHO:"seeds=[6.05444, 18.0085, -0.590766, 19.2072, -1.22781, -9.5463, 18.8305]"
ECHO:D = 3, key = 836094, n = 3000, $fn = 32, parts = 59, x = 1920, y = 1080
Rendering Polygon Mesh using CGAL...
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
[... 16 lines cut ...]
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
Geometries in cache: 9639
Geometry cache size in bytes: 359791816
CGAL Polyhedrons in cache: 36
CGAL cache size in bytes: 5214543280
Total rendering time: 1:46:16.268
   Top level object is a 3D object:
   Simple:        yes
   Vertices:   841077
   Halfedges:  4623596
   Edges:      2311798
   Halffacets: 2935506
   Facets:     1467753
   Volumes:      1269
Rendering finished.
```

System monitor shows a process AppRun.wrapped , Virtual 43GiB, Resident Memory 27GiB.
If you compare that to the cache sizes above, either there is a leak, a duplication in all the inter-process copying, or really bad structures.

My first 1¢ of theology, 80 nonplanar polysets...(possibly not multiprocess related), but for relatively simple test objects, something internal is not producing good geometry.

lf94, 2021-05-20:
@MichaelAtOz I spent yesterday just experiencing the changes while working on a project (yes, I had multi-processing --enabled). For sure there is something weird going on. I found hull() and minkowski() to be oddly slower?

I am starting to think this attempt at parallelization is no good.