openscad/openscad #3467: CGAL crashes while subtracting a generated mesh from an import, preview works fine

logiclrd, 2020-10-18:
I'm trying to create my own version of an articulated snake, because an otherwise really-nice model on Thingiverse has a ton of annoying errors, like separate segments intersecting each other with the intersections baked into the geometry. I've been working on a fairly sophisticated OpenSCAD script to do this, generating segments and placing them so that they interlock, and for the head of the snake, I extracted the head from the broken STL, did quite a bit of work on it with Meshmixer, and got it to a point where both Meshmixer and Rhino3D say there's nothing at all wrong with it.

The script has a module that generates the cutout for each end of a piece -- "up" toward the tail, "down" toward the head (chosen arbitrarily, probably would do it the other way around if I was starting over :-P ) -- and the module for the head piece takes the imported STL and subtracts an appropriately-proportioned "up" cutout from it:

```
module up_cutout(
  up_width, up_height, previous_down_height, joint_bar_size,
  previous_rounding, rounding)
{
  translate([0, 0.5 * joint_bar_size - rounding, 0])
  minkowski()
  {
    translate([-0.25 * up_width + 0.5 * joint_rounding, 0, 0.5 * up_height + previous_rounding - rounding])
    rotate([0, 90, 0])
    difference()
    {
      cylinder(
        h = up_width * 0.5 - joint_rounding,
        r = previous_down_height * 0.5 + joint_tolerance * 1.25, // TODO: why 1.25?
        $fn = detail * 2.5);

      cylinder(
        h = up_width * 0.5 + joint_rounding,
        r = joint_bar_size * 0.5 + joint_rounding,
        $fn = detail * 1.5);
    }
    
    sphere(
      r = joint_rounding,
      $fn = detail);
  }
}

module apply_up_cutout(
  up_width, up_height, previous_down_height, joint_bar_size,
  length,
  down_width, down_height, joint_hook_width, next_joint_bar_size,
  previous_rounding, rounding)
{

  difference()
  {
  children();

    up_cutout(
      up_width, up_height, previous_down_height, joint_bar_size,
      previous_rounding, rounding);
  }
}

module head_segment(height, rounding)
{
  translate([0, -rounding, 0])
  scale(height * 0.1)
  translate([176, -72.5, 0])
  rotate([0, 0, 90])
  import("Snake Head.stl", convexity = 10, center = true);
}
...
    apply_up_cutout(
      up_width, up_height, previous_down_height, joint_bar_size,
      length,
      down_width, down_height, joint_hook_width, next_joint_bar_size,
      0, 0)
    {
      head_segment(previous_down_height, rounding);
    }
```

When I press F5, I get the expected result:

[image]

But when I press F6, apparently CGAL crashes:

[image]

Here are the complete files in question:

[OpenSCAD bug files.zip](https://github.com/openscad/openscad/files/5397377/OpenSCAD.bug.files.zip)

Am I doing something wrong? Is my STL file damaged in some way that the tools I've tried can't detect? Thanks!

MichaelAtOz, 2020-10-18:
Self intersections:
[image]

logiclrd, 2020-10-18:
Huh! None of the tools at my disposal highlighted those, and also that surface was created by doing a Plane Cut in Meshmixer, strange that Meshmixer created self-intersections like that. Do you know of a good technique/tool for fixing them?

MichaelAtOz, 2020-10-18:
That's Netfabb, there has been debate about its SI, but it inevitably leads to CGAL errors.
I have little success fixing SIs. There used to be an online thing, perhaps others know????

Looking closer there are also tiny non-manifolds.
MeshLab [image]
Netfabb 
[image]
MeshLab 
[image]

logiclrd, 2020-10-19:
I put in quite a bit more work and I think I got a well-formed mesh out of it. I figured out how to use MeshLab to track down the self-intersecting polygons, deleted them in Meshmixer and Erased & Filled. Then it boiled down to this weird scar on the side. My guess is that it is 3D scanning noise/error. Anyway, I just deleted it outright, turned it into a smooth surface, then put a "scar" through it to disguise the flatness.

Before:

[image]

After:

[image]

[image]

Thanks for the help, and I learned some new skills identifying these problems and fixing them :-)

closed 2020-10-19 by logiclrd