openscad/openscad #3468: difference renders as union on macOS 10.15

sodiumnitrate, 2020-10-24:
I tried to do the example in the documentation for difference. Here's what should have happened: http://2.bp.blogspot.com/-_99RwKvT9aM/VBBnOHnvs4I/AAAAAAAACxc/Hn8k1MoDDCE/s1600/cube_cylinder_manifold.png

What happens instead:

[image]

Making the sphere transparent does not reveal a "carved out" cube either:

[image]

MichaelAtOz, 2020-10-24:
@sodiumnitrate 
Please post the contents of Help/Library-info.
What is your graphics H/W?

sodiumnitrate, 2020-10-24:
OpenSCAD Version: 2019.05 System information: Mac OS X Version 10.15.6 (Build 19G2021) MacBookPro16,1 6 CPUs 16.00 GB RAM  User Agent: OpenSCAD/2019.05 (Mac OS X Version 10.15.6 (Build 19G2021) MacBookPro16,1) Compiler: Clang "10.0.1 (clang-1001.0.46.4)" 64bit MinGW build: No Debug build: No Boost version: 1_65_1 Eigen version: 3.3.7 CGAL version, kernels: 4.13, Cartesian, Extended_cartesian, Epeck OpenCSG version: OpenCSG 1.4.2 Qt version: 5.9.7 QScintilla version: 2.10.8 InputDrivers:  GLib version: 2.56.3 lodepng version: 20180910 libzip version: 1.5.1 fontconfig version: 2.13.1 freetype version: 2.9.1 harfbuzz version: 2.3.1 lib3mf version: 1.8.1 Application Path: /Users/nano/Dropbox/My Mac (Nanos-MacBook-Pro.local)/Desktop/OpenSCAD.app/Contents/MacOS Documents Path: /Users/nano/Documents User Documents Path: /Users/nano/Documents Resource Path: /Users/nano/Dropbox/My Mac (Nanos-MacBook-Pro.local)/Desktop/OpenSCAD.app/Contents/Resources User Library Path: /Users/nano/Dropbox/My Mac (Nanos-MacBook-Pro.local)/Documents/OpenSCAD/libraries User Config Path: /Users/nano/Library/Application Support/OpenSCAD Backup Path: /Users/nano/Dropbox/My Mac (Nanos-MacBook-Pro.local)/Documents/OpenSCAD/backups OPENSCADPATH:  OpenSCAD library path: /Users/nano/Dropbox/My Mac (Nanos-MacBook-Pro.local)/Documents/OpenSCAD/libraries /Users/nano/Dropbox/My Mac (Nanos-MacBook-Pro.local)/Desktop/OpenSCAD.app/Contents/Resources/libraries  OPENSCAD_FONT_PATH:  OpenSCAD font path: /usr/share/fonts /Users/nano/.local/share/fonts /Users/nano/.fonts /usr/X11/lib/X11/fonts /System/Library/Fonts /Library/Fonts /Users/nano/Library/Fonts /usr/X11/lib/X11/fonts/100dpi /usr/X11/lib/X11/fonts/75dpi /usr/X11/lib/X11/fonts/OTF /usr/X11/lib/X11/fonts/Speedo /usr/X11/lib/X11/fonts/TTF /usr/X11/lib/X11/fonts/Type1 /usr/X11/lib/X11/fonts/cyrillic /usr/X11/lib/X11/fonts/encodings /usr/X11/lib/X11/fonts/misc /usr/X11/lib/X11/fonts/util /System/Library/Fonts/Supplemental /usr/X11/lib/X11/fonts/encodings/large GLEW version: 2.1.0 OpenGL Version: 2.1 ATI-3.10.16 GL Renderer: AMD Radeon Pro 5300M OpenGL Engine GL Vendor: ATI Technologies Inc. RGBA(8888), depth(0), stencil(0) GL_ARB_framebuffer_object: yes GL_EXT_framebuffer_object: yes GL_EXT_packed_depth_stencil: yes  Qt graphics widget: QOpenGLWidget QSurfaceFormat: RGBA(888-1), depth(-1), stencil(-1)  GL Extensions: GL_APPLE_aux_depth_stencil GL_APPLE_client_storage GL_APPLE_element_array GL_APPLE_fence GL_APPLE_float_pixels GL_APPLE_flush_buffer_range GL_APPLE_flush_render GL_APPLE_object_purgeable GL_APPLE_packed_pixels GL_APPLE_pixel_buffer GL_APPLE_rgb_422 GL_APPLE_row_bytes GL_APPLE_specular_vector GL_APPLE_texture_range GL_APPLE_transform_hint GL_APPLE_vertex_array_object GL_APPLE_vertex_array_range GL_APPLE_vertex_point_size GL_APPLE_vertex_program_evaluators GL_APPLE_ycbcr_422 GL_ARB_color_buffer_float GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_depth_texture GL_ARB_draw_buffers GL_ARB_draw_elements_base_vertex GL_ARB_draw_instanced GL_ARB_fragment_program GL_ARB_fragment_program_shadow GL_ARB_fragment_shader GL_ARB_framebuffer_object GL_ARB_framebuffer_sRGB GL_ARB_half_float_pixel GL_ARB_half_float_vertex GL_ARB_imaging GL_ARB_instanced_arrays GL_ARB_multisample GL_ARB_multitexture GL_ARB_occlusion_query GL_ARB_pixel_buffer_object GL_ARB_point_parameters GL_ARB_point_sprite GL_ARB_provoking_vertex GL_ARB_seamless_cube_map GL_ARB_shader_objects GL_ARB_shader_texture_lod GL_ARB_shading_language_100 GL_ARB_shadow GL_ARB_shadow_ambient GL_ARB_sync GL_ARB_texture_border_clamp GL_ARB_texture_compression GL_ARB_texture_compression_rgtc GL_ARB_texture_cube_map GL_ARB_texture_env_add GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_float GL_ARB_texture_mirrored_repeat GL_ARB_texture_non_power_of_two GL_ARB_texture_rectangle GL_ARB_texture_rg GL_ARB_transpose_matrix GL_ARB_vertex_array_bgra GL_ARB_vertex_blend GL_ARB_vertex_buffer_object GL_ARB_vertex_program GL_ARB_vertex_shader GL_ARB_window_pos GL_ATI_blend_equation_separate GL_ATI_blend_weighted_minmax GL_ATI_separate_stencil GL_ATI_texture_compression_3dc GL_ATI_texture_env_combine3 GL_ATI_texture_float GL_ATI_texture_mirror_once GL_EXT_abgr GL_EXT_bgra GL_EXT_bindable_uniform GL_EXT_blend_color GL_EXT_blend_equation_separate GL_EXT_blend_func_separate GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_clip_volume_hint GL_EXT_debug_label GL_EXT_debug_marker GL_EXT_depth_bounds_test GL_EXT_draw_buffers2 GL_EXT_draw_range_elements GL_EXT_fog_coord GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXT_framebuffer_object GL_EXT_framebuffer_sRGB GL_EXT_geometry_shader4 GL_EXT_gpu_program_parameters GL_EXT_gpu_shader4 GL_EXT_multi_draw_arrays GL_EXT_packed_depth_stencil GL_EXT_packed_float GL_EXT_provoking_vertex GL_EXT_rescale_normal GL_EXT_secondary_color GL_EXT_separate_specular_color GL_EXT_shadow_funcs GL_EXT_stencil_two_side GL_EXT_stencil_wrap GL_EXT_texture_array GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_s3tc GL_EXT_texture_env_add GL_EXT_texture_filter_anisotropic GL_EXT_texture_integer GL_EXT_texture_lod_bias GL_EXT_texture_mirror_clamp GL_EXT_texture_rectangle GL_EXT_texture_sRGB GL_EXT_texture_sRGB_decode GL_EXT_texture_shared_exponent GL_EXT_timer_query GL_EXT_transform_feedback GL_EXT_vertex_array_bgra GL_IBM_rasterpos_clip GL_NV_blend_square GL_NV_conditional_render GL_NV_depth_clamp GL_NV_fog_distance GL_NV_light_max_exponent GL_NV_texgen_reflection GL_NV_texture_barrier GL_SGIS_generate_mipmap GL_SGIS_texture_edge_clamp GL_SGIS_texture_lod GL_SGI_color_matrix

Graphics: AMD Radeon Pro 5300M 4 GB
and Intel UHD Graphics 630 1536 MB

MichaelAtOz, 2020-10-24:
I'm not a Mac guy, but try setting Edit/Preferences/Advanced/Force-Goldfeather, it often helps with Intel graphics.

thehans, 2020-10-25:
That is exactly what thrown together mode should look like btw.
Look in the "View" menu and make sure that "Preview (F9)" is checked and not "Thrown Together (F12)", as it may have been set accidentally.

t-paul, 2021-01-05:
Closing, no feedback.

closed 2021-01-05 by t-paul