openscad/openscad #3469: incorrect modulo

DanielYWoo, 2020-10-26:
The rowNum is not an integer here, so it should not be allowed to modulo. However there is no compliation error and the result is very interesting.

code
```
ballDiameter = 0.8;

// --------------- pillars ------------------
pillarStart = 80;
pillarHeight = ballDiameter * 3.5;
for (row = [pillarStart : pillarHeight : 135]) {
    rowNum = (row - pillarStart) / pillarHeight;
    echo(rowNum % 2);
}
```

result
```
1
2
1
2
...
```

t-paul, 2020-10-26:
OpenSCAD has no integers. This is just following the C++ implementation of [fmod](http://www.cplusplus.com/reference/cmath/fmod/).

The 2 is simply the result of rounding, e.g. `echo(5.999999 % 2);`.

If you want to guard against floating point rounding issues you could use for example `floor(rowNum)`.

closed 2020-10-26 by t-paul

jordanbrown0, 2020-10-26:
A few additional comments...

Modulo is perfectly well-defined for non-integers.  `a % b` is `a - floor(a/b)*b`.  Intuitively, if you take `a` and you subtract away the largest integer multiple of `b` that is still less than or equal to `a`, what you have left is the modulus.

The value of `rowNum` is mathematically an integer.  `row` is `n*pillarHeight + pillarStart`, for `n` from 0 to some value.  `rowNum` is that, less `pillarStart`, divided by `pillarHeight`, and so is n.

However, computer arithmetic is not mathematics, and in particular many division cases do not result in exact results.  (If you don't understand why this is, try calculating 1/3*3 on paper.  1 divided by 3 is 0.333... and no matter how many digits you use, when you multiply by 3 you get 0.999... rather than 1.  It's a similar concept for computers, but in base 2 rather than base 10.)

It is illuminating to change the `echo` to
```
    echo(rowNum - 2);
```
The results are
```
ECHO: -2
ECHO: -1
ECHO: -2.22045e-15
...
```
It seems that the 2 is actually about 1.999999999999998.  Modulo 2 is the same, because it's less than 2.  Displayed, it gets rounded back up to 2.

DanielYWoo, 2020-10-27:
Many thanks to both of you. OpenSCAD is easy to use so people like me often skip the manual. ;-)