openscad/openscad #3470: Modifying STL render problem

cbldwn, 2020-10-26:
Hello,  I’m trying to modify an STL so I can change some text on it. I’m able to import the STL, render it, and export it no problem but as soon as I add a cube to fill in the lettering and try to render it my STL I’m trying to modify disappears and I’m left with just the cube, yet the preview works fine. Any help would be appreaciated. 

Thanks,

cbldwn, 2020-10-26:
I forgot to mention I’ve run also run the STL through mesh mixer and netfabb with no change.

t-paul, 2020-10-26:
I'm sorry, but I don't see how anyone can help based on that information. Please add at least a link to the STL or the STL itself, the error message and what OpenSCAD version you are using.

cbldwn, 2020-10-26:
@t-paul   Im not sure how to add an stl on here, the link no longer works on thingiverse, its a minions chest piece. 

here is my code

    translate([-9,-7,0])
    rotate([0,0,.005])
        import("C:/minion_chess pawn (repaired).stl", convexity = 2,center=true);
translate([-5,-11.8012,13])
    //color("green")
        cube([10,2,10]);

and here is what the console says when I hit F5 the models shows in preview, Then F6 and I'm only left with the cube.

Parsing design (AST generation)...
Saved backup file: C:/Users/Craig/Documents/OpenSCAD/backups/unsaved-backup-hrROpdDU.scad
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 11
Geometry cache size in bytes: 69831304
CGAL Polyhedrons in cache: 2
CGAL cache size in bytes: 22688
Compiling design (CSG Products normalization)...
Normalized CSG tree has 2 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

Parsing design (AST generation)...
Saved backup file: C:/Users/Craig/Documents/OpenSCAD/backups/unsaved-backup-hrROpdDU.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h Line: 329
Geometries in cache: 13
Geometry cache size in bytes: 93089048
CGAL Polyhedrons in cache: 3
CGAL cache size in bytes: 34032
Total rendering time: 0 hours, 0 minutes, 41 seconds
Top level object is a 3D object:
Simple: yes
Vertices: 8
Halfedges: 24
Edges: 12
Halffacets: 12
Facets: 6
Volumes: 2
Rendering finished.

cbldwn, 2020-10-26:
OpenSCAD version 2019.05

cbldwn, 2020-10-26:
I made a drop box account to upload the file, here is the link.
https://www.dropbox.com/s/svtgqvkscf3l7hc/minion_chess%20pawn.stl?dl=0

nophead, 2020-10-27:
The STL has degenerate triangles and self intersections. OpenSCAD can only
handle manifold STL without degenerate triangles.

cbldwn, 2020-10-27:
> The STL has degenerate triangles and self intersections. OpenSCAD can only handle manifold STL without degenerate triangles.
> […](#)

By any chance do you know how I can repair it? Im pretty sure at least 1 person has managed to edit it with openscad to change the arms and added hair using the hairizer scrip. (https://www.thingiverse.com/thing:4570022)

MichaelAtOz, 2020-10-27:
I don't know how missed it in Netfabb:

 

There is lots of damage behind the shield, whoever added that didn't do a good job. Also between
fingers in both hands.

 

That is not easy to fix.

 

 

  _____  

 

<https://github.com/notifications/beacon/AAPBVYI5IVC54AQB4RZN6SLSMXGFBA5CNFSM4S72PKRKYY3PNVWWK3TUL5
2HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFK4S6NY.gif> 

 

This email has been checked for viruses by AVG.
https://www.avg.com

cbldwn, 2020-10-27:
Thats quite a bummer, I seen someone else changed the arms and a bunch of other things including making them larger because they had issues printing them. I’m not really a fan of all the changes they made and had already printed the first half with no problem so was hopping this would of been something I could do. Any recommendations?

MichaelAtOz, 2020-10-27:
Well that image in the email worked well...

Manual post...

> I don't know how missed it in Netfabb:
[image]

Here's another:
[image]
(red lines Self-Intersections)

MichaelAtOz, 2020-10-27:
> 
> Any recommendations?

As I said, not easy. Repair is an art you should practice, google 'self intersection' & 'non manifold'.

cbldwn, 2020-10-27:
So I spend most of the day trying to figure out how to fix the model in blender but didn’t have a ton of luck as the version I had was newer and I couldn’t find all of the tools I needed. Ended up giving up with blended and decided to try converting to solid to see if I could eliminate some of the internal intersections causing problems. To my surprise it actually worked, but I had a couple blemishes in the shield from where I deleted the G. I then decided go back to the original model and generate face groups. I then deleted all the faces in the areas I was getting the blemishes and then I converted to a solid. I then exported it out of mesh mixer and into openscad and to my surprise it actually worked. Considering what I was expecting it was a very easy fix. Thanks for the help.

cbldwn, 2020-10-28:
Ok so I'm able to modify the stl and render it for export to 3d print but I noticed a small problem with the Hairizer scrip that I'm using off thingiverse and was wondering if anyone might be able to help me fix it. looking at the 2 sides I noticed the left one tapers in and causes issues with the inner most hairs as the wall doesn't print properly. Would anyone by chance know what I have to change to fix it? Script is pasted below.

// Hairizer 1.1 By Magonegro (10-MAY-2017)
// Small script to give a hairy look to an existing .STL file
/* [Hair radial count] */
RadialN=6; //[1:32]
/* [Contour wall, 0 for none] */
// This is in case you want a surrounding cilinder to keep hair tight, keep this greater than nozzle diameter
WALLThickness=0.6; //[0.0:1.2]
/* [Cylinder height] */
// Your .stl should be included in the cylinder
CHeight=50;
/* [Cylinder radius] */
// Your .stl should be included in the cylinder
CRadius=20;
/* [Hair Z-Start Height] */
HZStart=40.0;
/* [Printer Layer Height] */
LayerHeight=0.2;
/* [Printer nozzle size] */
HairWidth=0.4;
/* [Cylinder angle] */
CYLAngle=340;
/* [Cylinder angle offset] */
CYLAngleOffset=-80.2;

/* [Filename of .STL] */
STLModel="Minions Pawn.stl";
/* [hidden] */
Layers=floor((CHeight-HZStart)/LayerHeight);
Angle=360/RadialN;

module cyl_pie(radius, deg)
{
    degn = deg % 360;
    rotate([0,0,CYLAngleOffset])
    difference()
        {
        cylinder(r=radius+2,h=CHeight*2,$fn=100);
        if (degn > 180)
            intersection_for(a = [0, 1, 180 - degn]) 
                rotate([0,0,a])
                    translate([-radius, 0, 0]) cube(CHeight * 2);
        else
            union() 
                for(a = [0, 1, 180 - degn])
                    rotate([0,0,a])
                        translate([-radius, 0, 0])
                            cube(CHeight * 2);
        }
}

// Creates the surrounding hairy cilinder for the model
module Hair()
{
difference()
    {
    union()
        {
        for (j=[0:Layers])
           {
            Twist=Angle/(RadialN)*j;
            rotate([0,0,j+Twist])
                translate([0,0,HZStart+j*LayerHeight]) 
                for (i=[1:RadialN])
                    {
                    rotate ([0,0,i*Angle])
                    translate([0,-HairWidth/2,0])
                    union()
                        {
                        cube([CRadius,HairWidth,LayerHeight]);
                        translate([CRadius,HairWidth/2,0])
                            cylinder(r=HairWidth/2,LayerHeight);            }
                        }
            }
        if(WALLThickness>0)
            {
            difference()
                {
                cylinder(r=CRadius+WALLThickness,h=CHeight);
                cylinder(r=CRadius,h=CHeight);
                }
            }
        }
    cyl_pie(CRadius,360-CYLAngle);
    }
} // module

union()
{
   // We simply sum the grid to the model
   Hair();
   translate([.25,0,0]) // Center your model in the grid, just in case
   *import("C:/M Minion Pawn.stl", convexity = 2,center=true);
}

MichaelAtOz, 2020-10-28:
Hopefully someone can help you, but GitHub is for OpenSCAD development/bugs, not for general .scad programming questions.
Please ask such questions on the [Forum](http://forum.openscad.org/).

closed 2020-10-28 by MichaelAtOz

cbldwn, 2020-10-28:
I realize that, wasn’t sure if it’s a bug in openscad or in the code tho. Not sure where else I could have asked.