openscad/openscad #3472: incorrect rendering

vuvova, 2020-10-31:
This works:
```
LL=100;
W=38;
module nut() {
  module twosides() { square([LL,W], center=true); }
  intersection() {
    twosides();
    rotate([0,0, 60]) twosides();
    rotate([0,0,120]) twosides();
  }
}
linear_extrude(4)
difference() {
  circle(d=60);
  nut();
  translate([0, LL/2+W/2-0.0001]) square([LL,LL], center=true);
}
```
but if you remove `-0.0001` the object is rendered incorrectly, solid, without a hole

t-paul, 2020-10-31:
This is a caching issue, I'm almost sure we have an issue for that somewhere. It works with just F6 (or Design->Flush Caches + F6).

vuvova, 2020-11-01:
It doesn't seem to work with just F6 (just = directly after opening the file) or with Design->Flush Caches. At least not with the version 2019.05.

Scopeuk, 2020-11-14:
This appears to come in during the linear extrude.
preview appears to show the shape is correctly derived in two dimensions however running a render results in a solid shape, a circle with a missing edge) which matches the extruded shape, possibly due to the 0 width edge closing the shape(someone with more knowledge of the pipeline may be able to help here).
substituting the -0.0001 for a +0.001 also produces a valid shape unfilled which then exports.
-0.0001 (2D)
[image]
0 (2D)
[image]
+0.001
[image]

kintel, 2024-03-31:
Status as of 2024-03-24: Works now!

closed 2024-03-31 by kintel