openscad/openscad #3474: Default included library

espr14, 2020-11-01:
OpenSCAD has very few primitive solids. There is MCAD library already included in OpenSCAD installation but it is not automatically included in the environment. Therefore, very few people use it. Could you make the inclusion automatic?

jordanbrown0, 2020-11-01:
Please don't.  Automatically including libraries pollutes the namespace, making things harder for those who do not use this particular library.  (Including, say, those who might want to use an experimental or alternate version of it.)  Any *other* library that wants to supply, say, a triangle or a gear would have to use alternate names for its modules.

By all means make it more prominent in the documentation - which would be necessary anyway, for people to know that the functionality is available and how to use it.  It is that knowledge that is the barrier to adoption, not the effort involved in typing the "include".

espr14, 2020-11-01:
1. cube, sphere, cylinder also pollute the namespace so let's remove them. This is about making some ordinary solids to be standardized and easy and flexible to use.
2. You cannot use two versions of the same library already. If you want to use different version of library or same name of the module, you just (re)move it from the installation directory. Or you use different version of OpenSCAD.
3. If you have powerful library without bugs you don't need different one. And if you need, you can rename modules that you need.
4. The barrier is the hell of 1000 libraries that provide a triangle module.

jordanbrown0, 2020-11-02:
Q:  MCAD appears to include somewhere in the neighborhood of 30 individual packages.  The documentation only goes into details on MCAD/regular_shapes.scad and MCAD/involute_gears.scad; the others are only listed.  Which of those are included in this request?

espr14, 2020-11-02:
The goal is to start with something. You can pick one:
1. Remove packages that don't have documentation.
2. Add documentation over time (build-in modules also don't have complete documentation).

jordanbrown0, 2020-11-02:
Does any major language designed in the last 30 years pull in more than a minimal library by default?  Java doesn't.  Python doesn't.  I don't think Rust does.  Even C, a namespace disaster, doesn't force you to use its standard library.

Considering your points in order:
(1) Although `cube`, `sphere`, and `cylinder` can be derived from `polyhedron`, and `cube` and `cylinder` from `linear_extrude`, doing so is, especially for `sphere`, significantly more complex than creating a triangle using `circle($fn=3)` or a cone using `cylinder(d1=10,h=10,d2=0)`.  Also, there is a legacy aspect.  If we were starting from zero, indeed perhaps they should have been in a library requiring a specific `include`.  But they weren't.  That doesn't set law for all to come.

(2) You don't need to modify your installation directory to use an alternate library.  I just copied MCAD into a work directory, modified it, and now a program in that directory gets my modified version of MCAD.  Or I could have an MCAD2 in the installation directory with the same set of files, modules, and functions, and refer to them instead of MCAD.    Or BOSL2, which overlaps MCAD.  Today these can all peacefully coexist, with the root program saying which one (if any) to pull in.  With this proposal, what will the status be of BOSL2's `gear` be?

(3) "Powerful" and "without bugs" do not coexist.  How would an end user rename modules in, say, BOSL2, to avoid conflicts with the now-built-in MCAD?  By modifying BOSL2? That doesn't seem sustainable.

(4) I'm baffled by why we need even one library that provides `triangle`.  But indeed there is a possible hell there.  But why does it happen?  Is it because MCAD's `triangle` isn't in the default namespace?  Is it because MCAD doesn't get prominent enough positioning in the documentation?  Is it because MCAD's `triangle` is not flexible enough for real use?  Is it because it is so cluttered with the trivial (`triangle`) and the specialized (`motors.scad`) that the general-and-tricky (like `sweep` and Bezier curves), the places where you really need a standard library, are lost?  Is it because modifying MCAD is a heavier-weight process than writing your own library?  I don't know, but I don't think that it's because you have to type an `include` to use it.

espr14, 2020-11-02:
Great point. C is very old so everything has to be included as a library. Still, they standardized STD library and made it a part of the language somehow. Now it has a lot of documentation, so few bugs and support from different compilers. Python does not need to include so much because it has a lot of features already built-in (set etc.). Why they didn't put it in the library as an optional thing?

Every big change in a programming language destroys compatibility to some extent. You always have to accommodate old code to new versions.

1. This is a matter of decision where to put the boundary what is built-in, what is supported library and what is left to users to create.
2. Like this now: `cube(); module cube() circle(1);`
3. Supported library gets more attention, more feedback, more documentation than unsupported (third-party) one. So it can coexists.
4. If we don't need triangle, why we need cube? It's not about the `include` but about something more than just a cube. Why the intersection of e.g. MCAD and BOSL2 is not in OpenSCAD already? Why people had to create the same twice (and each time differently)? Because there's no such a building block.

t-paul, 2020-11-02:
I'd be ok with shipping some kind of standard library providing extended "primitives" and/or "functions". The problem here is that there's no such thing at this time. It would need people who maintain both code and documentation.

MCAD is special for legacy reasons. Maybe we need to cut the default MCAD at some point to give the DEV version a chance which is slowly cooking in the MCAD repo. The current way of trying to develop new stuff while being compatible with current MCAD makes development even more difficult compared to having a MCAD2 and letting MCAD go into archive mode.

So far there's a couple of great libraries which tick both the maintained and well documented boxes, but those have lots of other functionality (like mechanical parts) which go far beyond a standard library. Fixing those to OpenSCAD releases is not a good option as that limits ways how those libraries can evolve.

I'll leave the issue open a bit for further discussion, but this is not something that can be solved by the OpenSCAD team (at it's current size). I'm happy to help if someone would be interested in getting a library project started. There's a couple of tickets with suggestions of features that could/should be included. Maybe even collecting those into a wiki page would already help shaping the scope of a possible future standard library.

jordanbrown0, 2020-11-03:
@espr14  I think you've got the evolution of languages backwards.  Historically - BASIC, FORTRAN - you didn't have to say anything to get access to library functions.  You just said `abs(x)` and it worked.  Original C was half-and-half; you only had to declare a library function if it returned something other than an integer.  More recent languages (ANSI C, C++, Python, Java) require that you explicitly declare what library functions you want to access using `include` or `import` directives.  The movement has been **away** from having "standard" functions be part of the core language, and **toward** having them be in standard libraries that are always available but need not necessarily be included in the program's namespace.  In C++, `printf("hello world")` is an error if you haven't included `stdio.h`.  In Python, if you want to get access to the program's command line arguments, you have to `import sys`.

OpenSCAD follows the more recent pattern:  MCAD is installed by default, but if you want to use its features you must explicitly `include` an appropriate file.  It is in a sense "part" of overall language, but as in other modern languages you must explicitly say that you want to use it.

And that's exactly right.  We could include an MCAD2 and it would peacefully coexist with MCAD, with older programs including MCAD and newer programs including MCAD2, and no problem with confusion or possible conflicts.

juanccilleruelo, 2020-11-03:
I think a good solution can be the creation, optionally, of a Project file.
In this one, are going to be listed the libraries that are part of the 
project, not his files.

After, in each drawing file, we need to include the individual 
references to the files used, but without the complete path, because it 
is in the project file.

Is like Pascal evolution to Delphi.

When you are going to open a file, you can open a project or a file.

If you compile externally, at the command line, you can also compile the 
project or the file.

Of course, if the file is created using a project file and is compiled 
without it, is not compilable because don't know the paths to the 
libraries involved.

I think this is a solution that does not modify very much the current 
style of OpenSCAD projects.

nophead, 2020-11-03:
You don't need to use the complete path to library files. You can put those in the OPENSCADPATH environment variable.

Putting absolute paths in a project file would make in non-portable between different machines.

Rather than a project file you can simply have a normal openscad file that you include in every other file that includes the library files needed.

juanccilleruelo, 2020-11-04:
The application I am developing, integrates OpenSCAD inside. This means, 
my application generates the instructions and save it into a file. 
After, my application call OpenSCAD with this file as parameter.

Are you sure that it's better use OPENSCADPATH to indicate where the 
source of the libraries?

Can I send the OPENSCADPATH as parameter in the call to OpenSCAD?

Can I have a standard type of call for multiple operating system? 
(currently I need Windows and MacOS X).

Thanks!!!

nophead, 2020-11-04:
Yes you can pass environment variables when you start another process. 

It needs semicolon separators between paths for Windows and colons for Linux and MacOS I think. In Python the path separator is available a constant ```os.path.pathsep```.

Not sure you should be setting it though as normally the users decide where they want the libraries and set it to configure OpenSCAD for their machine.

closed 2024-11-23 by jordanbrown0