openscad/openscad #3479: Package manager: Discussion

HKalbasi, 2020-11-09:
There were tries to implement a package manager, Or something with same goals (#2377, #2384, #3124, #1913, ...) But they were left without result. I open this issue for a place to solution-neutral discussion and as a tracking issue for package manager feature.

# Goals
1. **Isolated libraries**: Every project needs its own libraries, and two projects may depend on different versions of library. Or there maybe similar version issues between two dependencies that have a common nested dependency.
2. **Discoverability**:  A central registry provides an easy place to look for existing packages. Currently it is a "library" part in the website, but it has very limited features and is unscalable. A package manager, combined with tagging, makes it possible for a registry to provide ecosystem-wide information, such as a list of the most popular or most-depended-on packages.
3. **Micro libraries**: Package manager will make it easy to reuse parts between multiple projects, without copy pasting files and loss of sync. It also makes possible to publish micro libraries for other peoples. Currently, libraries are enormous and general and this means we should repeat ourselves and other people for specialized works.

# Solutions
* **A simple package manager**: Something similar to node `npm`, ruby `gem`, rust `cargo`, ... For example, `scadpm`
  * Pros
    * Simple, battle proven solution.
  * Cons
    * Huge development cost.
    * Hosting cost of a service
* **An alternative module path resolve method**: An alternative to global library folder, for each project. Something like `node_modules`. Solves 1 and 3 and opens a door for future developments.
  * Cons
    * It destroys the way back. With this implemented, Some of other options will be infeasible, And solution for 2 maybe missed.
* **Use a ready package manager**: Current package managers are agnostic to what is stored into them. For example, We can use `npm` as openscad package manager, which is the case for many scripting languages (though, most of them are closely related to js).
  * Pros
     * Little development cost
  * Cons
     * It destroys openscad independence.
* **Http import and use**: Add support for some statement like `include<https://github.com/xyz/abc/folderA/fileB>`
  * Pros
    * Little development cost
    * Very close to current state, It is a small, backward compatible change, relative to others.
  * Cons
    * Web pages die sooner or later. Hard coding urls into source files and repeating them many times is not a good idea, Specially for changing them. A package manager can install packages from urls, and keep other useful metadata like content-hash, so in source files we can only mention the name of packages which is cleaner.
    * Goal 2 remains unsolved.

I'm not native English speaker, Sorry for my English errors and please fix them. And Add more (Or tell me to Add) solutions and goals if I missed them.

alexose, 2020-12-31:
A package manager would be a hugely important milestone for OpenSCAD.  I put $50 towards the bounty on this one.

Working package management would unlock a more cohesive development experience.  It would encourage developers to publish good APIs and help bubble the most useful libraries to the top.  Solutions to common problems would become standardized, and more functional-style programming would emerge.

As an OpenSCAD and Node.js user, It's my belief that an approach similar to `npm` would work for this project.  And while package management isn't quite solved problem, virtually every platform shares the same main characteristics:  Version controlled packages served from a central hub, with a web interface, with links back to the project homepages (generally GitHub or GitLab).

I don't think this project should try to reinvent the wheel with regards to decentralization; If anything, centralizing the packages is paramount to achieving the discoverability goal mentioned above.

Irev-Dev, 2021-06-06:
Thanks @HKalbasi, really well laid out discussion of the problem.

Honestly I'd be okay with **Http import and use** as a stop gap, I don't think broken URLs would actually be that big of a problem since most of the community would probably move to wards a defacto standard of using Github releases anyway and that should make for robust URLs.

But Also like **An alternative module path resolve method** as something that could continue to be built on.

@t-paul mentioned to me https://github.com/entropic-dev/entropic which is supposed to be an agnostic package manager, but it doesn't seem very active these days.

fearthecowboy, 2021-11-11:
O_O.

This is a place in OpenSCAD that I can contribute. 

I have a lot of experience in package management, and I'm actually working on something right this very moment, which with very little effort could hit most of the important points. With very minor additions to OpenSCAD itself, I think we could hit a very high level of joy.

Step zero is ... finishing the process of actually open-sourcing the current code I have.  I'm building it for my employer, but it's intended to be released and MIT licensed. 

Even if I just hard fork it and tailor it to suit OpenSCAD only, it'll be really fast to put something together. 

I can't say more at the moment, but ... I'm not very far out from being able to do so.

t-paul, 2021-11-11:
That sounds awesome. I do have some rough ideas what would be nice to have (e.g. ideally not requiring another scripting language to be installed), but there's no fixed plan. So we should be able to find some common direction. This is certainly a feature that would be welcomed by lots of users.

If you want to talk, feel free to drop by our IRC channel [#openscad](https://web.libera.chat/?channel=#openscad) at [Libera.Chat](https://libera.chat/).

fearthecowboy, 2021-11-12:
Good news; I finally got the signoff to be able to publish the stuff I'm working on from my Employer 😄 

Gimme a couple days and I'll write up something that talks about the approach and how it works.

fearthecowboy, 2021-11-20:
*Sorry for the delay; I had written out what I wanted to say, and then forgot to save it before closing a screen, and then got real busy this week.*
<hr>

Up Front: I'm on the Embedded/Cross-platform C++ team at Microsoft. Don't let that bother you. 

We've been building a tool for Embedded C++ developers to help acquire tools, sdks, libraries, and other 'artifacts' needed for their embedded development, and then configure the development environment to use them.

Instead of writing a somewhat typical "package manager" that requires repackaging of existing content, curation of large repositories, and some kind of walled garden, I've designed the whole thing to instead take advantage of the heterogeneous nature of the ecosystem and work with everything where it already is.

The tool is called `VCPKG-CE` which is a component of `VCPKG` (a package manager for building C++ libraries) -- but does not have a dependency on `VCPKG`; it can be standalone without issue.

`CE` stands for 'configure-environment' and can be used for entirely different things.  I'll refer to it as just `CE` for now.

**Here's the idea:**

Think of it as a manifest-driven desired state configuration for projects.

Instead of 'packages', we have `artifacts`. `Artifacts` can be any tools, libraries, sdks, etc. The physical artifacts are an archive -- .zip files, .tar.gz/bz2 files,... or even a *git repository*.
More importantly, artifacts are **not** repackaged, rebuilt, rehosted, or modified to be used with the tool.

All we write is a metadata file (a `manifest`) that references the `artifact`s that we want to use. 

Instead of requiring a single centralized registry that requires ongoing effort to curate and maintain, we make a `registry` easy to create. Just make a repo of some manifest files, and generate an index. These can be local directories, or hosted on the web (like a GitHub repository). No service is necesarry at all. 

If you do want a centralized repository, it can be created with a simple github repo, and a couple of github actions to keep the index up-to-date. 

Benefits of `CE`:
  - Cross-platform (Windows/Linux/OSX).  
   `CE` is written in TypeScript and requires node.js to run -- BUT -- don't worry. 
    The script that installs it takes care of that and brings along an isolated copy of node, and doesn't require to you even notice that it's there. You should never need to see it or worry about what version it needs.

  - No admin/root access requires ever. At All. Even the installer. The entire thing is designed to be run with just regular use permissions. 

  - Bootstrapping is super easy. a single 'one-liner' is available for all the shells, and you can run it in on your desktop, in CI/CD, or anywhere you need to use it. No elevated permissions used at all.

  - Works from a blank system. No prerequisites or dependencies that it doesn't manage all by itself.
    *(On Linux IIRC, the assumption is that glibc is used. I probably would need to tweak a couple things for a MUSLC based distro )*

  - Does not make any changes to your system configuration. Does not modify the system PATH/$PATH. 
    Entirely constrained to the shell [`Bash`/`Zsh` (really,any Posix compatible shell should work), `PowerShell`,`Pwsh`, `CMD`] window you are working in.

  - Today, it's a command-line based tool, but we're looking down the road for integration into things like VSCode, VS, etc. 

  - Isolation: Every artifact is installed into a distict folder, and you can have multiple versions of the same artifact installed concurrently without any concern or conflict.
   
  - Projects: a project `manifest` declares the artifacts/versions that a given project needs. When you want to work on that project, you open up your preferred shell and `activate` the project.  
    That will acquire the specific artifacts that project needs (if they aren't already acquired) and configure the shell's environment to use them

    >In C/C++ terms, the `artifact` `manifest` expresses the paths, binaries, include files, environment variable, and other settings that an artifact exposes.

    When the `artifact`s are pulled down, they get unpacked into a (user-global) folder, and we update the environment so that the binaries are on the PATH, and set any other variables that are necessary. 

    `Artifact`s are never dumped into the project folder, they're *referenced* from the (user-global) folder.

  - Restoration is always trivial. Run the bootstrap script and call `activate`.  
    Think something got broken somehwere? Blow away the whole installation folder and re run the bootstrap script and 'activate'  
    No state is difficult to restore at all. 

  - `Artifact Metadata` is incredibly simple to write. A simple file with name/version/description and references to the artifact itself. That's it. 
    You can even specify different conditions that affect what file/settings are used for the artifact. 
    >In C++, this lets you specify a different .zip or .tgz file for a tool depending on platform/architecture/etc, so that 
    and artifact can be use on multiple platforms without changing the project manifest.

    Artifact metadata is in YAML/JSON format. (I prefer YAML simply beacuse it is simpler to edit manually, and doesn't suffer from particularly painful punctuation problems perpetuated purposefully by JSON or XML)

[... cut ...]
[... 78 lines cut ...]
    
requires:
  openscad:openscad: "* 2021.1.0"

```
and
> `sample.scad`
``` scad
use <MCAD/boxes.scad>;

translate([180, 0, 60]) rotate([0, 90, 90]) roundedCube(size = [60, 180, 180], r = 5, sidesonly = true);
```

After that, on a completely blank computer:

``` powershell
iex (iwr -useb aka.ms/install-ce.ps1)

ce activate

openscad.exe ./sample.scad
```

A video of it in action:
https://user-images.githubusercontent.com/133736/142738644-bcdca593-0553-43b3-9571-69dc6edb7108.mp4

t-paul, 2021-11-21:
I've not seen the link, so for completeness: https://github.com/microsoft/vcpkg-ce

As mentioned before, it would be nicer to not have additional dependencies. But I suppose having something that works is much more useful to get started with this topic. The project that had given some hope of being an option for OpenSCAD `entropic` would have been nods.js based too.

Access to decentral storage without need for a central registry seems like a big bonus. Having a central index for discovering libraries would be nice, but it might even be possible to have that completely separate.

It's not yet clear to me how the installed stuff looks like, it may take some time to go through some of the examples. What I'm not seeing is something most package manager are using now, a distinction between a requirements and a lock file. The `pipenv` [talk by Kenneth Reitz](https://youtu.be/GBQAKldqgZs?t=592) gives some nice insight in how package management can be made more user friendly. How is that handled in `vcpkg-ce`?

Another question would be how customizable the behavior is. For example it could be useful supporting Thingiverse as source, which is obviously a very OpenSCAD specific case.

Are there any features built-in that allow execution of commands? From security perspective it would be nice not having to worry about random commands being executed, e.g. when used in CI pipelines or platforms like cadhub.

fearthecowboy, 2021-11-21:
The lock file and the project manifest are one in the same. The version ends up being recorded as `<requested> <fulfilled>` so if you just add a reference without asking for a specific version, it'll put something like `* 1.0.0` . Eventually the tool will allow you to 'upgrade'  an artifact by finding the latest version that matches the 'requested', otherwise, on activate, it restores the 'fulfilled' version.

The intent it to support alternative sources and artifacts that can be crafted from other existing metadata (ie, for C++, an 'arduino' source that can point to existing feeds/libraries of stuff) .. making one for thingiverse would not be all that tricky. 

And, no there is ZERO scriptability for artifacts; zero additional code can get executed. The artifacts have to be 100% declarative -- anything else is just inviting a security nightmare. No random things get executed. 

As it is, this is *very* early, probably by mid-December, I'll be close to having something that has some additional error-handling in, right now, it fails pretty fast off the 'happy-path'.

t-paul, 2021-11-21:
With both requested and fulfilled in one file, how is it possible to see transitive dependencies? E.g. I declare library A in the manifest as this is the direct dependency, but A depends on B. So the lock part would need to list the B version. But I would not like to see this promoted to a declared dependency.

I found https://github.com/perpetual-motion/openscad-registry and download of MCAD with just `"*"` produced a `OPENSCADPATH` entry ending in `:undefined`.

> And, no there is ZERO scriptability for artifacts; zero additional code can get executed. The artifacts have to be 100% declarative -- anything else is just inviting a security nightmare. No random things get executed.

Perfect :smiley: 

I'll try to toy a bit more with this, maybe we can get one or 2 libraries to include the manifest for testing...

fearthecowboy, 2021-11-21:
It's not quite stable yet; there's gonna be a lot of little places it breaks down; In a few more weeks it'll be a lot more consistent. 

(hence the `undefined` on the path.)

I've got a whole lotta stuff coming up regarding transitivity. It will be possible to lock the nested dependencies; but ... gotta walk before flying. 

if you run into issues, or things I should look at, post them here (not in the vcpkg-ce repo) for now; I don't want to raise too much fuss in the project until I get past a bit more and not freak out some folks on the team 😄

t-paul, 2021-11-21:
How do I regenerate the `index.yaml` for https://github.com/openscad/openscad-registry? It's not clear to me how the `ce regenerate` works.

t-paul, 2021-11-21:
I'm testing via a docker container with `Dockerfile`:
```
FROM ubuntu:rolling

ENV DEBIAN_FRONTEND=noninteractive

WORKDIR /openscad

RUN apt-get update

RUN apt-get upgrade

RUN apt-get install -y --no-install-recommends curl vim apt-utils ca-cacert ca-certificates openscad

RUN bash -c ". <(curl aka.ms/install-ce.sh -L)"

RUN bash -c 'echo ". ~/.vcpkg/ce" >> ~/.bashrc'
```
```
$ docker build -t vcpkg-ce .
```
```
$ docker run -it vcpkg-ce
```

fearthecowboy, 2021-11-21:
> How do I regenerate 

I think I botched something on Friday. I'll have the regenerate command fixed on Monday. (It's not properly allowing file paths like it should).

fearthecowboy, 2021-11-23:
@t-paul  I updated CE yesterday to 0.6.67 to fix the regenerate command. Now you can just give it the path `ce regenerate /foo/bar` (or the name of a registry in a manifest.)

t-paul, 2021-11-24:
Looks good. I've added some more libraries and a quick readme to https://github.com/openscad/openscad-registry.

@Irev-Dev What do you think, would that work for cadhub?

Irev-Dev, 2021-11-28:
Sorry @t-paul, I'm caught up now.

I do like the sounds of it, seems like it's been kept as simple as possible.

I'm not sure exactly how it would fit in with CadHub only in the sense that I'm still figuring out the backend pipeline, but would be keen to experiment with this one that's sorted.

@fearthecowboy one question I do have is does this handle dependencies of dependancies? will it recognise a metadata yml file within the artifact and grab its dependancies too?

iOmkarNikam, 2024-03-03:
Hello Mentors Team BRL-CAD !

@kintel 

I'm Omkar Nikam ,I am second year student studying Artificial Intelligence and Machine Learning (CSE) from KITCOEK(INDIA).
I am Very Keen Learner , My Expertise Include Python , C , C++ , JAVA , HTML, CSS , JavaScript , PHP .

I'm beginner in Open Source , I'm Passionate for Technology .
I have read and gone through the description , And I am interested to work with BRL-CAD  Projects for GSOC 2024 ,
particularly in ;

Simple Library Manager / Downloader for OpenSCAD scripts ([#57]

I am having good exposure to Machine Learning Concepts .
Please , guide me for next steps !

Sincerely ,
Omkar

kintel, 2024-03-03:
Best next step is to join our IRC channel and start familiarizing yourself with OpenSCAD development topics and discuss ideas: https://openscad.org/community.html#irc

t-paul, 2024-10-28:
You missed the issue linked just above your comment. We do have 95% of an initial package manager which needs some finishing touches and activation. It's a relatively simple start, similar to the Arduino one, but hopefully can get things going eventually.

Status: https://github.com/openscad/openscad/issues/5268
Repo: https://github.com/openscad/openscad-library-manager
First try packaging the client as Snap package: https://snapcraft.io/olman
PyInstaller builds: https://ci.openscad.org/teams/olman/pipelines/olman-pyinstaller-amd64
First (very much not polished) library for testing: https://github.com/t-paul/tp-scad-utils
Support in OpenSCAD: Small patch, only local, I'll PR that
Docu: None yet

Anyone who would like to help moving things forward is very welcome on IRC. If nobody is motivated to help, it will move along based on my personal spare time and motivation which both are under my sole discretion.