openscad/openscad #3482: Fix global library location on MacOS

jottr, 2020-11-10:
According to the [documentation](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Libraries#Library_Locations), globally available libraries must be installed into the `Documents`-folder: `$HOME/Documents/OpenSCAD/libraries`.

1. This pollutes the users `Documents` folder 
2. Does not follow the commonly used MacOS configuration path, see Apple's official [documentation](https://developer.apple.com/library/archive/documentation/MacOSX/Conceptual/BPFileSystem/Articles/WhereToPutFiles.html)

I would like to suggest that openSCAD loads global libraries in the following order on MacOS:

```
$HOME/Library/Preferences/openSCAD/libraries
$HOME/.local/share/OpenSCAD/libraries # match Linux XDG path for portability
$HOME/Documents/OpenSCAD/libraries # for backwards compatibility
$HOME/.openSCAD/libraries # follow common dotfiles convention
```

t-paul, 2020-11-10:
Duplicate of #1524.

closed 2020-11-10 by t-paul

jottr, 2020-11-10:
Sorry for the duplicate.