openscad/openscad #3485: Intersection fails for relatively simple bodies

jpmaterial, 2020-11-18:
When calculating an intersection between bodies, I get an error on render.

One body is a box. The other consists of two shells each of which was created as a merger of two boxes. The latter used to be more coarsely meshed (no problems there) and was then remeshed more finely, which seems to have caused the problems.

Both bodies are imported from .off files. Since no attachments seem to be possible, I'll post them below (~45kB)

This is the offending script:

`intersection(){import("tmpfh1z4mti.off");import("tmpxsmrjc52.off");}`

Body 1:

```
OFF
8 12 0
-1.0125007629 -2.0458002090 -14.9805002213
-1.0125007629 -2.0458002090 14.5354995728
-1.0125007629 0.4541997910 -14.9805002213
-1.0125007629 0.4541997910 14.5354995728
1.4874992371 -2.0458002090 -14.9805002213
1.4874992371 -2.0458002090 14.5354995728
1.4874992371 0.4541997910 -14.9805002213
1.4874992371 0.4541997910 14.5354995728
3 1 3 0
3 4 1 0
3 0 3 2
3 2 4 0
3 1 7 3
3 5 1 4
3 5 7 1
3 3 7 2
3 6 4 2
3 2 7 6
3 6 5 4
3 7 5 6
```

Body 2:

```
OFF
656 1304 0
0.1312000006 0.6000000238 1.9470000267
-3.3519999981 3.4258000851 6.9176001549
-0.7919999957 0.0000000000 -12.4805002213
-0.7919999957 5.0000000000 -11.7965002060
-0.7919999957 5.0000000000 -9.5740995407
-0.7919999957 5.0000000000 -12.0538997650
-0.7919999957 5.0000000000 -8.5272998810
-0.7919999957 5.0000000000 -11.9252004623
-0.7919999957 5.0000000000 -12.4805002213
-0.7919999957 5.0000000000 -7.4805002213
-0.7919999957 5.0000000000 -11.6677999496
-0.7919999957 5.0000000000 -12.3315000534
-0.7919999957 5.0000000000 -12.1826000214
-0.7919999957 5.0000000000 -10.6209001541
-0.7919999957 -5.0000000000 -9.8315000534
-0.7919999957 -5.0000000000 -12.4805002213
-0.7919999957 -5.0000000000 -7.4805002213
-0.7919999957 -5.0000000000 -11.0069999695
-0.7919999957 -5.0000000000 -12.3315000534
-0.7919999957 -5.0000000000 -12.1826000214
-0.7919999957 -5.0000000000 -8.6560001373
-0.7919999957 4.7425999641 -12.3315000534
-0.7919999957 4.7425999641 -12.1826000214
-0.7919999957 2.0531001091 -12.2569999695
-0.7919999957 2.0531001091 -12.1826000214
-0.7919999957 2.2021000385 -12.3315000534
-0.7919999957 4.4853000641 -12.1826000214
-0.7919999957 2.3510000706 -12.4060001373
[... 1908 lines cut ...]
3 533 535 80
3 75 533 80
3 399 535 533
3 65 80 535
3 216 401 214
3 15 216 214
3 396 401 216
3 399 214 401
3 212 217 402
3 396 212 402
3 2 217 212
3 397 402 217
3 536 402 530
3 70 536 530
3 396 402 536
3 397 530 402
3 30 212 216
3 15 30 216
3 2 212 30
3 396 216 212
3 213 217 31
3 8 213 31
3 397 217 213
3 2 31 217
```

rcolyer, 2020-11-20:
Your second input object is actually not manifold.  It throws a degeneracy error from the CGAL library.  I didn't examine the 45kB of values, but from looking at the preview it might be from the two pieces of the upper volumes being distinct volumes, but having zero space between them, giving a zero-volume external space between them.

To test them separately, uncomment only one at a time:
```
union() {
  //import("tmpfh1z4mti.off");
  import("tmpxsmrjc52.off");
  cube(1);
}
```

kintel, 2024-03-31:
Status as of 2024-03-24: Works with Manifold, still broken with CGAL