openscad/openscad #3488: Short cut "Ctrl + P" (Reset Movement) does not work

fe60, 2020-11-20:
OpenSCAD 2019.05 on Windows 10 (2004)

The short cut `Ctrl + P` does not appear in the "View" menu of the application, but in the documentation [1].
Nothing happens when I try to use this short cut.

Needs this to be fixed in the application or should I remove it from the documentation?

[1] https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_User_Interface#View_navigation

t-paul, 2020-11-20:
Hmm, the list in the [github wiki](https://github.com/openscad/openscad/wiki/Keyboard-Shortcuts) says
> Used to be View/Center, reserved for Print

All the "good" keys are already gone, so `Ctrl+Y` maybe?

fe60, 2020-11-20:
I was not aware of the list in the [github wiki](https://github.com/openscad/openscad/wiki/Keyboard-Shortcuts).
If it's about defining an alternate/unused short cut, I propose `Ctrl+Shift+0`.
`Ctrl+0` sets the viewing angle back to the initial view. Adding `Shift` moves the view to its initial position (back to "zero").
With `Ctrl+Shift+V` there is already one short cut related to the adjustment of the viewing position, that uses `Ctrl+Shift`.

tombsar, 2020-11-20:
If you're looking for inspiration, Blender uses `Home` to centre the camera about the origin, `Alt+Home` to centre the camera about the 3D cursor, and `Shift+C` to both move the 3D cursor to the origin and centre the camera about the origin.

t-paul, 2020-11-20:
No, it's just about a default that still works with all the existing shortcuts (and `Ctrl+Shift+0` sounds like a good suggestion). The preferences UI to configure shortcuts will hopefully come soon, it's mostly ready in #3338, but needs review and maybe some minor updates.

MichaelAtOz, 2020-11-21:
Note that wiki is slightly out of date, 2019.05 changes not added (eg F8=3DPrint, new underscore-menu-shortcuts on many View menu items, probably others).

`Ctrl+P` was in 2013.06. 

'Reserved for print' ie in most other apps `^p` is `File/Print`, if ever the editor needs to print the code, or whatever...

Should also try to include `View/Reset View` into the pattern, perhaps `Ctrl+Alt+0`?

Also the underscore-menu-shortcuts, ie `Alt+v` then `t` selects `View/Top` should also be defined.
`Cen_ter` has `n`, perhaps `V_iew All` gets `v`, and `Re_set View` gets `e`?

While looking at `View` menu, I think `Animate` needs a shortcut-key, perhaps `Ctrl+Shift+a`, and `A_nimate` underscore-menu-shortcut of 'a'?

t-paul, 2020-11-21:
Hmm, `Ctrl+Shift+0` collides with `Ctrl+=` which is supposed to be a synonym for `Ctrl++`. But strangely it does not work. I have to check where that was coming from, maybe that's for english keyboard or something? Maybe we can drop that and defer to the shortcut setup GUI.

t-paul, 2020-11-21:
And that came in via #752.

MichaelAtOz, 2020-11-21:
See [Qt doc](https://doc.qt.io/qt-5/qkeysequence.html#keyboard-layout-issues).
[This](https://defkey.com/what-means/ctrl-shift-0) shows a number of programs use `Ctrl+Shift+0` so it should be doable??

MichaelAtOz, 2020-11-21:
On my system `Ctrl+Shift+0` types a `0` in the editor??

MichaelAtOz, 2020-11-21:
I just updated the [GitHub wiki](https://github.com/openscad/openscad/wiki/Keyboard-Shortcuts) with 2019.05 shortcuts visible on menus.

commit 67f6397d (2020-11-25): Set keyboard shortcut for "reset translation" as Ctrl+Shift+0 (fixes #3488).

This removes the special case for editActionZoomTextIn which had defined
an additional Ctrl+= shortcut as this collides with the new one and did
not seem to work anyway (at least collides on German keyboard where =
is Shift+0).
commit b1fb32b5 (2020-11-25): Merge pull request #3498 from openscad/fix-viewport-center-shortcut

Set keyboard shortcut for "reset translation" as Ctrl+Shift+0 (fixes #3488).
closed 2020-11-25 by t-paul