openscad/openscad #3494: 3d preview stays blank

emdete, 2020-11-22:
My scad shows an empty 3d-preview. No objects, no coords, just background (which changes color if the background is changed in the settings).

I get the following output from openscad:

```
GLEW Error: Unknown error
OpenGL Error: invalid operation
OpenGL Program Linker Error:
error: fragment shader lacks `main'
```

I dont know if that is related. I am running "OpenSCAD version 2019.05" on a debian system (version sid). It may well be a problem with the distribution libraries (because i run unstable) but i doubt that. Previous versions ran fine on the same system - but hey, everything is moving in sid.

Do you have any hints on how to track down the problem or even solving it (while i'm an developer i don't know much about opengl)?

t-paul, 2020-11-22:
Please add the info from `Help->Library Info`.

t-paul, 2020-11-22:
Oh, also the latest snapshot AppImage (or Snap package) might be interesting as that uses a new set of shaders.

emdete, 2020-11-22:
```
OpenSCAD Version: 2019.05
System information: Linux 5.9.0-1-amd64 #1 SMP Debian 5.9.1-1 (2020-10-17) x86_64 Devuan GNU/Linux 4 (chimaera/ceres) 8 CPUs 31.24 GB RAM
User Agent: OpenSCAD/2019.05 (Linux x86_64; Devuan GNU/Linux 4 (chimaera/ceres))
Compiler: GCC "9.3.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_71
Eigen version: 3.3.7
CGAL version, kernels: 5.0.2, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.15.1
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.64.3
lodepng version: 20180910
libzip version: 1.6.1
fontconfig version: 2.13.1
freetype version: 2.10.1 (runtime: 2.10.2)
harfbuzz version: 2.6.4 (runtime: 2.6.7)
lib3mf version: 1.8.1
Application Path: /usr/bin
Documents Path: /home/mdt/.local/share
User Documents Path: /home/mdt
Resource Path: /usr/share/openscad
User Library Path: /home/mdt/.local/share/OpenSCAD/libraries
User Config Path: /home/mdt/.config/OpenSCAD
Backup Path: /home/mdt/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/mdt/.local/share/OpenSCAD/libraries
/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
/usr/share/fonts
/usr/local/share/fonts
/home/mdt/.local/share/fonts
/home/mdt/.fonts
/usr/share/texmf/fonts/opentype/public/lm
/usr/share/texmf/fonts/opentype/public/lm-math
/usr/share/fonts/X11
/usr/share/fonts/cMap
/usr/share/fonts/cmap
/usr/share/fonts/eot
/usr/share/fonts/opentype
/usr/share/fonts/svg
/usr/share/fonts/truetype
/usr/share/fonts/type1
/usr/share/fonts/woff
/usr/share/fonts/X11/100dpi
/usr/share/fonts/X11/75dpi
/usr/share/fonts/X11/Type1
/usr/share/fonts/X11/encodings
/usr/share/fonts/X11/misc
/usr/share/fonts/X11/util
/usr/share/fonts/cmap/adobe-cns1
/usr/share/fonts/cmap/adobe-gb1
/usr/share/fonts/cmap/adobe-japan1
/usr/share/fonts/cmap/adobe-japan2
/usr/share/fonts/cmap/adobe-korea1
/usr/share/fonts/eot/font-awesome
/usr/share/fonts/opentype/cantarell
/usr/share/fonts/opentype/font-awesome
/usr/share/fonts/opentype/fonts-hosny-amiri
/usr/share/fonts/opentype/ipafont-gothic
/usr/share/fonts/opentype/ipafont-mincho
/usr/share/fonts/opentype/mathjax
/usr/share/fonts/opentype/roboto
/usr/share/fonts/opentype/urw-base35
[... 175 lines cut ...]
GL_OES_shader_io_blocks
GL_OES_shader_multisample_interpolation
GL_OES_standard_derivatives
GL_OES_stencil8
GL_OES_surfaceless_context
GL_OES_tessellation_point_size
GL_OES_tessellation_shader
GL_OES_texture_3D
GL_OES_texture_border_clamp
GL_OES_texture_buffer
GL_OES_texture_cube_map_array
GL_OES_texture_float
GL_OES_texture_float_linear
GL_OES_texture_half_float
GL_OES_texture_half_float_linear
GL_OES_texture_npot
GL_OES_texture_stencil8
GL_OES_texture_storage_multisample_2d_array
GL_OES_texture_view
GL_OES_vertex_array_object
GL_OES_vertex_half_float
GL_OES_viewport_array
```

Will check the snapshot next. Give me a bit..

emdete, 2020-11-22:
I installed the nightly, it does the same but logs:

```
GLEW Error: Unknown error
OpenGL Error: invalid operation
OpenGL Error: invalid operation
OpenGL Error: invalid operation
OpenGL Error: invalid operation
OpenGL Program Linker Error:
error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shaderOpenGL error start:
 invalid enumerant

OpenGL error render:
 invalid operation

```

t-paul, 2020-11-22:
That's probably the issue:
```
OpenGL Version: OpenGL ES 3.2 Mesa 20.2.2
```
OpenSCAD does not support GLES, I don't know why that happens, it could be a system issue with how Qt is compiled. I don't know if there is a way to force a Desktop OpenGL version, I think Debian has different packages and does not support parallel installation of those (`libqt5gui5` vs. `libqt5gui5-gles`).

emdete, 2020-11-23:
You made my day! A simple command solves the problem:

```
apt purge libqt5gui5-gles
```

it deinstalls the gles one and installs the other. Maybe the debian package should conflict with `libqt5gui5-gles`?

closed 2020-11-23 by emdete

t-paul, 2020-11-23:
Yes, having a conflict makes sense. It's currently not set as the `-gles` packaging is newer than the openscad package. I'll make a note to pass that on to the debian people... :grinning:.