openscad/openscad #3497: Building with cgal-5.1 failing

waebbl, 2020-11-23:
Trying to build openscad-2019.05 against the new cgal-5.1 fails with error

```
x86_64-pc-linux-gnu-g++ -c -O2 -pipe -march=bdver2 -mprefer-avx128 -mvzeroupper -frecord-gcc-switches -Wno-narrowing -fcommon -DST
ACKSIZE=8388608 -fno-strict-aliasing -std=c++11 -std=gnu++11 -I/usr/include -I/usr/include/hidapi -I/usr/include/libxml2 -I/usr/in
clude/lib3mf -I/usr/include/freetype2 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/harfbuzz -I/usr/include/freetype
2 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -DEIGEN_DONT_ALIGN -
frounding-math -D_REENTRANT -Wall -Wextra -Wno-unused-local-typedefs -fPIC -DOPENSCAD_VERSION=2019.05 -DOPENSCAD_SHORTVERSION=2019
.05 -DOPENSCAD_YEAR=2019.0 -DOPENSCAD_MONTH=05.0 -DOPENSCAD_DAY=.0 -DDEBUG -DSTACKSIZE=8388608 -DUSE_QOPENGLWIDGET -DENABLE_DBUS -
DENABLE_JOYSTICK -DUSE_SCINTILLA_EDITOR -DQSCINTILLA_DLL -DENABLE_SPNAV -DENABLE_HIDAPI -DENABLE_LIBZIP -DENABLE_LIB3MF -DENABLE_O
PENCSG -DENABLE_CGAL -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQ
T_NETWORK_LIB -DQT_DBUS_LIB -DQT_CORE_LIB -I. -Isrc -Isrc/ext/libtess2/Include -I/usr/include/qt5 -I/usr/include/qt5/QtPrintSuppor
t -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtMultimedia -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtConcurrent -I/usr/inc
lude/qt5/QtNetwork -I/usr/include/qt5/QtDBus -I/usr/include/qt5/QtCore -Iobjects -I/usr/include/eigen3 -I/usr/lib64/qt5/mkspecs/li
nux-g++ -o objects/src/transform.o src/transform.cc
In file included from /usr/include/CGAL/user_classes.h:21,
                 from /usr/include/CGAL/Kernel/global_functions_2.h:25,
                 from /usr/include/CGAL/Kernel/global_functions.h:23,
                 from /usr/include/CGAL/Cartesian/Cartesian_base.h:22,
                 from /usr/include/CGAL/Simple_cartesian.h:20,
                 from /usr/include/CGAL/Extended_cartesian.h:19,
                 from src/cgal.h:14,
                 from src/cgalutils.h:3,
                 from src/polyset-utils.cc:9:
/usr/include/CGAL/Point_2.h:89:19: error: ‘is_nothrow_swappable_v’ is not a member of ‘std’; did you mean ‘is_nothrow_swappable’?
   89 |     noexcept(std::is_nothrow_swappable_v<Rep>)
      |                   ^~~~~~~~~~~~~~~~~~~~~~
      |                   is_nothrow_swappable
```

followed by many errors of the same type and follow-up errors, when building with c++11. 

Trying to use c++17 instead gives:

```
x86_64-pc-linux-gnu-g++ -c -O2 -pipe -march=bdver2 -mprefer-avx128 -mvzeroupper -frecord-gcc-switches -Wno-narrowing -fcommon -DST
ACKSIZE=8388608 -fno-strict-aliasing -std=gnu++1z -I/usr/include -I/usr/include/hidapi -I/usr/include/libxml2 -I/usr/include/lib3m
f -I/usr/include/freetype2 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/in
clude/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -DEIGEN_DONT_ALIGN -frounding-m
ath -D_REENTRANT -Wall -Wextra -Wno-unused-local-typedefs -fPIC -DOPENSCAD_VERSION=2019.05 -DOPENSCAD_SHORTVERSION=2019.05 -DOPENS
CAD_YEAR=2019.0 -DOPENSCAD_MONTH=05.0 -DOPENSCAD_DAY=.0 -DDEBUG -DSTACKSIZE=8388608 -DUSE_QOPENGLWIDGET -DENABLE_DBUS -DENABLE_JOY
STICK -DUSE_SCINTILLA_EDITOR -DQSCINTILLA_DLL -DENABLE_SPNAV -DENABLE_HIDAPI -DENABLE_LIBZIP -DENABLE_LIB3MF -DENABLE_OPENCSG -DEN
ABLE_CGAL -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_NETWORK_L
IB -DQT_DBUS_LIB -DQT_CORE_LIB -I. -Isrc -Isrc/ext/libtess2/Include -I/usr/include/qt5 -I/usr/include/qt5/QtPrintSupport -I/usr/in
clude/qt5/QtWidgets -I/usr/include/qt5/QtMultimedia -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtConcurrent -I/usr/include/qt5/Qt
Network -I/usr/include/qt5/QtDBus -I/usr/include/qt5/QtCore -Iobjects -I/usr/include/eigen3 -I/usr/lib64/qt5/mkspecs/linux-g++ -o 
objects/src/GeometryUtils.o src/GeometryUtils.cc
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/10.2.0/include/g++-v10/functional:54,
                 from src/ext/polyclipping/clipper.hpp:58,
                 from src/clipper-utils.h:3,
                 from src/clipper-utils.cc:1:
/usr/lib/gcc/x86_64-pc-linux-gnu/10.2.0/include/g++-v10/tuple: In instantiation of ‘constexpr const size_t std::tuple_size_v<Clipp
erLib::ClipType>’:
/usr/lib/gcc/x86_64-pc-linux-gnu/10.2.0/include/g++-v10/tuple:1733:24:   required from ‘constexpr decltype(auto) std::apply(_Fn&&,
 _Tuple&&) [with _Fn = std::vector<std::vector<std::vector<ClipperLib::IntPoint> > >&; _Tuple = ClipperLib::ClipType&]’
src/clipper-utils.cc:147:41:   required from here
/usr/lib/gcc/x86_64-pc-linux-gnu/10.2.0/include/g++-v10/tuple:1251:61: error: incomplete type ‘std::tuple_size<ClipperLib::ClipType>’ used in nested name specifier
 1251 |     inline constexpr size_t tuple_size_v = tuple_size<_Tp>::value;
      |                                                             ^~~~~
```

For more details please see https://bugs.gentoo.org/755842 or the attached build log.
[openscad-2019.05-r3:20201123-202836.log](https://github.com/openscad/openscad/files/5585828/openscad-2019.05-r3.20201123-202836.log)

t-paul, 2020-11-23:
CGAL manual states c++14 compatibility for version 5.1.1 https://doc.cgal.org/latest/Manual/thirdparty.html but `is_nothrow_swappable_v` is c++17 according to https://en.cppreference.com/w/cpp/types/is_swappable. Sound like that needs to be reported against CGAL.

waebbl, 2020-11-24:
Ok thank you, I'm going to check cgal then.

lrineau, 2020-11-27:
> CGAL manual states c++14 compatibility for version 5.1.1 [doc.cgal.org/latest/Manual/thirdparty.html](https://doc.cgal.org/latest/Manual/thirdparty.html) but `is_nothrow_swappable_v` is c++17 according to [en.cppreference.com/w/cpp/types/is_swappable](https://en.cppreference.com/w/cpp/types/is_swappable). Sound like that needs to be reported against CGAL.

Our usage of `is_nothrow_swappable_v` in CGAL-5.1.1 is protected by `#ifdef __cpp_lib_is_swappable`:
https://github.com/CGAL/cgal/blob/8b649c42a258e3db346f19cb3ae89eca5fea877d/Kernel_23/include/CGAL/Point_2.h#L87-L94

The problem is that there is a strange feature in gcc (version 10.2 at least):
```
[lrineau@bonnard]~% g++ --version                                                                                       
g++ (GCC) 10.2.1 20201016 (Red Hat 10.2.1-6)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

[lrineau@bonnard]~% g++ -std=c++14 -E -dM -xc++ /usr/include/c++/10/version | grep __cpp_lib_is_swappable
[lrineau@bonnard]~% g++ -std=gnu++11 -E -dM -xc++ /usr/include/c++/10/version | grep __cpp_lib_is_swappable
#define __cpp_lib_is_swappable 201603
[lrineau@bonnard]~% g++ -std=gnu++11 -E -dM -xc++ /usr/include/c++/10/version | grep __cpp_lib_is_swappable
#define __cpp_lib_is_swappable 201603
```

That means that `__cpp_lib_is_swappable` is defined with the compiler options `-std=gnu++11` and `-std=gnu++14` (but not `-std=c++11` or `-std=c++14`), even if it is supposed to be only in C++17.

*But* with `-std=gnu++11`, `<type_traits>` only defines the template class `std::is_nothrow_swappable` but not the template variable `std::is_nothrow_swappable_v` (because template variables are a core language feature of C++14).

So, that bug can only occur with `-std=gnu++11`. In my opinion, that is not a bug of CGAL, because CGAL-5.0 and CGAL-5.1 only support C++14 or later, and that bug cannot occur with `-std=c++14`, `-std=gnu++14`, `-std=c++17`, `std=gnu++17`, or anything more recent.

thehans, 2020-11-27:
@waebbl 
Current OpenSCAD master(HEAD) is built on c++14, but I guess 2019.05 release was just before we switched from c++11 to c++14.  

Do you still get errors if you patch qmake to use c++14 ?

waebbl, 2020-11-27:
@thehans I was just trying that. Backporting the patches from commit 9c4d5ae let's the package build successfully. 
Thank you all. Closing this issue.

closed 2020-11-27 by waebbl

t-paul, 2020-11-27:
Thanks @lrineau for the background info. That's a very strange issue. Going forward I tried to get rid of any setup using `gnu` variants, trying to enforce plain standard support (e.g. for qmake `CONFIG += c++14 strict_c++`). I'll make a note to also check that's happening in future `cmake` support.