openscad/openscad #3500: Object runs into a infinite loop

lampentom, 2020-11-25:
running the attached code, drawing the preview needs a long time also each redraw after any movment. running it under my old Athlon cpu desktop PC, it runs slowly but it works. Under my new HP notebook with Rythen 5 cpu the redraw leads into an infinite loop. Not possible to work with it.
[Turm - Kopie.txt](https://github.com/openscad/openscad/files/5597796/Turm.-.Kopie.txt)

lampentom, 2020-11-25:
Today the object makes also problems on my old Athlon. I modified the code to make it easier, but it does not help. After the first preview of the objekt, the cursor changes periodically into the running circle and working is no longer possible.
[Tower.txt](https://github.com/openscad/openscad/files/5598077/Tower.txt)

thehans, 2020-11-26:
There's no infinite loop, just inefficient script and too much complexity.

I made some optimizations by:
1) using `$fs` and `$fa` instead of `$fn`
2) reorganizing modules/transformations for better cache re-use
3) resized radial cylinders to not intersect in the center, slightly faster
4) used render() on some smaller parts to compute results in CGAL and cache them

In the future please use the mailing list/forums or IRC for help with scripting.

```
$fs=0.5;
$fa=2;
module stufe1(){
    difference(){
        union(){
        cylinder (h=7, d1=150, d2=130);
        cylinder (h=1, d=150);
        };
        translate ([0,0,9]) ring(147.5/2,16);
    };
  cylinder (h=60, d1=136, d2=106);
};

translate ([0,0,37]) ring(178/2,4);
difference(){
    translate ([0,0,24]) ring(112/2,72);
    cylinder(h=37*2, d=200, center=true);
    translate ([0,0,24]) ring(112/2,68);
};
translate ([0,0,55]) cylinder (h=5, d=112);

module stufe2voll(){
  for (i=[1:1:13]) rotate ([0,90,i*360/13]) {
    render() difference() {
      translate([0,0,110/2-8]) cylinder (h=8,d=18);
      translate ([0,0,106/2+3]) ring(18/2,6);
    }
  }
};
module stufe2hol(){
  render() for (i=[1:1:13]) rotate ([0,90,i*360/13]) {
    translate([0,0,111/4]) cylinder(h=111/4,d=11);
  }
};

module stufe2(){
    difference (){
        union() {
          translate ([0,0,60]) cylinder (h=7, d=108);
          translate ([0,0,60]) cylinder (h=80, d=106);
          translate ([0,0,82.5]) stufe2voll();
          translate ([0,0,102.5]) rotate ([0,0,360/26]) stufe2voll();
        }
        union() { 
          translate ([0,0,59]) cylinder (h=82, d=100);
          translate ([0,0,82.5]) stufe2hol();
          translate ([0,0,102.5]) rotate ([0,0,360/26]) stufe2hol();
        }
    };
};

module ring(dr,dw){ //dr = Ringdurchmesser, dw=Wulstdurchmesser
  rotate_extrude() translate([dr, 0, 0])
    circle(d=dw);
};
stufe1();
stufe2();
```

closed 2020-11-26 by thehans

MichaelAtOz, 2020-11-26:
@thehans beat me to it, but I've got pictures...

Your Turm - Kopie code has a few problems.
Firstly all those cylinders meeting in the middle, in stufe2():
[image]
That's just one layer.

And in stufe2ganz():
[image]

That creates a huge amount of work for OpenCSG trying to work out which bits are visible.

Also just make the cylinders as big as they need to be:
[image]
[image]

That improved it from almost infinite to very slow.

The $fn=100 makes for more work, the $fn=50 also, try lower values until you finished then higher values for the final product, or as thehans said, use $fa/$fs.

Also you have many cylinders with co-incident faces, ideally two should be either just apart, or just embedded, not just touching.

e.g.
```
s=1/128; s2=s*2;
module stufe1(){
  difference(){
    union(){
      cylinder (h=7, d1=150, d2=130);
      *cylinder (h=1, d=150); // not needed
    };
    translate ([0,0,9])
      ring(147.5/2,16);
  };
  translate([0,0,s])
    cylinder (h=60-s, d1=136, d2=106);
};
```

You have others aligned at z=60.

lampentom, 2020-11-26:
> Screenshot from 2020-11-26 13-24-00
> Screenshot from 2020-11-26 14-53-00
> Screenshot from 2020-11-26 15-04-25
> Screenshot from 2020-11-26 14-58-02

Hi Michael, hi Hans,

great, thanks for your support. Now I know there is a vital community. 
Sorry for using a defect instead of the mailing list/forum. for me it 
looks like a bug.

Reading your explanations I now understand where I have to keep an eye on.

It also shows me the limitations of OpenSCAD, currently I do not have 
more complex projects, so it ok.

One point I'm still wondering: What is the programm periodically doing 
in the background running in the described situation, unable to work 
with it any longer? I can understand that it will take a while to creat 
the preview or full compilation and I'm willing to wait therefor. But 
for me there is no need to do anything after that.

Thanks and regards

Thomas

MichaelAtOz, 2020-11-26:
> One point I'm still wondering: What is the programm periodically doing in the background running in the described situation, unable to work with it any longer? 

OpenSCAD Preview uses [OpenCSG](http://opencsg.org/), it uses graphics card techniques to project the front face of the objects, only those parts which are not hidden behind others. See [technical paper](http://wscg.zcu.cz/wscg2004/Papers_2004_Full/M11.pdf) particularly the 'The SCS Algorithm'.

So when you have lots of objects which intersect or difference, like those cylinders in the middle, it has lots of work to do, discovering which part is visible from the front and not hidden. The more faces, the more work; the $fn=100 applied to a subtracted ring * 13 * 2, results in mannnny faces it needs to place in z-depth order to work out what is visible.

That is why it is best to make objects clearly inside another (overlap) or clearly separate (air-gap) than touching. It makes it easier to work out which part is in front.

It has to do that again whenever you move the view, or it needs to repaint.

thehans, 2020-11-26:
Also related to what Michael is saying, see this FAQ: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_the_preview_so_slow?

Basically it has to convert the whole tree of boolean operations into a [Sum of Products](https://en.wikipedia.org/wiki/Canonical_normal_form) form.  So depending on the operations and how they are organized, this expression can end up exploding in the number of terms, even sometimes many multiple times the original number of geometric objects.

You can also see get an idea of the overall complexity in the Console at the end of preview, where a line says: 
`Normalized tree has 554 elements!`  from your Tower.txt, or 
`Normalized tree has 93 elements!`  after my edits

And if you are curious to see more info about these boolean expressions, you can also click on the OpenSCAD menu: Design -> Display CSG Products... to show a detailed text expression for what OpenCSG has to render.

Lastly keep in mind that a carefully tuned $fn (or $fs and $fa) is especially important in the case of sphere and torus.   Because the number of faces is proportional to `$fn` **squared** for them!