openscad/openscad #3504: WARNING: Normalized tree is growing past X elements. Aborting normalization.  Execution aborted

JoshMcCullough, 2020-11-29:
I'm confused about this warning: 

> Parsing design (AST generation)...
> Saved backup file: /home/josh/.local/share/OpenSCAD/backups/bbBrix-backup-OMLQKrmE.scad
> Compiling design (CSG Tree generation)...
> Compiling design (CSG Products generation)...
> Geometries in cache: 23
> Geometry cache size in bytes: 75304
> CGAL Polyhedrons in cache: 48
> CGAL cache size in bytes: 6276416
> Compiling design (CSG Products normalization)...
> **WARNING: Normalized tree is growing past 1200000 elements. Aborting normalization. **
> Execution aborted

In my settings, I have the limit set to 600,000, but the warning is stating that it reached twice that amount (1,200, 000):

[image]

Furthermore, once this warning appears, I have to close and reopen OpenSCAD before I can preview _or render_ again. Nothing happens when I click preview/render, no log, no rendering.

thehans, 2020-12-07:
Can you share a pared down scad script which reproduces this?

t-paul, 2020-12-07:
It seems to do that for a while. E.g. set config value to 1000 and
```
difference() {
    cube();
    for (a = [0:9]) {
        intersection() {
            sphere();
            cylinder();
        }
    }
}
```
-> `WARNING: Normalized tree is growing past 2000 elements. Aborting normalization. location unknown `

thehans, 2020-12-07:
Hmm, I see.  Looks like its been that way ever since there was a configurable limit added about 8yrs ago, but not sure why exactly.
https://github.com/openscad/openscad/commit/e93a23fdb8f478c2ebbda68df3d5c6d1441fc5bd

Actually, I can vaguely recall from a couple years ago when I was deep in the normalization code... i think due to something about the way normalization happens, it ends up over-counting the total nodes.  I wanted to fix that at the time but found the problem to be insurmountable.  So the CSG limit remains more of a general hint, not a hard limit.

I guess the more concerning part of this report is if its really preventing further preview/renders, because that should not be the case.  I have no such effect after generating the warning on 2019.05, Linux Mint 20, nor on nightly OpenSCAD.

JoshMcCullough, 2020-12-08:
You can grab [this repo](https://github.com/JoshMcCullough/bbBrix), open `bb-brix.scad`, set "turn off rendering at" to `10000` elements and add to the bottom put ...

```
Brick(8, 2);
```

--- and render the preview. You'll see this ...

>WARNING: Normalized tree is growing past 200000 elements. Aborting normalization. 
>Execution aborted

UBaer21, 2022-12-12:
Problem persists with windows and v 2022.11.27. 
After the Warning no preview or render is possible until restart.

nophead, 2022-12-12:
Try 1 million elements on the preferences.

UBaer21, 2022-12-12:
@nophead  how would that help.  differences with intersections cause a massive build up which can be solved by using render()  .. I am using 5 million but still hitting that limit - the problem is that i have to restart the program after that warning.
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_the_preview_so_slow%3F

So you do want a warning to inform you about this - but you want to work after you fixed the cause for the warning and not restart open scad.

nophead, 2022-12-12:
I haven't seen it for a long time. Probably because I changed my coding style, but I don't remember having to restart OpenSCAD after the warning. Have you tried just clearing the cache. What problem do you see if you don't restart OpenSCAD?

UBaer21, 2022-12-13:
Yes also new Window doesn't work. The F5/F6 is just unresponsive the preview is empty but can change the view. The console will have no further output.

nophead, 2022-12-13:
You probably need to add a render() somewhere. Also have you got the VBO rendering turned on? That is much faster and make it less likely the GUI will bog down.

UBaer21, 2022-12-13:
Yes as i said "render()" solves this but I don't need a solution - I need to be able to use openSCAD after i implemented the solution, The "Execution aborted" seems to be irreversible.
I have no option for vertex buffer object but activated all vertex object render (deactivation doesn't change behavior)

jstebbins, 2025-06-12:
I just ran into this today. I was debugging a model by using the '#' to highlight an element.  Everything seemed to be working as I wanted so I removed the '#'. Single character change to a fairly complex model and... *Ka-Boom*.

I increased elements to 1 million and that allowed me to preview, but extremely sluggishly. 'render()' works around the sluggishness.

A side note, after getting this warning, it is no longer possible to preview again until shutting down and restarting OpenSCAD.

jordanbrown0, 2026-01-21:
Since it seems to kill the preview attempt, "Warning" doesn't seem like the right label.

jordanbrown0, 2026-01-21:
I don't duplicate the "need to restart OpenSCAD" behavior on 2026.01.03 on Windows.