openscad/openscad #3509: Unable to Focus Error Log with Screen Reader

jage9, 2020-12-02:
Using OpenSCAD 2020.11.30.ci6340
With a screen Reader (I am using NVDA), there seems to be no way to bring keyboard focus to the error log, and potentially other windows.

From the main window, create code that causes an error and press F6.
Expected result: The error is displayed, and either focus would jump straight to the error or there would be a way to move focus manually.

Current result: The error is displayed, but screen reader-specific navigation commands are needed to read the text.

There are a couple of possible ideas. First, adding Ctrl+Tab and Ctrl+Shift+Tab to move between windows would allow for keyboard access. Additional shortcuts to jump to specific windows would also be helpful.

Once in an additional window, such as the error log, there should be a way to move keyboard focus through the list, for instance by using the arrow keys to go by line, or right and left arrow to move by column. Enter should move focus to the line of the error in question.

t-paul, 2020-12-12:
Could you give this version a try?

[Windows 64-bit ZIP](https://6434-1049088-gh.circle-artifacts.com/0/64-bit/OpenSCAD-2020.12.12.ci6434-x86-64_fix-keyboard-focus.zip)
[Windows 64-bit Installer](https://6434-1049088-gh.circle-artifacts.com/0/64-bit/OpenSCAD-2020.12.12.ci6434-x86-64_fix-keyboard-focus-Installer.exe)

This adds a `Window` menu and enables jump to source from the ErrorLog window by pressing the `Enter` key. I've added Prev/Next window too, but for now we can't use `Ctrl+Tab` and `Ctrl+Shift+Tab` as those are already used to navigate editor tabs. There's a change pending which makes the keyboard shortcuts configurable.

MichaelAtOz, 2020-12-13:
Works on W7.

> enables jump to source from the ErrorLog window by pressing the `Enter` key. 

That confused me for a while, I thought after Window/Error-log, press Enter to go to Editor...

\:|

No, Window-Error-log, scroll through the list of errors (with Up/Down keys) press Enter to go to the line with that error. 
I haven't played with Error-log.  

Thought-bubble, listing all the Editor tabs on the bottom of the 'Window' menu would also aid navigation...

closed 2020-12-13 by t-paul

t-paul, 2020-12-13:
I wanted to keep things minimal so we can include as accessibility fix for the release. It would be great to collect more ideas for improving accessibility.

commit d3239b10 (2020-12-13): Add window menu to allow keyboard access to all docked windows (fixes #3509).
commit d35dbf44 (2020-12-13): Merge pull request #3530 from openscad/fix-keyboard-focus

Add window menu to allow keyboard access to all docked windows (fixes #3509).
commit c2ee7ce2 (2020-12-17): Add window menu to allow keyboard access to all docked windows (fixes #3509).