openscad/openscad #3510: Parts that should be subtracted keep flashing up

eeucalyptus, 2020-12-03:
I am using OpenSCAD version 2020.11.29 from snap using Ubuntu 18.04 LTS with the following GL driver:
`Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel Open Source Technology Center (0x8086)
    Device: Mesa DRI Intel(R) UHD Graphics 620 (WHL GT2) (0x3ea0)
    Version: 20.0.8
    Accelerated: yes
    Video memory: 3072MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
`
In a model I wrote, some of the parts keep flashing up when rotating around the preview in perspective-mode. 
`$fn=100; 

difference() {
    scale([21, 21, 42]) {
        sphere(r=1);
    }
    
    translate([-100,-100,-100]) {
        cube([200,200,100]);
    }
}`
Here is a screen capture:
[image]
Let me know if further data is needed.

t-paul, 2020-12-03:
That happens when the camera ends up inside the object that is subtraced. I think that's an inherent property of the way preview works.

eeucalyptus, 2020-12-03:
I see. I'm not familiar with the code, but I guess that the blending is done using the clip plane. Shouldn't it be possible to (manually or automatically) adjust it to enable closer inspections? I know that I can just add a cut plane or perform a full render, but sometimes the behaviour bothers me

nophead, 2020-12-03:
Use a smaller cube, use an intersection instead of difference, or wrap it
with render().

thehans, 2020-12-03:
You can use "orthogonal" view to workaround this issue in most cases, but I don't believe its possible for it to ever be fixed in perspective preview.

For more explanation why see this old comment: https://github.com/openscad/openscad/issues/1591#issuecomment-457985428

t-paul, 2020-12-04:
As there's currently no way known how to handle this issue, I'm closing the ticket. In case someone reading this has a pointer to some algorithm / paper that could help, please reopen.

closed 2020-12-04 by t-paul