openscad/openscad #3511: The stack trace would be better if shown in reverse with the error message at the end

nophead, 2020-12-04:
I think it would be better if the warnings / errors and the stack trace where shown in the reverse order, so the error is last, preceded by the line that called it, etc. The reason being the log scrolls, so only the last n lines are shown. When the stack trace is shown it typically doesn't fit in the window size I have allocated for the log, so I have to scroll backwards to see the error message. 

If the order was reversed I would always be able to see the error message and where it was called from. I rarely need to go all the way down the stack, but obviously I could be scrolling backwards if I needed to instead of for every error I get.

"Called by" would need to become "called from".

MichaelPFrey, 2022-04-17:
I get the general idea of this request, but I would like to talk specifics of this idea.

```
module a(){
    b();
}

module b(){
    echo("test");
    c();
}

module c(){
    assert(false);
}

use<does not exist>

a();
```

currently, the console looks like
```
WARNING: Can't open library 'does not exist'. in file , line 14
Compiling design (CSG Tree generation)...
ECHO: "test"
ERROR: Assertion 'false' failed in file , line 11
TRACE: called by 'assert' in file , line 11
TRACE: called by 'c' in file , line 7
TRACE: called by 'b' in file , line 2
TRACE: called by 'a' in file , line 16
```

what would be easy is to do this with the error log:
[image]

just moving the error down would be easy:
```
WARNING: Can't open library 'does not exist'. in file , line 14
Compiling design (CSG Tree generation)...
ECHO: "test"
TRACE: called by 'assert' in file , line 11
TRACE: called by 'c' in file , line 7
TRACE: called by 'b' in file , line 2
TRACE: called by 'a' in file , line 16
ERROR: Assertion 'false' failed in file , line 11
```

duplicating the error would be easy:

```
WARNING: Can't open library 'does not exist'. in file , line 14
Compiling design (CSG Tree generation)...
ECHO: "test"
ERROR: Assertion 'false' failed in file , line 11
TRACE: called by 'assert' in file , line 11
TRACE: called by 'c' in file , line 7
TRACE: called by 'b' in file , line 2
TRACE: called by 'a' in file , line 16
ERROR: Assertion 'false' failed in file , line 11
```

Doing this:

```
WARNING: Can't open library 'does not exist'. in file , line 14
Compiling design (CSG Tree generation)...
INFO: module 'a' called in file , line 16
INFO: module 'b' called in file , line 2
ECHO: "test"
INFO: module 'c' called in file , line 7
INFO: module 'assert' called in file , line 11
ERROR: Assertion 'false' failed in file , line 11
```
is tricky. Either we need to generate the info message in advance or we need to somehow reorder the trace messages. (Note that I call that "INFO" for now)

**Generating the info messages in advance**
Generating the info messages in advance comes at a serious performance penalty - I do not want to have that in a normal build.
Putting the echo calls in the right place is trivial that way.

**Reordering the messages**

Reordering the messages means first collecting the messages and to only output them after all are collected (which can take time).
Putting the echo calls in the right place is difficult.

MichaelPFrey, 2022-04-17:
I would like to go into details about trace: In part, the trace looks like it does, because it was "easy" to implement (just catch and re-throw an exception that is already thrown to unwind the stack) and the implementation is (relatively) inexpensive during normal run time.

But there is more do it:

I have a background in Java programming, there _most inner call first_ is normal: https://en.wikipedia.org/wiki/Stack_trace#Java

Looking at the Wikipedia page, I assume you mean python style Traceback (_most recent call last_): https://en.wikipedia.org/wiki/Stack_trace#Python

That is tricky. Coming from java, _most inner call first_ looks normal, but it is easy to get used to _most recent call last_.

Coming from python, _most inner call first_  looks messy, as the most relevant information is hidden in the middle of noise (normal outputs, **error**, trace).

nophead, 2022-04-17:
Yes like Python.

I don't think it needs to be a big change taking echos into account. I don't expect those to be merged in as it isn't a chronological record. Just a LIFO stack to store and regurgitate the stack trace as it happens but if it is created by catch and re-throw I can see that would be tricky. I had assumed it was just a loop unwinding the stack.

My stacks are typically much deeper because they are often deep in a library called from a part deep in a hierarchical assembly. I tried to generate an example but I found I am only getting 13 levels of stack with OpenSCAD 2021.09.25.ci8698.  The outer levels are missing.

MichaelPFrey, 2022-04-17:
> Yes like Python.

OK - that is a good reference.

> Just a LIFO stack to store and regurgitate the stack trace as it happens

No, trace actually does not use any special data structures during normal run time.
_I assume try/catch has a hidden cost and there is some inline/noinline to prevent stack allocation happening for things that should not happen "most of the time". (The compiler does not know that the code is rarely executed nor that we need to optimize for stack usage during normal operation, not runtime during error handling)_

Note that most user module and user function calls cause a group of C++ functions to be called.
So when talking stack, I mean both "call stack caused by the user openscad file" and "c++ call stack (and stack memory in RAM) of openscad program it self". _Tail recursion is the exception to that - there openscad uses a loop._

> but if it is created by catch and re-throw I can see that would be tricky.

Yes, this is how TRACE is implemented.

> I had assumed it was just a loop unwinding the stack.

The trace is created when the c++ exception is handled, when the c++ stack is unwinding.

> My stacks are typically much deeper because they are often deep in a library called from a part deep in a hierarchical assembly. I tried to generate an example but I found I am only getting 13 levels of stack with OpenSCAD 2021.09.25.ci8698. The outer levels are missing.

That was/is an arbitrary limit, that has to be set somewhere as trace would get messy for recursive functions and modules. The test framework has a few of does, so any feature in or close to the language core has to handle that with stable, reproducible and not too intensive output.

With #4222 this limit will become adjustable. (Mainly because show parameters also decrements the trace-depth counter, despite not actually showing a trace level, as the trace output is partially redundant)