openscad/openscad #3512: Unable to close undocked windows in 2020.11.30.ci6340

nyponen, 2020-12-05:
When windows are undocked, and you close the preview window, the other windows are not closed, and cannot be closed normally. This occurs in the development snapshot 2020.11.30.ci6340 (git 9a08c222f)

MichaelAtOz, 2020-12-05:
I cannot reproduce this.
With 2020.11.30.ci6340/64, I undocked Editor & Console, leaving Preview with nothing else (I don't normally use Customizer or error log).
Clicking the close button in the Preview Titlebar, closed all three. Ditto File/Quit.
Rerunning, all three come up undocked as before.

Both Window-7/64 & Debian.

> cannot be closed normally

The close `X` button does nothing?

nyponen, 2020-12-06:
> I cannot reproduce this.
> With 2020.11.30.ci6340/64, I undocked Editor & Console, leaving Preview with nothing else (I don't normally use Customizer or error log).
> Clicking the close button in the [...]

When I click the 'X' button, Windows issues a popup with the message "openscad.exe is not responding" and gives a choice to "Close the program" or "Wait for the program to respond". When I click "Close the window" Windows reports the problem to Microsoft. This is on Windows 10.

MichaelAtOz, 2020-12-06:
Hmmm...loopy.scad....
Are you trying to close it while it is in still previewing?

thehans, 2020-12-07:
Sounds like duplicate of #1195 if that's the issue.

nyponen, 2021-02-08:
> Are you trying to close it while it is in still previewing?

No.

nyponen, 2021-02-08:
This still occurs in 2021.01 when OpenSCAD windows are undocked on separate monitors.

kintel, 2024-12-22:
@nyponen Following up: Does this still fail on a recent development snapshot?

nyponen, 2024-12-23:
I just tried this on 2024.12.22 (git 401f06ed4) and it seems to work correctly.

closed 2024-12-23 by kintel