openscad/openscad #3515: Critical error in render in 2020.11.30.ci6340

nyponen, 2020-12-06:
This occurs sporadically on Windows 10. The render does not complete. I don't get a stack trace or any other information. I opened this as #3514 a few minutes ago, but for some reason, it was marked "Closed".
[image]

t-paul, 2020-12-06:
I explained why it's closed. Without any additional info, there's nothing we can do. If you have some small code snippet that triggers the issue, someone can have a look. Also again: please add the detail info asked for in the linked FAQ entry. Otherwise I'll close this again as there's no action possible.

nyponen, 2020-12-06:
I gave you the operating system name and version and the OpenSCAD version. As I said, the problem is sporadic during render. There is no particular scenario or code snippet that causes the problem. There are certainly actions you could take to debug this problem. For example, you could add a stack trace that would show where the error was thrown from. That would give you a clue where the bug is.

MichaelAtOz, 2020-12-06:
He said please...

> Please read this [FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs?) for bug reports that can be processed.

"The bug report should give..."
"Most of the technical version information can..."

As it is a crash you could also look in Event Viewer, it may have something helpful.

> It should be easy to duplicate on Windows 10. 

Yes? 2020.11.30.ci6340 renders fine for me.

nyponen, 2020-12-07:
There was nothing in the Event Viewer.

Because of this crash and other bugs, 2020.11.30.ci6340 was too unstable for me to use, so I went back to 2019.05 and can't display the library and build info for 2020.11.30.ci6340.

t-paul, 2020-12-07:
Would you give the Release Candidate a try? https://www.openscad.org/downloads.html#rc

MichaelAtOz, 2020-12-09:
I've had a couple of crashes with RC2, both were when I was fiddling with multiple tabs and, at least one with customizer. Also may involve Animation. Also using s/w OpenCSG rendering. [yeh not helpful]

Library Info [Gist](https://gist.github.com/MichaelAtOz/a65ef7be310366dedb9c65137c40608c).

Latest crash, Eventviewer says was Exception code: 0xc0000005 Fault offset: 0x0000000001545a30, hazy memory says the other was too.

t-paul, 2021-07-27:
Closing due to no feedback.

closed 2021-07-27 by t-paul