openscad/openscad #3520: Feature request: Use multiple processes (at least on Mac)

whosawhatsis, 2020-12-08:
On windows, it is easy to launch multiple instances that each have their own process. The Mac app architecture assumes that there will never be more than one instance of an app open at once, and that the app will handle multiple windows and processes as necessary. Creating a second instance can only be launched as a terminal process (using the full path to the executable file inside the application package), and that terminal window has to stay open.

Using only a single process means that you can't have two preview/render processes running at the same time, and can't make use of multiple processor cores. It would be much more useful if each window was its own process (though it may still make sense for tabs to share a process, if that's necessary for them to share cache data).

whosawhatsis, 2021-03-09:
You're right, I forgot about the -n option. Still, it would be much nicer if you didn't have to launch multiple instances of the program (which is not how Mac apps are intended to be used) for each document to have its own process for previewing/rendering.

monroewilliams, 2021-03-09:
A better way to improve performance here would be to improve the way the OpenSCAD app handles threading/task management. Forking off multiple processes seems like it would complicate caching and cause other issues, what this really needs is for the render to happen with proper threading (or preferably dispatch queues, which better manage thread pools and relative priority) and have the render operations for different windows not block each other.

jordanbrown0, 2021-03-09:
MT makes a program much more complicated, and puts all of your eggs in one basket - if one thread hits a bug and crashes, you lose all of your windows/tabs/whatever.

monroewilliams, 2021-03-09:
Threading/MP is nontrivial to get right, but it's not _that_ hard, and it's necessary for good performance. I'm running on a machine with 24 cores, and I'm waiting ridiculously long times for a render while OpenSCAD pegs a single one of my CPUs. There's no way spawning multiple processes is going to speed up a single render, that needs actual multithreading.

jordanbrown0, 2021-03-09:
In general, it's *very* hard to get right.

But in limited cases - which this likely is - yes, it's only non-trivial.

kintel, 2024-11-28:
Closing as not planned

closed 2024-11-28 by kintel