openscad/openscad #3522: 2020.12.RC2 Customizer preset dialogue

MichaelAtOz, 2020-12-09:
Customizer active, click '+' button, dialogue 'Create New...', click Cancel, another dialogue pops-up.
Clicking Cancel clears it, but I had an instance when there were three.
Doesn't happen with 2020.11.30.

t-paul, 2020-12-27:
I can't reproduce that. I suspect that might be the text-cache issue (e.g. #3119).

MichaelAtOz, 2020-12-27:
No, not like that.
[image]
That '+' button, cancel brings up the dialogue again, two or three or more.
It appears to be related to multiple tabs, more tabs, more repeats.
RC3, 2020.12.12, 2020.12.14.

commit 1aa845a3 (2020-12-29): Merge pull request #3561 from openscad/fix-customizer-presets

Fix customizer presets (fixes #3522).
commit ca9ca26f (2020-12-29): Fix customizer presets (fixes #3522).
commit 74565b79 (2020-12-30): Fix customizer presets (fixes #3522).
closed 2020-12-29 by t-paul