openscad/openscad #3523: exporting STL triggers a coredump ( filedialog could not be displayed )

mjrider, 2020-12-09:
openscad version OpenSCAD-2020.12-RC2-x86_64.AppImage

output on the console:
```
(AppRun.wrapped:296322): Gtk-WARNING **: 16:29:52.025: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Mint-Y-Dark/actions/16/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /usr/lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end (gdk-pixbuf-error-quark, 5)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Mint-Y-Dark/actions/16/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /usr/lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end (gdk-pixbuf-error-quark, 5)
Aborted (core dumped)
```

did some googling and it seems to be a version incompatibility between the svg lib and cairo.

cairo is included in the appimage, but to old for the system libs, extracting the appimage, and replaceing the cairo lib fixes this.

thehans, 2020-12-24:
I'm on Linux Mint 20, and was able to reproduce.

Although it may be a library issue as you've said, I also noticed that it is specific to which icon theme is selected. "Mint-Y" (what I was running) and "Mint-Y-Dark" (from your console messages) both have the problem. 

But for example switching the Icon theme to "Mint-X" or "Mint-X-Dark" will also stop it from crashing, although it displays the "missing image" icons on all the folders.

Here is a screenshot of "Save As..." dialog with Mint-X icons:
[image]

Also there is still a warning message on the console when I do this, just no crash:
```
(AppRun.wrapped:2639794): Gtk-WARNING **: 16:14:46.483: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
```

JeffreyRodriguez, 2020-12-26:
I also get this same crash if I attempt to "Open" a file, from either the splash screen or File menu. Export STL crashes as well, anything that opens a file picker dialog.

OpenSCAD-2020.12-RC3-x86_64.AppImage

```
(AppRun.wrapped:1307395): Gtk-WARNING **: 11:10:09.925: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Mint-Y/actions/16/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /usr/lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end (gdk-pixbuf-error-quark, 5)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Mint-Y/actions/16/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /usr/lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end (gdk-pixbuf-error-quark, 5)
Aborted (core dumped)
```

[image]

I can confirm that @thehans workaround of changing to `Mint-X-Dark` instead of `Mint-Y-Dark` works for me, with the same console error:
```
(AppRun.wrapped:1308216): Gtk-WARNING **: 11:27:34.133: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
```

mattheath, 2021-01-03:
I'm getting the same crash with RC4 on Ubuntu 20.04.1 LTS when opening the file picker (as above, by clicking Open on the splash screen). There's no visual output, but when running from a terminal I get the folllowing:

```
./OpenSCAD-2020.12-RC4-x86_64.AppImage 

(AppRun.wrapped:2333279): Gtk-WARNING **: 21:40:09.254: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Humanity/status/16/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /usr/lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end (gdk-pixbuf-error-quark, 5)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Humanity/status/16/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /usr/lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end (gdk-pixbuf-error-quark, 5)
Aborted (core dumped)
```

KjellMorgenstern, 2021-01-16:
Similar crash but in my case  ( IoebSCAD-2021.01-RC6,  Ubuntu 20.04.1 LTS)
1. opened one of the examples
2. click "saved as"
=> Crash
```
~/appimage$ ./OpenSCAD-2021.01-RC6-x86_64.AppImage 

(AppRun.wrapped:41608): Gtk-WARNING **: 15:20:04.206: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Faba/16x16/status/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end (gdk-pixbuf-error-quark, 5)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Faba/16x16/status/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end (gdk-pixbuf-error-quark, 5)
Aborted (core dumped)

```
Let me know if you need the core dump.

M4rc3lv, 2021-02-19:
A work around that can be done by end users and that works for my Linux Mint system is copying all icons that are at `/usr/share/icons/Mint-X` but not at `/usr/share/icons/Mint-Y` from `/usr/share/icons/Mint-X` (recursively) to `/usr/share/icons/Mint-Y`. This way OpenScad does not crash when opening the file dialog.

t-paul, 2021-02-19:
Is the nightly snapshot AppImage having the same problem? (e.g. latest one https://files.openscad.org/snapshots/OpenSCAD-2021.02.17.ai7235-2021.02.17.ai7235-x86_64.AppImage)

This is work in progress to update the AppImage builds to a newer platform. It would be interesting to see if that helps with this issue.

M4rc3lv, 2021-02-19:
> Is the nightly snapshot AppImage having the same problem? (e.g. latest one https://files.openscad.org/snapshots/OpenSCAD-2021.02.17.ai7235-2021.02.17.ai7235-x86_64.AppImage)

Yes, it has the same problem.I tested it.

yagui, 2021-02-28:
@M4rc3lv I tried  `cp -r -n Mint-X/* Mint-Y` to copy only what is in Mint-X and not to overwrite any file in Mint-Y but I still getting this error. Is this the way you did it?

M4rc3lv, 2021-02-28:
> @M4rc3lv I tried `cp -r -n Mint-X/* Mint-Y` to copy only what is in Mint-X and not to overwrite any file in Mint-Y but I still getting this error. Is this the way you did it?

I did it with Nemo (file manager) started as `sudo Nemo`

yagui, 2021-02-28:
I used `sudo` when I ran `cp` (forgot to added, now I edited it) but I'm still getting this error.
Did you only copy files between those two folders? or any other folder?
Did you overwrite any file present both in Mint-X and Mint-Y? or just copied what was absent?
Thanks!

M4rc3lv, 2021-02-28:
I only copied files between those folders (recursively) from X to Y (I use Mint Y as a theme)  and I did not overwrite any file, so the end result was that only files from X that were not in Y were copied.

Maybe you should test that it works first by setting Mint-X as your theme, and then test OpenScad. I use version version 2021.01 (git 41f58fe).

yagui, 2021-02-28:
When I went to select Mint-X I saw that for some reason I was using a 'custom' theme instead the one I once selected, don't know why. Once I selected Mint-Y-Dark it started working. Thanks

sixtyfive, 2021-05-14:
DXF, too.

atroly, 2021-06-15:
The OP applied what appears to be a straightforward fix for this in December 2020 - update the version of cairo in the AppImage. Is there a reason why this can't be done for the release version?

atroly, 2021-06-15:
If anyone would like a step-by-step version of mjrider's solution, which does not involve using outdated icon themes:

Open a terminal and cd to the directory containing the OpenSCAD AppImage.

Extract the AppImage:
`./OpenSCAD-2021.01-x86_64.AppImage --appimage-extract`
(This creates a directory called squashfs-root)

Find your current libcairo shared library and copy it over the OpenSCAD version:
`cp /usr/lib/x86_64-linux-gnu/libcairo.so.2 ./squashfs-root/usr/lib/libcairo.so.2`

Download the AppImage creation tool and make it executable:
`wget https://github.com/AppImage/AppImageKit/releases/download/continuous/appimagetool-x86_64.AppImage`
`chmod +x appimagetool-x86_64.AppImage`

Make a new AppImage using the updated directory:
`./appimagetool-x86_64.AppImage squashfs-root`

This will create an updated AppImage named OpenSCAD-x86_64.AppImage

This worked for me, and OpenSCAD no longer reports warning on the console when opening files.

ianrolfe, 2021-09-30:
I tried the instructions in @atroly post and they fixed the crashing on save of a new file for me. It was very straightforward.

ipsod, 2022-05-15:
I had this issue, but updating to the latest AppImage fixed it.

Zefiro, 2022-06-23:
Thank you.
I had the same issue and this really helped me!

newest Linux Mint 20.3 (Una) (which package manager still ships the 2019.5 openscad)
newest Openscad AppImage 2021.1
crashed on every "save" or "open" (tried both with and without AppImageLauncher)

Both solutions above helped: switching the icon theming to "Mint X" as well as replacing the embedded libcairo version with the one on my system (which was a symlink to libcairo.so.2.11600.0), though not sure if that has further side-effects.
(I did not try the suggestion to "copy missing Y icons over from the X folder", as this felt too invasive to the rest of the system)

timnolte, 2025-06-27:
> This is still an issue with the latest available AppImage download. I'm running Pop!_OS 22.04. There hasn't been a new release in 4 years so this application almost seems to be dead and unsupported now. :-/

I was able to create a new AppImage with the update Cairo library as outlined and that does seem to fix the crash when trying to open a file.

kintel, 2025-06-27:
@timnolte By "latest available AppImage download" are you referring to the 2021.01 latest stable, or the 2025.06.23 latest nightly?

kintel, 2025-06-27:
https://openscad.org/downloads.html#snapshots

timnolte, 2025-06-27:
@kintel many projects like OrcaSlicer/PrusaSlicer/etc all have GitHub pre-releases setup for their nightly builds, I use this feature as well for my own projects for alpha & RC releases. So it is a feature that is just not being used here when it could.

[image]

t-paul, 2025-06-27:
I'm not sure it's a good idea to entangle the different services. Pushing releases from Circle-CI seems relatively easy, but it's just another thing to maintain. So at this point I would vote against it.

t-paul, 2025-06-27:
Another point is that this would still exclude at least Snap, Flatpak and miss the much easier use of the OBS repos for Debian/Ubuntu/Raspbian. Maybe the "beta-release" could just be a README pointing to the main website?

timnolte, 2025-06-27:
So there are already releases in GitHub so I'm not clear on how things would be any more tangled.

kintel, 2025-06-27:
We don't have automation for collecting and uploading binaries to GitHub releases, and automation is definitely needed for nightlies. Having a pointer to the website is reasonable though, to at least enable people coming here to find them.