openscad/openscad #3525: OpenSCAD crash durint rendering

ckhmer1, 2020-12-10:
Hi,
first of all, thanks for the great work, OpenSCAD is fantastic.

Unfortunately, it crashes with the following script (I'm trying to add a hole to the raspberry PI case found 
 here https://www.prusaprinters.org/prints/21430-pi4-case-fancover-wcamera-slot/files):

--------------------------
r=10.2/2;
r1=r+1;
h=13.9;
d=11;
difference() {
    union() {
        translate([4.5, 0, 0]) import("hexcase-base.stl", convexity=3);
        translate([0, 24.5-11, h]) rotate([0,90,0]) cylinder(h=3,r=r1,$fn=100);
    }
    translate([-1, 24.5-11, h]) rotate([0,90,0]) cylinder(h=20,r=r,$fn=100);
}
----------------------------

I've tried with the current stable version and the 2020 RC2 one.
If you tell me how to generate the logs, I'll provide them.

Thanks
Claudio

ckhmer1, 2020-12-10:
Hi,
I've also tried with another PC, please find below the details:

OpenSCAD version 2020.12-RC2 
Windows 10 64bit Version 20H2 (buildSO 19042.685)

Library info:

OpenSCAD Version: 2020.12-RC2
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 4 CPUs 15.91 GB RAM
User Agent: OpenSCAD/2020.12-RC2 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.3.7
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: unknown, 
Qt version: 5.13.0
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.2
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.4.0
cairo version: 1.16.0
lib3mf version: 1.8.0
Features: input-driver-dbus, lazy-union
Application Path: C:/Users/Claud/Desktop/openscad-2020.12-RC2
Documents Path: C:\Users\Claud\Documents
User Documents Path: C:\Users\Claud\Documents
Resource Path: C:/Users/Claud/Desktop/openscad-2020.12-RC2
User Library Path: C:/Users/Claud/Documents/OpenSCAD/libraries
User Config Path: C:\Users\Claud\AppData\Local/OpenSCAD
Backup Path: C:/Users/Claud/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/Claud/Documents/OpenSCAD/libraries
C:/Users/Claud/Desktop/openscad-2020.12-RC2\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
C:/WINDOWS/fonts
C:/Users/Claud/.local/share/fonts
C:/Users/Claud/.fonts
GLEW version: 1.12.0
OpenGL Version: 4.3.0 - Build 20.19.15.4835
GL Renderer: Intel(R) HD Graphics 4600
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QGLWidgetGL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
[... 182 lines cut ...]
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_ARB_buffer_region
WGL_ARB_create_context
WGL_ARB_create_context_profile
WGL_ARB_create_context_robustness
WGL_ARB_extensions_string
WGL_ARB_framebuffer_sRGB
WGL_ARB_make_current_read
WGL_ARB_multisample
WGL_ARB_pbuffer
WGL_ARB_pixel_format
WGL_ARB_pixel_format_float
WGL_EXT_create_context_es2_profile
WGL_EXT_create_context_es_profile
WGL_EXT_depth_float
WGL_EXT_extensions_string
WGL_EXT_pixel_format_packed_float
WGL_EXT_swap_control
WGL_EXT_swap_control_tear
WGL_INTEL_cl_sharing
WGL_NV_DX_interop
WGL_NV_DX_interop2

thehans, 2020-12-10:
Duplicate of #2847 
The crash comes from CGAL due to bad input.  The STL likely needs repairing.

closed 2020-12-10 by thehans

thehans, 2020-12-10:
The STL has at least one hole in the mesh.  I ran a preview of just the imported STL, using Thrown Together mode (F12) and you can see a sliver of magenta peeking through here.  That coloring means the back side of a face in the mesh is showing, so its not "watertight":
[image]