openscad/openscad #3529: Performance regression between 2019 release and 2020.12-RC2

thehans, 2020-12-11:
Originally reported by nophead on the mailing list: 
> Here is an example where the RC is 3 times slower than the release. https://github.com/nophead/NopSCADlib/blob/master/tests/pulleys.scad

OpenSCAD 2019.05
```
...
Used file cache size: 18 files
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 259
Geometry cache size in bytes: 7707888
CGAL Polyhedrons in cache: 9
CGAL cache size in bytes: 5571112
Compiling design (CSG Products normalization)...
Normalized CSG tree has 36 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 4 seconds
```

OpenSCAD 2020.12-RC2
```
...
Used file cache size: 18 files
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 245
Geometry cache size in bytes: 5961344
CGAL Polyhedrons in cache: 23
CGAL cache size in bytes: 21865784
Compiling design (CSG Products normalization)...
Normalized tree has 36 elements!
Compile and preview finished.
Total rendering time: 0:00:12.699
```

I have confirmed that its also about 3x slower on my own hardware running Linux Mint 20.

tt seems to be related to differences in cache handling, which can be noted in the console outputs differing significantly in number of objects and size.  

Still doing some debugging to trace down the exact cause, but I think it also has something to do with `render()` statements being used in this example.

nophead, 2020-12-12:
Is lazy union forced on in this release? If so that might be why different geometries get cached.

thehans, 2020-12-12:
No, lazy union is still considered incomplete and experimental.

The official RC is built with experimental features disabled, so its not even an option.

nophead, 2020-12-12:
I asked because the library info says it includes lazy-union as a feature but I have no features tab.

[image]

thehans, 2020-12-12:
The source is still technically there for those features, so it "knows" about their existence, but they should not have any way to actually be enabled (at least after the very recent fix here #3518, can't remember if that was just before or after RC2 was put out there).

If it were truly enabled, then there should be a `*` next to the feature name, based on the source:
https://github.com/openscad/openscad/blob/8afe126f59dc2f6ab048537eb01744e7961b0c5a/src/feature.cc#L82-L89

Perhaps we should omit that line from LibraryInfo for non-experimental builds, to avoid confusion (I wasn't even aware of it until now).

MichaelAtOz, 2020-12-13:
@thehans RC2 has the bug.

@nophead if you used a snapshot and enabled Lazy-union, it is enabled in RC2, run a snapshot to disable it in preferences, then RC2 won't use it.

nophead, 2020-12-13:
Looking at my registry I don't think I have it set, just every other feature that is no longer a feature!
[image]

Also I get similar time difference running from the command line.

thehans, 2021-01-02:
The reason for cache differences was identified while debugging for #3568, empty groups were being unioned with geometry (to no effect, except that it forces a conversion from Polyset to Nef polyhedron).  Whereas previously (before #3449) empty groups, and empty geometry were basically completely ignored by the code.

Unions now pre-filter empty groups so that union over a single geometry is once again is treated as NOOP and does not convert to Nef until truly needed. 

This was also partly alleviated by #3555 reducing the empty group occurences.  

Performance is pretty close to same now:
```
$ time openscad-nightly -o pulleys-nightly.png pulleys.scad
Compiling design (CSG Products normalization)...
Normalized CSG tree has 36 elements
Geometries in cache: 259
Geometry cache size in bytes: 7707888
CGAL Polyhedrons in cache: 9
CGAL cache size in bytes: 5571112
Total rendering time: 0:00:01.699

real    0m1.801s
user    0m1.551s
sys     0m0.068s

$ time openscad -o pulleys-2019.png pulleys.scad
Compiling design (CSG Products normalization)...
Normalized CSG tree has 36 elements

real    0m1.861s
user    0m1.544s
sys     0m0.064s
```

Closing this.  

Thanks again to @nophead for the report and let me know if you notice any other performance regressions.

closed 2021-01-02 by thehans