openscad/openscad #3531: 2020.12.RC2 - Warning file wrong

MichaelAtOz, 2020-12-12:
I opened RC2 with an existing .scad file.
File/New.
Enter:
```
$fn=12;
cylinder(10,h=10);
```
F5.
`WARNING: argument h supplied more than once in file Dyson 35 Holder, line 2`
(valid warning after thinking for a second)
The 'Dyson 35 Holder' is the directory of the file in Tab-1.

Open an example, logo.scad & repeat, it says 'Basics', the directory where logo.scad is.

Paste those two lines in a new instance with only one tab gets:
`WARNING: argument h supplied more than once in file openscad-2020.12-RC2, line 2`
The 'current directory' of the .exe file in 'program files'.

2020.11.30 on Debian does similar, (I haven't got RC2 there yet).

MichaelAtOz, 2020-12-12:
It's wrong in 2019.05 too...

MichaelAtOz, 2020-12-25:
I'm mentioning this here as it is probably related, I'll post a new issue if someone says otherwise.
It also saves to the filename of the first tab.
[image]

F5-F6-render-backup-zCuLPJJv.scad actually contains the rotate-vpr.scad code.

t-paul, 2026-04-25:
I believe I saw this recently, so it's probably still open.