openscad/openscad #3532: 2020.12.RC2 Truncated console output on CGAL error.

MichaelAtOz, 2020-12-12:


MichaelAtOz, 2020-12-12:
Oops - tip don't press Enter after typing the title...

[Forum](http://forum.openscad.org/2020-12-RC2-bogus-model-does-not-give-CGAL-error-td31096.html).

RC2. Render with CGAL error does not report cache stats or render time.
```
Parsing design (AST generation)...
Saved backup file: C:/Users/MeB/Documents/OpenSCAD/backups/adrianv-path_sweep-backup-AncrtKeh.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGALUtils::applyUnion3D: CGAL ERROR: assertion violation! Expr: G.mark(v1,0)==G.mark(v2,0)&& G.mark(v1,1)==G.mark(v2,1) File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_overlayer.h Line: 303, location unknown 
UI-WARNING: No top level geometry to render location unknown 
```

2019.05
```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: G.mark(v1,0)==G.mark(v2,0)&& G.mark(v1,1)==G.mark(v2,1) File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_overlayer.h Line: 303
Geometries in cache: 9
Geometry cache size in bytes: 1971896
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 4 seconds
Rendering finished.
```

MichaelAtOz, 2020-12-12:
Also should it say 'location unknown'?
`UI-WARNING: No top level geometry to render location unknown `

rcolyer, 2020-12-16:
It appears the old code flow went through applyOperator(...) which starts with a nullptr CGAL_Nef_polyhedron*, allocates it upon adding the first item, built it up gradually, and then upon encountering an error returned the partially constructed result.  The new code instead appears to go through the new applyUnion3D, constructs a priority_queue, and then only constructs a CGAL_Nef_polyhedron to return if it completes the process successfully.  Otherwise it returns a nullptr.  Then I suppose the stats are then not available if there's no top level geometry.  Given that it's a failure to perform the operation, I'm not sure that there's any advantage to returning a partially constructed result as was done previously.

MichaelAtOz, 2020-12-16:
Those stats are the cache, not the last rendered object.
I believe cache will be built, up to that point, even if the global render asserts.
Time is useful even if it fails.

A successful render shows:
```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
Geometries in cache: 2
Geometry cache size in bytes: 3184
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 22096
Total rendering time: 0:00:00.270
   Top level object is a 3D object:
   Simple:        yes
   Vertices:       16
   Halfedges:      48
   Edges:          24
   Halffacets:     20
   Facets:         10
   Volumes:         2
Rendering finished.
```
Those after the time are the object.

MichaelAtOz, 2020-12-16:
Also, I have a console scraper, which is not used much, but having consistent output, such as 'Total rendering time...' & 'Rendering finished.' is much easier to parse.

MichaelAtOz, 2020-12-16:
> 
> 
> The new code instead appears to go through the new applyUnion3D, constructs a priority_queue, and then only constructs a CGAL_Nef_polyhedron to return if it completes the process successfully. Otherwise it returns a nullptr. 

I'm not a deep C programmer, but I think the old code returned an empty object not a nullptr

https://github.com/openscad/openscad/blob/02da69559fda1583132fcd5965aae63e39681554/src/GeometryEvaluator.cc#L155-L158

I'm not a good Git driver either, so I'm not sure how to reference the commit, here goes [this one](https://github.com/openscad/openscad/commit/31c8fc904cb5bdbcd79262e222563b2b0f67328b#diff-22030df361bf07235d8df4eb92255e0777455ea10174e8e25706bfad26ddc96a).

commit ff49907d (2020-12-27): Show render statistics even if error resulted in no top level geometry (fixes #3532).
commit c49f9374 (2020-12-30): Show render statistics even if error resulted in no top level geometry (fixes #3532).
closed 2020-12-27 by t-paul