openscad/openscad #3535: Preview crashes for design with 53 Facets

FlipSchaumal, 2020-12-12:
Dear OpenScad team,

I'm trying to model a snail wheel for a striking clock. 
The target design is not too complex:
[image]

With the following code

```
snail_height = 3.6;
snail_par = [40.0, 2.0];

hours = [0 :  1];     // <-- preview works fine
// hours = [0 :  5];     // <-- preview works, but very slowly
// hours = [0 :  6];     // <-- preview crashes (does not finish / computes endlessly)
// hours = [0 : 11];     // <-- target

difference()
{
   cylinder(d = 60.0, h = snail_height, $fn = 18, center = true);

   for (ihour = hours)
      difference()
      {
         rotate(a = [0, 0, -(ihour - 0.25) * 360 / 12])
            translate(v = [0.0, -snail_par[0], 0.0])
               cylinder(d = 2 * snail_par[0] + snail_par[1], h = snail_height + 1.0, $fn = (ihour == 0 ? 50 : 16), center = true);
         rotate(a = [0, 0, -(ihour + 0.75) * 360 / 12])
            translate(v = [0.0, -snail_par[0], 0.0])
               cylinder(d = 2 * snail_par[0] - snail_par[1], h = snail_height + 2.0, $fn = 50, center = true);
         dia = 4 * snail_par[0] * sin(5.0 + ihour / 11 * 15.0);
         cylinder(d = dia, h = snail_height + 3.0, $fn = 30, center = true);
      }
}

```
the rendering works perfect, but the preview crashes (does not finish / computes endlessly) if the snail wheel has >6 steps which is as little as 53 Facets.

(The snail wheel is part of the complex clock design. For design development, rendering of the complete clock design takes FAR too long, so rendering is not an alternative for me.)

What can I do to get a preview for hours = [0 : 11]?

nophead, 2020-12-12:
Since the result is flat I would model it in 2D with circles and linear_extrude it by snail_height to be faster. But even then it seems to need to be wrapped in render() to draw quickly. This version only takes 60ms.

```
snail_height = 3.6;
snail_par = [40.0, 2.0];

//hours = [0 :  1];     // <-- preview works fine
 //hours = [0 :  5];     // <-- preview works, but very slowly
 //hours = [0 :  6];     // <-- preview crashes (does not finish / computes endlessly)
 hours = [0 : 11];     // <-- target

render() linear_extrude(snail_height) difference()
{
   circle(d = 60.0, $fn = 18);

   for (ihour = hours)
      difference()
      {
         rotate(a = [0, 0, -(ihour - 0.25) * 360 / 12])
            translate(v = [0.0, -snail_par[0], 0.0])
               circle(d = 2 * snail_par[0] + snail_par[1], $fn = (ihour == 0 ? 50 : 16));
         rotate(a = [0, 0, -(ihour + 0.75) * 360 / 12])
            translate(v = [0.0, -snail_par[0], 0.0])
               circle(d = 2 * snail_par[0] - snail_par[1], $fn = 50);
         dia = 4 * snail_par[0] * sin(5.0 + ihour / 11 * 15.0);
         circle(d = dia, , $fn = 30);
      }
}

```

[image]

FlipSchaumal, 2020-12-12:
Thanks for the VERY FAST and VERY HELPFUL answer! :-)

(So as a lessons learned, it is generally better to work in 2d as much as possible before moving to 3d by xx_etrude?)

FlipSchaumal, 2020-12-12:
Thanks for Open Scad :-)

closed 2020-12-12 by FlipSchaumal