openscad/openscad #3540: Feature request - log Design/Flush-cache to Console

MichaelAtOz, 2020-12-15:
When debugging, both my code & OpenSCAD issues, I regularly use Flush-cache to ensure a clean state.
It would be helpful when reviewing the history in the console to know when it was cleared.

Not urgent.

t-paul, 2020-12-22:
I was wondering how to add some kind of current time indicator in the console, any ideas how to make that work in a non-intrusive way? I don't think adding a long time stamp to every line makes sense. Maybe some kind of indicator left or right at some dedicated lines that would reveal the time as tooltip?

MichaelAtOz, 2020-12-25:
> I was wondering how to add some kind of current time indicator in the console

I'm still thinking about this.
It would need to be Accessible, and preferably able to be scraped (which it probably would be if accessible).

Purpose of time stamp in console:
1. know how long specific operations take, ie duration of AST/CSG-Tree/CSG-Products Generation etc.
2. know when something happened, ie time of day, potentially compare to file time-stamps etc.
3. ?

So these want time-stamps, probably date-time-stamps (where/how TBD)
```
Saved design 'C:/Users/MeB/Documents/3D-REPRAP/Things/My Things/tests/F5-F6-render.scad'.
Loaded design 'C:/Users/MeB/Documents/3D-REPRAP/Things/My Things/tests/rotate-vpr.scad'.
```
For consistency this should too, even though it is likely to have a time-stamp nearby in the log.
```
Saved backup file: C:/Users/MeB/Documents/OpenSCAD/backups/rotate-vpr-backup-APpzAPHb.scad
```

As an aside, file should be quoted & have full-stop at the end to be consistent with the the other messages, and all have a colon or all without.

Sections of logs happen consecutively with no need for time discrimination,
e.g. 
```
ECHO:$vpt = [0, 0, 0], $vpr = [0, 0, 0]
{potentially many ECHO's}
```
```
Geometries in cache: 46
Geometry cache size in bytes: 3672368
CGAL Polyhedrons in cache: 10
CGAL cache size in bytes: 58009752
```
```
   Top level object is a 3D object:
   Simple:        yes
   Vertices:     8077
   Halfedges:   40270
   Edges:       20135
   Halffacets:  24152
   Facets:      12076
   Volumes:         2
```

Warnings & Errors probably don't need times??

So for standard log lines (I haven't gone looking for all possible entries),
these are of interest
```
Preview

Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Compiling design (CSG Products normalization)...
Compiling highlights (2 CSG Trees)...
Compiling background (1 CSG Trees)...
Compile and preview finished.
Total rendering time: 0:00:00.050

Render

Parsing design (AST generation)...
Compiling design (CSG Tree generation)... 
Rendering Polygon Mesh using CGAL...
Total rendering time: 0:00:00.068
Rendering finished.
```
As an aside, `...finished` & `Total rendering time...` should be in the same order for F5 & F6, IMO `...finished.` should be last.

Suggestions
1. If no file is loaded, Log date-time after the `This program is free software; you can redistribute` initial console output. (if a file is loaded at startup the load will do a date-time)
`[yyyymmdd hh:mm:ss]` on a seperate line (including '[' & ']'),
2. Log that also prior to `Parsing design...`, 
3. On the messages with '...' replace that with `[hh:mm:ss]`,
4. Save/Load, and anything similar, like Export wants a date-time. 
Maybe just do as per 1./2. on a line prior,
5. For `Saved backup file:...` which ATM follows `Parsing design...`, 
explore if the backup can done/logged prior to the `Parsing design...`, so it will follow the date-time at 2.

e.g.
```
OpenSCAD 2020.12-RC3
https://www.openscad.org/

Copyright (C) 2009-2020 The OpenSCAD Developers

This program is free software; you can redistribute it and/or modify it under the terms of the 
GNU General Public License as published by the Free Software Foundation; 
either version 2 of the License, or (at your option) any later version.

[20201225 18:55:16]
Loaded design 'C:/Users/MeB/Documents/3D-REPRAP/Things/My Things/tests/rotate-vpr.scad'.

{F5}
[20201225 18:59:03]
Parsing design (AST generation)   
Compiling design (CSG Tree generation) [18:59:06]
ECHO:$vpt = [0, 0, 0], $vpr = [55, 0, 25]
Compiling design (CSG Products generation) [19:00:02]
Geometries in cache: 9
Geometry cache size in bytes: 302344
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization) [19:01:12]
Compiling highlights (2 CSG Trees) [19:01:15]
Normalized tree has 21 elements!
Total rendering time: 0:00:00.064 
Compile and preview finished [19:01:18]

{F6 a long time later...}
[20201225 23:58:03]
Parsing design (AST generation)
Compiling design (CSG Tree generation) [23:58:06]
ECHO:$vpt = [0, 0, 0], $vpr = [55, 0, 25]
Rendering Polygon Mesh using CGAL - 23:58:50   <== maybe it look cleaner without []??
Geometries in cache: 26
Geometry cache size in bytes: 314720
CGAL Polyhedrons in cache: 10
CGAL cache size in bytes: 58009752
Total rendering time: 0:00:25.225 
   Top level object is a 3D object: 
   Simple:        yes
   Vertices:     8077
   Halfedges:   40270
   Edges:       20135
   Halffacets:  24152
   Facets:      12076
   Volumes:         2
Rendering finished [00:25:20]

[20201216 00:26:01]
ASCIISTL export finished: C:/Users/MeB/Documents/3D-REPRAP/Things/My Things/tests/rotate-vpr.stl

[20201226 00:27:12]
Loaded design 'C:/Users/MeB/Documents/3D-REPRAP/Things/My Things/tests/F5-F6-render.scad'.
```

MichaelAtOz, 2020-12-26:
BTW when I said `to know when it was cleared.` I meant at what place in the console stream, not a time, i.e. just log 'Cache Flushed'.

But if you want to do times, I'm not against that.

In addition to the above time discussion, `--debug` output may need to be different, probably prefix [hh:mm:ss.mmm] like a normal log?

commit 6e04dc9f (2021-01-01): show message in console when caches are flushed  (fixes #3540)
commit e94f5b4c (2021-01-01): Merge pull request #3569 from donc42/flush_caches_message

show message in console when caches are flushed  (fixes #3540)
commit eb3025d7 (2021-01-03): show message in console when caches are flushed  (fixes #3540)
closed 2021-01-01 by t-paul