openscad/openscad #3543: Setting $vpr z-axis rotates 3D space not viewport - in my opinion

MichaelAtOz, 2020-12-16:
Reported to [forum](http://forum.openscad.org/Unexpected-vpr-behavior-td31033.html).

There is discussion on how to interpret what is being rotated.
There is also debate in what the viewport axes actually are.
This may require someone with history to know what the intention is/was.

BUT I think the viewport $vpr z-rotate is wrong.

See this to experience what is being rotated:

```
Rotate_x=false;
Rotate_y=false;
Rotate_z=false;

cube([100,10,10]);
cylinder(r=5,h=100,center=true);
$vpr=[  (Rotate_x) ? $t*360 : $vpr.x,
        (Rotate_y) ? $t*360 : $vpr.y,
        (Rotate_z) ? $t*360 : $vpr.z];
```

Show customizer, start animation, then select which axis to rotate dynamically.
Remember you can Shift-Drag to manually rotate-Y.

Select one axis at a time, then use the Top/Bottom/Left/Right/Front/Back cube
buttons to explore what is rotating.

nophead, 2020-12-17:
If you look at the axis labels the world rotates around the z axis when you select rotate around z.  That is the axis that points up on the screen.

What do you expect to happen?

nophead, 2020-12-17:
If you drag horizontally with the left mouse button it does the same thing and $vpr.z changes as you spin round world.

nophead, 2020-12-17:
The code that does it is here: https://github.com/openscad/openscad/blob/master/src/GLView.cc#L111

The only thing special about vpr.z is the order of the rotations, so no matter which view you use, it always rotates around the z axis in the world. With x you can get it to rotate around y if you look from some directions.

Since the order of rotations matters you would get a different effect if those statements were in a different order.

thehans, 2020-12-17:
Also the rotations are applied in an order which is consistent with `rotate`.

If it were changed, then I don't think you could do this neat trick (try appending to your example):
```OpenSCAD
translate($vpt)           // camera translation independent
 rotate($vpr)             // camera rotation independent
  translate([0,0,$vpd/2]) // bring to foreground so not intersecting with scene in most cases
   scale($vpd/500)        // zoom independent
    color([1,0,0]) linear_extrude(10) text("oh hi!", halign="center");
```
One use case of that is to be able to label objects in a way that is pretty much always visible to the user, as long as animation is on, or preview is otherwise refreshed.

thehans, 2020-12-17:
The above is example acts kinda like a screen overlay.
Here is the more basic example of how you could make a user-facing label, positioned next to an object:
```OpenSCAD
Rotate_x=false;
Rotate_y=false;
Rotate_z=false;

cube([100,10,10]);
cylinder(r=5,h=100,center=true);
$vpr=[  (Rotate_x) ? $t*360 : $vpr.x,
        (Rotate_y) ? $t*360 : $vpr.y,
        (Rotate_z) ? $t*360 : $vpr.z];

// position label at end of cube
translate([100,10,10]) rotate($vpr)
    color([0,1,1]) linear_extrude(10) text("p.s. look over here!");
```

jordanbrown0, 2020-12-17:
I think it's doing what it's intended to do, and what it needs to do (for compatibility, if nothing else).  The problem is, I think, in **describing** what it does.

`$vpt` is a point in the model space.  That point is always in the middle of the viewport.  (Note that panning and zooming with the mouse change `$vpt`.)

`$vpr`, and mouse left button "rotation", cause the camera to **orbit** around $vpt.  The camera always faces `$vpt`, by the definition of `$vpt` above, so the camera does turn, but that's not the primary effect - the camera both moves around its orbit **and** turns so that it continues to face $vpt.

You can view this orbiting as moving the camera, or you can view it as rotating the model **around** `$vpt`.

nophead, 2020-12-17:
I think another way to look at it is there are three rotations between the viewport and the world. The one nearest the world appears to rotate the world, the one nearest the viewport end of the transformation stack appears to rotate the viewport.

jordanbrown0, 2023-12-11:
I noticed today that the relationship between the light source and the camera is fixed, so either you're rotating the model while they stay fixed, or the two of them are orbiting the model together.

kintel, 2024-12-22:
Closing as works as intended

closed 2024-12-22 by kintel